bluez/bluez #788: ABI break in  5.73

bbhtt, 2024-03-23:
Hi, going from 5.70 to 5.73, we are seeing an ABI break:

```
        ┌──────────────────────────────┐
        │ ABI Break: libbluetooth.so.3 │
        └──────────────────────────────┘
        Functions changes summary: 0 Removed, 1 Changed (2 filtered out), 0 Added functions
        Variables changes summary: 0 Removed, 0 Changed, 0 Added variable
        1 function with some indirect sub-type change:
          [C] 'function int sdp_get_string_attr(const sdp_record_t*, uint16_t, char*, int)' at sdp.c:2182:1 has some indirect sub-type changes:
            parameter 4 of type 'int' changed:
              entity changed from 'int' to compatible type 'typedef size_t' at stddef.h:214:1
                type name changed from 'int' to 'unsigned long int'
                type size changed from 32 to 64 (in bits)
```

Is this expected?

jonas2515, 2024-04-09:
That was https://github.com/bluez/bluez/commit/b978f979678d237f406f3209cf2d5f2da8e5b74e, that change is not strictly necessary as the string in question (`sdpdata->val.str`) should never be so long that the length can't fit into a 32 bit int. I'm assuming this could be safely reverted, but I guess since it already slipped through and the ABI break has happened, it's kinda also too late for that now.

bbhtt, 2024-04-09:
Yes we froze it already. Thanks for the reply.

closed 2024-04-09 by bbhtt