bluez/bluez #793: Intel BE200: Invalid exception type 03 & 04

drujd, 2024-03-27:
Now that firmware 53617 landed for my BE200NGW card and finally added cis-central/peripheral support, I am trying to make LE Audio work with my LE Audio enabled Sony WF-1000XM5 earbuds (which work almost perfectly with my Android phone & tablet). However, this just does not seem to work. I have tried this on various combinations of:

- Linux kernel: 6.8.1, 6.8.2 & latest 6.9 master (962490525cffee6270422206c9d5bdb12d472c2b) with latest bluetooth-next (db4597cc88b2621706dc39a7363e61f4e7734425) merged into it
- bluez 5.73 & latest master (41d6c4e1c)
- pipewire/wireplumber: stable & latest git versions

```
~ $ sudo journalctl -b | grep -iE 'blue|hci0|ibt|be200'
bře 27 21:34:17 box kernel: Bluetooth: Core ver 2.22
bře 27 21:34:17 box kernel: NET: Registered PF_BLUETOOTH protocol family
bře 27 21:34:17 box kernel: Bluetooth: HCI device and connection manager initialized
bře 27 21:34:17 box kernel: Bluetooth: HCI socket layer initialized
bře 27 21:34:17 box kernel: Bluetooth: L2CAP socket layer initialized
bře 27 21:34:17 box kernel: Bluetooth: SCO socket layer initialized
bře 27 21:34:17 box kernel: Bluetooth: hci0: Firmware timestamp 2024.8 buildtype 1 build 53617
bře 27 21:34:17 box kernel: Bluetooth: hci0: Firmware SHA1: 0x7c7b67b9
bře 27 21:34:17 box kernel: Bluetooth: hci0: No support for _PRR ACPI method
bře 27 21:34:17 box kernel: Bluetooth: hci0: Found device firmware: intel/ibt-0291-0291.sfi
bře 27 21:34:17 box kernel: Bluetooth: hci0: Boot Address: 0x100800
bře 27 21:34:17 box kernel: Bluetooth: hci0: Firmware Version: 113-8.24
bře 27 21:34:17 box kernel: Bluetooth: hci0: Firmware already loaded
bře 27 21:34:17 box kernel: Bluetooth: hci0: Fseq status: Success (0x00)
bře 27 21:34:17 box kernel: Bluetooth: hci0: Fseq executed: 00.00.04.157
bře 27 21:34:17 box kernel: Bluetooth: hci0: Fseq BT Top: 00.00.04.157
bře 27 21:34:22 box systemd[1]: Starting Bluetooth service...
bře 27 21:34:22 box bluetoothd[1325]: Bluetooth daemon 5.73
bře 27 21:34:22 box systemd[1]: Started Bluetooth service.
bře 27 21:34:22 box systemd[1]: Reached target Bluetooth Support.
bře 27 21:34:22 box bluetoothd[1325]: Excluding (cli) a2dp
bře 27 21:34:22 box bluetoothd[1325]: Excluding (cli) vcp
bře 27 21:34:22 box bluetoothd[1325]: Bluetooth management interface 1.22 initialized
bře 27 21:34:22 box bluetoothd[1325]: src/adapter.c:reset_adv_monitors_complete() Failed to reset Adv Monitors: Failed (0x03)
bře 27 21:34:22 box bluetoothd[1325]: Battery Provider Manager created
bře 27 21:34:22 box kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
bře 27 21:34:22 box kernel: Bluetooth: BNEP filters: protocol multicast
bře 27 21:34:22 box kernel: Bluetooth: BNEP socket layer initialized
bře 27 21:34:22 box kernel: Bluetooth: MGMT ver 1.22
bře 27 21:34:22 box kernel: Bluetooth: ISO socket layer initialized
bře 27 21:34:22 box bluetoothd[1325]: profiles/audio/avrcp.c:avrcp_controller_server_probe() Unable to register AVRCP service record
bře 27 21:34:22 box bluetoothd[1325]: avrcp-controller: Operation not permitted (1)
bře 27 21:34:22 box bluetoothd[1325]: profiles/audio/avrcp.c:avrcp_target_server_probe() Unable to register AVRCP target service record
bře 27 21:34:22 box bluetoothd[1325]: audio-avrcp-target: Operation not permitted (1)
bře 27 21:34:22 box bluetoothd[1325]: Failed to clear UUIDs: Failed (0x03)
bře 27 21:34:22 box bluetoothd[1325]: Adv Monitor Manager created with supported features:0x00000001, enabled features:0x00000001, max number of supported monitors:32, max number of supported patterns:16
bře 27 21:34:35 box bluetoothd[1325]: Path / reserved for Adv Monitor app :1.17
bře 27 21:35:11 box kernel: Bluetooth: hci0: Invalid exception type 03
bře 27 21:35:11 box systemd[1566]: Reached target Bluetooth.
bře 27 21:35:16 box kernel: Bluetooth: hci0: Invalid exception type 03
bře 27 21:35:16 box bluetoothd[1325]: profiles/battery/battery.c:parse_battery_level() Trying to update an unregistered battery
bře 27 21:35:24 box bluetoothd[1325]: profiles/battery/battery.c:parse_battery_level() Trying to update an unregistered battery
bře 27 21:35:55 box bluetoothd[1325]: Adv Monitor app :1.17 disconnected from D-Bus
bře 27 21:36:27 box bluetoothd[1325]: src/profile.c:ext_register_record() Failed to register service record
bře 27 21:36:27 box bluetoothd[1325]: src/profile.c:ext_register_record() Failed to register service record
bře 27 21:36:27 box kernel: Bluetooth: RFCOMM TTY layer initialized
bře 27 21:36:27 box kernel: Bluetooth: RFCOMM socket layer initialized
bře 27 21:36:27 box kernel: Bluetooth: RFCOMM ver 1.11
bře 27 21:36:28 box systemd[1566]: Starting Bluetooth OBEX service...
bře 27 21:36:28 box systemd[1566]: Started Bluetooth OBEX service.
bře 27 21:36:28 box bluetoothd[1325]: src/profile.c:ext_register_record() Failed to register service record
bře 27 21:36:28 box bluetoothd[1325]: src/profile.c:ext_register_record() Failed to register service record
bře 27 21:36:28 box bluetoothd[1325]: src/profile.c:ext_register_record() Failed to register service record
bře 27 21:36:28 box bluetoothd[1325]: src/profile.c:ext_register_record() Failed to register service record
bře 27 21:36:28 box bluetoothd[1325]: src/profile.c:ext_register_record() Failed to register service record
bře 27 21:36:28 box bluetoothd[1325]: src/profile.c:ext_register_record() Failed to register service record
bře 27 21:36:28 box bluetoothd[1325]: src/profile.c:ext_register_record() Failed to register service record
bře 27 21:39:40 box kernel: Bluetooth: hci0: Invalid exception type 04
bře 27 21:39:40 box kernel: Bluetooth: hci0: Invalid exception type 04
```

Am I attempting something unsupported yet? I get Invalid exception type 03 after pairing, 04 after disconnecting. Also of note is the 'No support for _PRR ACPI method', 'Failed to reset Adv Monitors: Failed (0x03)' and 'Failed to clear UUIDs: Failed (0x03)' errors. I have disabled classic stack, so there are errors linked to that, but that is normal.

Vudentz, 2024-03-27:
@drujd since when these earbuds start working with LE Audio? It is probably a good idea to collect the HCI trace using btmon so we can see what is going at HCI that cause these exceptions.

drujd, 2024-03-27:
> @drujd since when these earbuds start working with LE Audio? It is probably a good idea to collect the HCI trace using btmon so we can see what is going at HCI that cause these exceptions.

Sony allows to enable LE Audio only in their app and only on Xperia phones (and that has been true for half a year at least), but there is a modified app that allowed me to enable it on my Samsung phone: https://github.com/eiarhabi/MDR_LEAUDIO

And it works! On both my Samsung S23 Ultra and Tab S9+ - there are some quirks, but they are well worth the latency reduction.

Github won't let me upload the btsnoop file (and maybe it's for the best the MAC addresses remain non-public), so I will send that over e-mail.

drujd, 2024-03-29:
With the latest 6.9 kernel+bluetooth-next, pipewire compield from master and bluez compiled from master, removing the disablement of a2dp & vcp and allowing all bluez experimental features, I am now NOT getting the Invalid exception type XX errors, but instead:

```
bře 29 11:45:28 box wireplumber[1764]: Acquire /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_source0/fd0 returned error: org.bluez.Error.Failed
bře 29 11:45:28 box wireplumber[1764]: (bluez_input.C6_FC_CD_99_12_86.0-64) suspended -> error (Received error event)
bře 29 11:45:28 box wireplumber[1764]: Failure in Bluetooth audio transport /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_source0/fd0
bře 29 11:45:28 box wireplumber[1764]: (bluez_output.C6_FC_CD_99_12_86.1-65) running -> error (Received error event)
bře 29 11:45:28 box wireplumber[1764]: Failure in Bluetooth audio transport /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_sink0/fd1
bře 29 11:45:28 box pipewire[1760]: pw.node: (bluez_input.C6_FC_CD_99_12_86.0-64) suspended -> error (Received error event)
bře 29 11:45:28 box pipewire[1760]: pw.node: (bluez_output.C6_FC_CD_99_12_86.1-65) running -> error (Received error event)
bře 29 11:45:33 box wireplumber[1764]: Failed to release transport /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_sink0/fd1: Operation Not Authorized
```

Rishav1818, 2024-03-31:
Encountring same error

Rishav1818, 2024-03-31:
Mar 01 11:52:12 rishav-IdeaPad-Flex-5-14ARE05 bluetoothd[529]: src/adapter.c:reset_adv_monitors_complete() Failed to reset Adv Monitors: Failed (0x03)
Mar 01 11:52:14 rishav-IdeaPad-Flex-5-14ARE05 bluetoothd[529]: Failed to clear UUIDs: Failed (0x03)
Mar 01 11:52:14 rishav-IdeaPad-Flex-5-14ARE05 bluetoothd[529]: Failed to add UUID: Failed (0x03)
Mar 01 11:52:14 rishav-IdeaPad-Flex-5-14ARE05 bluetoothd[529]: Failed to add UUID: Failed (0x03)
Mar 01 11:52:14 rishav-IdeaPad-Flex-5-14ARE05 bluetoothd[529]: Failed to add UUID: Failed (0x03)
Mar 01 11:52:14 rishav-IdeaPad-Flex-5-14ARE05 bluetoothd[529]: Failed to set mode: Failed (0x03)

drujd, 2024-04-08:
Sony released a new firmware for WF-1000XM5: https://www.sony.com/electronics/support/wireless-headphones-bluetooth-headphones/wf-1000xm5/software/00296561

LE Audio now officially works with all supporting devices, not just Xperia phones. From my short testing, it now works perfectly with my phone/tablet. Still no dice with Intel BE200 controller, though - LE Audio still does not work in Linux (nor in Windows).

Opendreas, 2024-04-09:
I have Intel Wireless-AC 9260 Bluetooth Adapter and the same errors when booting the system.

`bluetoothd	src/adapter.c:reset_adv_monitors_complete() Failed to reset Adv Monitors: Failed (0x03)
bluetoothd	Failed to clear UUIDs: Failed (0x03)
bluetoothd	Failed to add UUID: Failed (0x03)
bluetoothd	Failed to add UUID: Failed (0x03)
bluetoothd	Failed to add UUID: Failed (0x03)
bluetoothd	Failed to add UUID: Failed (0x03)
bluetoothd	Failed to add UUID: Failed (0x03)
bluetoothd	Failed to add UUID: Failed (0x03)`

These errors are present on kernels 6.4, 6.5, 6.8, 6.9-rc1 and bluez 5.71.

I also have a Sony WH-1000XM5, but the only error that appears when connecting is the following
`profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info`

I don’t know if it makes sense to open a new bug report.

[hwinfo.txt](https://github.com/bluez/bluez/files/14922905/hwinfo.txt)
[journald-6.4.0.txt](https://github.com/bluez/bluez/files/14922906/journald-6.4.0.txt)
[journald-6.5.9.txt](https://github.com/bluez/bluez/files/14922908/journald-6.5.9.txt)
[journald-6.8.2.txt](https://github.com/bluez/bluez/files/14922909/journald-6.8.2.txt)
[journald-6.9.0-rc1.txt](https://github.com/bluez/bluez/files/14922910/journald-6.9.0-rc1.txt)

drujd, 2024-04-10:
Intel 9260 will NEVER work with LE Audio, so whatever you are experiencing is a different issue. The "Failed to add UUID: Failed (0x03)" error might indicate some common issue, but the point is nonfunctioning LE Audio on BE200.
 
 Please create a separate issue.

Opendreas, 2024-04-10:
> Intel 9260 will NEVER work with LE Audio, so whatever you are experiencing is a different issue. The "Failed to add UUID: Failed (0x03)" error might indicate some common issue, but the point is nonfunctioning LE Audio on BE200.
> 
> Please create a separate issue.

I created a new issue. It would be nice if everyone who has the same errors wrote about them.
https://github.com/bluez/bluez/issues/809

drujd, 2024-04-10:
Worth noting is that I also see the earbuds as two separate devices, aren't CSIP devices supposed to be "merged" as one device?

pv, 2024-04-10:
> aren't CSIP devices supposed to be "merged" as one device?

See https://gitlab.freedesktop.org/pipewire/pipewire/-/wikis/LE-Audio-+-LC3-support

I guess for debugging something here, btmon traffic dump and/or debug logs would be necessary. (You can attach btmon files, but they need to be renamed to end with `.log` (or `.txt`) I think github only cares about the file name.)

drujd, 2024-04-10:
I did see the wiki page and set everything accordingly. I also sent btsnoop to @Vudentz (didn't know about the suffix workaround).

drujd, 2024-04-10:
Wireplumber log: 
[wireplumber.log](https://github.com/bluez/bluez/files/14934228/wireplumber.log)

drujd, 2024-04-10:
Ok, here are the logs of pairing. Something seems to be wrong with matching the two LE devices into a CSIP set.

dmesg (these errors seem gone on 6.9 rc kernel):
```
[  290.754044] Bluetooth: hci0: Invalid exception type 03
[  295.450042] Bluetooth: hci0: Invalid exception type 04
[  295.699044] Bluetooth: hci0: Invalid exception type 03
[  297.660038] Bluetooth: hci0: Invalid exception type 03
```

bluetoothd:
```
dub 10 18:11:36 box bluetoothd[1337]: src/profile.c:register_profile() sender :1.27
dub 10 18:11:36 box bluetoothd[1337]: src/profile.c:create_ext() Created "Object Push"
dub 10 18:11:36 box bluetoothd[1337]: src/profile.c:ext_adapter_probe() "Object Push" probed
dub 10 18:11:36 box bluetoothd[1337]: src/profile.c:ext_start_servers() Object Push listening on PSM 4105
dub 10 18:11:36 box bluetoothd[1337]: src/profile.c:ext_start_servers() Object Push listening on chan 9
dub 10 18:11:36 box bluetoothd[1337]: src/profile.c:ext_register_record() Failed to register service record
dub 10 18:11:36 box bluetoothd[1337]: src/profile.c:register_profile() sender :1.27
dub 10 18:11:36 box bluetoothd[1337]: src/profile.c:create_ext() Created ":1.27/org/bluez/obex/00005005_0000_1000_8000_0002ee000001/00005005-0000-1000-8000-0002ee000001"
dub 10 18:11:36 box bluetoothd[1337]: src/profile.c:ext_adapter_probe() ":1.27/org/bluez/obex/00005005_0000_1000_8000_0002ee000001/00005005-0000-1000-8000-0002ee000001" probed
dub 10 18:11:36 box bluetoothd[1337]: src/profile.c:ext_register_record() Failed to register service record
dub 10 18:14:58 box bluetoothd[1337]: src/profile.c:register_profile() sender :1.55
dub 10 18:14:58 box bluetoothd[1337]: src/profile.c:create_ext() Created "Hands-Free Voice gateway"
dub 10 18:14:58 box bluetoothd[1337]: src/profile.c:ext_adapter_probe() "Hands-Free Voice gateway" probed
dub 10 18:14:58 box bluetoothd[1337]: src/profile.c:ext_start_servers() Hands-Free Voice gateway listening on chan 13
dub 10 18:14:58 box bluetoothd[1337]: src/profile.c:ext_register_record() Failed to register service record
dub 10 18:14:58 box bluetoothd[1337]: src/profile.c:register_profile() sender :1.55
dub 10 18:14:58 box bluetoothd[1337]: src/profile.c:create_ext() Created "Hands-Free unit"
dub 10 18:14:58 box bluetoothd[1337]: src/profile.c:ext_adapter_probe() "Hands-Free unit" probed
dub 10 18:14:58 box bluetoothd[1337]: src/profile.c:ext_start_servers() Hands-Free unit listening on chan 7
dub 10 18:14:58 box bluetoothd[1337]: src/profile.c:ext_register_record() Failed to register service record
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:register_app() Registering application: :1.55:/MediaEndpoint
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:register_app() Registering application: :1.55:/MediaEndpointLE
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/ldac, iface: org.bluez.MediaEndpoint1
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:media_endpoint_create() Unable initialize endpoint for UUID 0000110a-0000-1000-8000-00805f9b34fb
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:media_endpoint_destroy() sender=:1.55 path=/MediaEndpoint/A2DPSource/ldac
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:app_register_endpoint() Unable to register endpoint :1.55:/MediaEndpoint/A2DPSource/ldac: Success
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSink/aptx_hd, iface: org.bluez.MediaEndpoint1
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:media_endpoint_create() Unable initialize endpoint for UUID 0000110b-0000-1000-8000-00805f9b34fb
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:media_endpoint_destroy() sender=:1.55 path=/MediaEndpoint/A2DPSink/aptx_hd
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:app_register_endpoint() Unable to register endpoint :1.55:/MediaEndpoint/A2DPSink/aptx_hd: Success
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/aptx_hd, iface: org.bluez.MediaEndpoint1
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:media_endpoint_create() Unable initialize endpoint for UUID 0000110a-0000-1000-8000-00805f9b34fb
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:media_endpoint_destroy() sender=:1.55 path=/MediaEndpoint/A2DPSource/aptx_hd
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:app_register_endpoint() Unable to register endpoint :1.55:/MediaEndpoint/A2DPSource/aptx_hd: Success
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSink/aptx, iface: org.bluez.MediaEndpoint1
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:media_endpoint_create() Unable initialize endpoint for UUID 0000110b-0000-1000-8000-00805f9b34fb
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:media_endpoint_destroy() sender=:1.55 path=/MediaEndpoint/A2DPSink/aptx
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:app_register_endpoint() Unable to register endpoint :1.55:/MediaEndpoint/A2DPSink/aptx: Success
dub 10 18:14:58 box bluetoothd[1337]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/aptx, iface: org.bluez.MediaEndpoint1
[... cut ...]
dub 10 18:16:27 box bluetoothd[1337]: profiles/audio/media.c:client_disconnect_cb() Client disconnected
dub 10 18:16:27 box bluetoothd[1337]: src/gatt-database.c:client_disconnect_cb() Client disconnected
dub 10 18:16:27 box bluetoothd[1337]: src/gatt-database.c:profile_remove() Removed ":1.56/midi/profile/03b80e5a-ede8-4b33-a751-6ce34ec4c700"
dub 10 18:16:27 box bluetoothd[1337]: src/gatt-database.c:profile_release() Releasing ":1.56"
```

[wireplumber.log](https://github.com/bluez/bluez/files/14934890/wireplumber.log)
[btsnoop.log](https://github.com/bluez/bluez/files/14934893/btsnoop.log)

Vudentz, 2024-04-10:
Seems to be something with the matching logic:

Set1:

```
dub 10 18:16:00 box bluetoothd[1337]: profiles/audio/bap.c:bap_ready() bap 0x60c3dcd15640
dub 10 18:16:00 box bluetoothd[1337]: profiles/audio/bap.c:pac_register() lpac 0x60c3dcd05260 rpac 0x60c3dcd1b370
dub 10 18:16:00 box bluetoothd[1337]: profiles/audio/bap.c:ep_register() ep 0x60c3dcd1a8d0 lpac 0x60c3dcd05260 rpac 0x60c3dcd1b370 path /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_source0
dub 10 18:16:00 box bluetoothd[1337]: profiles/audio/bap.c:pac_register() lpac 0x60c3dcd04b30 rpac 0x60c3dcd1f5a0
dub 10 18:16:00 box bluetoothd[1337]: profiles/audio/bap.c:ep_register() ep 0x60c3dccd2b40 lpac 0x60c3dcd04b30 rpac 0x60c3dcd1f5a0 path /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_sink
```

Set2:

```
dub 10 18:16:08 box bluetoothd[1337]: profiles/audio/bap.c:bap_ready() bap 0x60c3dcd36850
dub 10 18:16:08 box bluetoothd[1337]: profiles/audio/bap.c:pac_register() lpac 0x60c3dcd05260 rpac 0x60c3dcd216c0
dub 10 18:16:08 box bluetoothd[1337]: profiles/audio/bap.c:ep_register() ep 0x60c3dcd382e0 lpac 0x60c3dcd05260 rpac 0x60c3dcd216c0 path /org/bluez/hci0/dev_AC_80_0A_29_C1_C4/pac_source0
dub 10 18:16:08 box bluetoothd[1337]: profiles/audio/bap.c:pac_register() lpac 0x60c3dcd04b30 rpac 0x60c3dcd4cdc0
dub 10 18:16:08 box bluetoothd[1337]: profiles/audio/bap.c:ep_register() ep 0x60c3dcd37e90 lpac 0x60c3dcd04b30 rpac 0x60c3dcd4cdc0 path /org/bluez/hci0/dev_AC_80_0A_29_C1_C4/pac_sink
```

Perhaps it could be something with capabilities, since it seems these earbuds has only 32Khz and 16Khz enabled, are you using pipewire as endpoint? Have you tried with bluetoothctl?

drujd, 2024-04-10:
Are you sure about the sampling frequency? `Sampling Frequency: 48 Khz (0x0080)`
If yes, there is certainly wrong, they definitely run at 48 when paired to Android. I would hear the loss of treble :)

In pipewire, I see it as inactive device (and it is only one channel, the two buds don't match into set).

What exactly am I supposed to try with bluetoothctl? pairing? Connecting?

The logs are from pairing with powerdevil (+plasma-pa+pipewire+wireplumber).

pv, 2024-04-10:
Looks like bt_bap_select() thinks there is no suitable channel mapping, probably logic problem there. You can try this: [0001-shared-bap-fix-channel-allocation-in-bt_bap_select.patch.txt](https://github.com/bluez/bluez/files/14935948/0001-shared-bap-fix-channel-allocation-in-bt_bap_select.patch.txt) 

This is not a full proper fix though as it'll only allocate one channel per device, to find out how many streams it should create, it probably should count free ASEs.

Vudentz, 2024-04-10:
```
dub 10 18:16:07 box bluetoothd[1337]: src/shared/bap.c:bap_parse_pacs() PAC(s) 4
dub 10 18:16:07 box bluetoothd[1337]: #0: len 0x03 type 0x01
dub 10 18:16:07 box bluetoothd[1337]: Sampling Frequencies: 0x0004
dub 10 18:16:07 box bluetoothd[1337]: Sampling Frequency: 16 Khz (0x0004)
dub 10 18:16:07 box bluetoothd[1337]: #1: len 0x02 type 0x02
dub 10 18:16:07 box bluetoothd[1337]: Frame Duration: 0x01
dub 10 18:16:07 box bluetoothd[1337]: Frame Duration: 7.5 ms (0x01)
dub 10 18:16:07 box bluetoothd[1337]: #2: len 0x02 type 0x03
dub 10 18:16:07 box bluetoothd[1337]: Audio Channel Count: 0x01
dub 10 18:16:07 box bluetoothd[1337]: Audio Channel Count: 1 channel (0x01)
dub 10 18:16:07 box bluetoothd[1337]: #3: len 0x05 type 0x04
dub 10 18:16:07 box bluetoothd[1337]: Frame Length: 30 (0x001e) - 30 (0x001e)
dub 10 18:16:07 box bluetoothd[1337]: #4: len 0x02 type 0x05
dub 10 18:16:07 box bluetoothd[1337]: Max SDU: 1 (0x01)
dub 10 18:16:07 box bluetoothd[1337]: src/shared/bap.c:bap_parse_pacs() PAC #0: type 2 codec 0x06 cc_len 19 meta_len 4
dub 10 18:16:07 box bluetoothd[1337]: #0: len 0x03 type 0x01
dub 10 18:16:07 box bluetoothd[1337]: Sampling Frequencies: 0x0020
dub 10 18:16:07 box bluetoothd[1337]: Sampling Frequency: 32 Khz (0x0020)
dub 10 18:16:07 box bluetoothd[1337]: #1: len 0x02 type 0x02
dub 10 18:16:07 box bluetoothd[1337]: Frame Duration: 0x01
dub 10 18:16:07 box bluetoothd[1337]: Frame Duration: 7.5 ms (0x01)
dub 10 18:16:07 box bluetoothd[1337]: #2: len 0x02 type 0x03
dub 10 18:16:07 box bluetoothd[1337]: Audio Channel Count: 0x01
dub 10 18:16:07 box bluetoothd[1337]: Audio Channel Count: 1 channel (0x01)
dub 10 18:16:07 box bluetoothd[1337]: #3: len 0x05 type 0x04
dub 10 18:16:07 box bluetoothd[1337]: Frame Length: 60 (0x003c) - 60 (0x003c)
dub 10 18:16:07 box bluetoothd[1337]: #4: len 0x02 type 0x05
dub 10 18:16:07 box bluetoothd[1337]: Max SDU: 1 (0x01)
dub 10 18:16:07 box bluetoothd[1337]: src/shared/bap.c:bap_parse_pacs() PAC #1: type 2 codec 0x06 cc_len 19 meta_len 4
dub 10 18:16:07 box bluetoothd[1337]: #0: len 0x03 type 0x01
dub 10 18:16:07 box bluetoothd[1337]: Sampling Frequencies: 0x0004
dub 10 18:16:07 box bluetoothd[1337]: Sampling Frequency: 16 Khz (0x0004)
dub 10 18:16:07 box bluetoothd[1337]: #1: len 0x02 type 0x02
dub 10 18:16:07 box bluetoothd[1337]: Frame Duration: 0x02
dub 10 18:16:07 box bluetoothd[1337]: Frame Duration: 10 ms (0x02)
dub 10 18:16:07 box bluetoothd[1337]: #2: len 0x02 type 0x03
dub 10 18:16:07 box bluetoothd[1337]: Audio Channel Count: 0x01
dub 10 18:16:07 box bluetoothd[1337]: Audio Channel Count: 1 channel (0x01)
dub 10 18:16:07 box bluetoothd[1337]: #3: len 0x05 type 0x04
dub 10 18:16:07 box bluetoothd[1337]: Frame Length: 40 (0x0028) - 40 (0x0028)
dub 10 18:16:07 box bluetoothd[1337]: #4: len 0x02 type 0x05
dub 10 18:16:07 box bluetoothd[1337]: Max SDU: 1 (0x01)
dub 10 18:16:07 box bluetoothd[1337]: src/shared/bap.c:bap_parse_pacs() PAC #2: type 2 codec 0x06 cc_len 19 meta_len 4
dub 10 18:16:07 box bluetoothd[1337]: #0: len 0x03 type 0x01
dub 10 18:16:07 box bluetoothd[1337]: Sampling Frequencies: 0x0020
dub 10 18:16:07 box bluetoothd[1337]: Sampling Frequency: 32 Khz (0x0020)
dub 10 18:16:07 box bluetoothd[1337]: #1: len 0x02 type 0x02
dub 10 18:16:07 box bluetoothd[1337]: Frame Duration: 0x02
dub 10 18:16:07 box bluetoothd[1337]: Frame Duration: 10 ms (0x02)
dub 10 18:16:07 box bluetoothd[1337]: #2: len 0x02 type 0x03
dub 10 18:16:07 box bluetoothd[1337]: Audio Channel Count: 0x01
dub 10 18:16:07 box bluetoothd[1337]: Audio Channel Count: 1 channel (0x01)
dub 10 18:16:07 box bluetoothd[1337]: #3: len 0x05 type 0x04
dub 10 18:16:07 box bluetoothd[1337]: Frame Length: 80 (0x0050) - 80 (0x0050)
dub 10 18:16:07 box bluetoothd[1337]: #4: len 0x02 type 0x05
dub 10 18:16:07 box bluetoothd[1337]: Max SDU: 1 (0x01)
```

32Khz - 7.5-10ms and 16Khz - 7.5-10ms, which is sort of WTF is going on, why they are doing 4 records like that, makes no sense to me.

pv, 2024-04-10:
Looks like they didn't want to use the bitmasks in the PACs for whatever reason... AFAIK per spec it is equivalent though, so the channel allocation shouldn't get confused.

Vudentz, 2024-04-10:
> Looks like bt_bap_select() thinks there is no suitable channel mapping, probably logic problem there. You can try this patch for BlueZ: [0001-shared-bap-fix-channel-allocation-in-bt_bap_select.patch.txt](https://github.com/bluez/bluez/files/14935948/0001-shared-bap-fix-channel-allocation-in-bt_bap_select.patch.txt)
> 
> This is not a full proper fix though as it'll only allocate one channel per device, to find out how many streams it should create, it probably should count free ASEs. Another point the BAP spec is not so clear about.
> 
> Galaxy Buds2 Pro has actually two locations (FL,SL & FR,SR) and two ASEs on each of the earbuds, so it looks as if it's intended to allow two output streams per earbud + one input. Looks like this 4 connections is too much for AX210 though it fails with Connection Limit Exceeded in LE Set CIG Parameters. When connecting just one earbud, the FL,SL location streams get established fine with no error, but only FL results to audio output on the device the other apparently is ignored. So now I'm not quite sure how this is supposed to work via the information available in BAP, and how.

Yeah I don't think that is going to fly if you are trying to setup anything other than stereo, does Pipewire even sets more than FL, FR locations?

This is the important part that perhaps is not working:

```
			/* Check if location was set otherwise attempt to
			 * assign one based on the number of channels it
			 * supports.
			 */
			if (!rc->location) {
				rc->location = bt_bap_pac_get_locations(rpac);
				/* If channel count is 1 use a single
				 * location
				 */
				if (rc->count == 0x01)
					rc->location &= BIT(i);
			}

			/* Try matching the channel location */
			if (!(map.location & rc->location))
				break;

			lpac->ops->select(lpac, rpac, map.location &
						rc->location, &rpac->qos,
						func, user_data,
						lpac->user_data);
```

So we only select if there is a valid location, so we need to check what is the locations supported here, perhaps they do Center Left, Center Right, which is something we don't mark as supported in the local endpoints.

Vudentz, 2024-04-10:
```
> ACL Data RX: Handle 2049 flags 0x02 dlen 9                                                                             #569 [hci0] 41.433311
      ATT: Read Response (0x0b) len 4
        Value[4]: 02000000
        Handle: 0x1205 Type: Sink Audio Locations (0x2bca)
        Value[4]: 02000000
           Location: 0x00000002
              Front Right (0x00000002)
> ACL Data RX: Handle 2049 flags 0x02 dlen 9                                                                             #581 [hci0] 41.656645
      ATT: Read Response (0x0b) len 4
        Value[4]: 02000000
        Handle: 0x120b Type: Source Audio Locations (0x2bcc)
        Value[4]: 02000000
           Location: 0x00000002
              Front Right (0x00000002)
```

I don't see this information for the left side though.

pv, 2024-04-10:
The `rc->location &= BIT(i);` is not right at least, there's no relation between PAC ordering and the location bitmask. This also looks wrong:
```
			/* Check if there are any channels left to select */
			map.count &= ~(map.count & rc->count);
			if (!map.count)
				break;
```
The PAC channel bitmasks indicate how many channels each PAC configuration can do, I don't see something in 

> Yeah I don't think that is going to fly if you are trying to setup anything other than stereo

I think logic in the patch is OK, the problem is that in practice devices don't support much else than the fixed BAP configurations. Maybe what bt_bap_select() should do is to loop through the configurations in the BAP spec in some priority order, and try to allocate the channels for them.

> does Pipewire even sets more than FL, FR locations?

It supports any number of streams, in any locations whatever BlueZ gives in ChannelAllocation.

Vudentz, 2024-04-10:
> The `rc->location &= BIT(i);` is not right at least, there's no relation between PAC ordering and the location bitmask. This also looks wrong:
> 
> ```
> 			/* Check if there are any channels left to select */
> 			map.count &= ~(map.count & rc->count);
> 			if (!map.count)
> 				break;
> ```
> 
> The PAC channel bitmasks indicate how many channels each PAC configuration can do, and I don't see in the spec that they would indicate how many channels the device supports, but only how many you can put on a single ASE. Cf. BAP v1.0.1 §3.5.3. (Also, e.g. rc->count = 0x3 and lc1->count=0x1, lc2->count=0x2 would here try to do select() for both lc, even though the rc bitmask indicates the rpac supports either 1 or 2 channels.)

The general idea is that we stop allocating channels if we run out of locations, I understand you want to keep allocating until most locations are assigned while I think that would generally create problems at latter stage thus the I think going with the least amount channels is a better idea.

So if channel = 1 and there are 2 ASEs we would allocate just 1 ASE, not 2 since that requires a bigger topology, otherwise we would probably need to failover and retry if the controller don't have enough resources, but I would only consider doing that once we have cache support for the configurations so only do the failover and retry logic once per device.

> I think logic in the patch is OK, the problem is that in practice devices don't support much else than the fixed BAP configurations. Maybe what bt_bap_select() should do is to loop through the [...]

Will not going to fly either, the problem here is that the number of CIS(s) is limited at the controller level and there is no way to know that ahead of time.

> > does Pipewire even sets more than FL, FR locations?
> 
> It supports any number of streams, in any locations whatever BlueZ gives in ChannelAllocation. As the supported locations, Pipewire indicates all bits are OK.

Hmm, Id say that is probably not a good idea when we are the server, since it very likely would run into the problem that the controller can't set setup that many CIS(s) if the central attempts something like a 6(i) where each location requires a different CIS rather than multiplexing, for now I would recommend just keep it simple with FL/FR and we should probably avoid 6(i) as much as possible given there is no command to check how many CISes the controller can maintain.

Vudentz, 2024-04-10:
@drujd do you have a step by step how I flash the LE Audio firmware from sony, also if you could please attach it here I would try to make it work later this week.

drujd, 2024-04-10:
Just to be clear if you missed my previous comment regarding firmware: There is no need to install a modified apk anymore, with the latest official firmware LE Audio can be enabled on any Android phone that supports it (but you **do** need to enable it, the buds/firmware basically have 2 modes (there are no distinct 2 firmwares): A2DP only (the default, allows you to enable multipoint), and A2DP + BLEA (but no multipoint).

I installed the latest firmware through the official Android Sony Headphones app: https://play.google.com/store/apps/details?id=com.sony.songpal.mdr

I don't think there is any other official way: https://www.sony.com/electronics/support/wireless-headphones-bluetooth-headphones/wf-1000xm5/software/00296561 (there might be unofficial routes, but I have never tried that: https://github.com/lzghzr/MDR_Proxy )

Once you pair/connect the headphones, it starts updating the firmware automatically in background (you can check in System/Update Software cogwheel/Update Status). After updating to the latest firmware (which is a requirement to enable LE Audio functionality on anything that is not an Xperia phone), enable BLEA in System/LE Audio connection setting for headphones/LE Audio Priority. That's it, the should buds now support BLEA - you can then even uninstall the app.

Vudentz, 2024-04-10:
> Just to be clear if you missed my previous comment regarding firmware: There is no need to install a modified apk anymore, with the latest official firmware LE Audio can be enabled on any Android/iOS device that supports it (but you **do** need to enable it). The buds/firmware basically have 2 modes (there are no distinct 2 firmwares): A2DP only (the default, allows you to enable multipoint), and A2DP + BLEA (but no multipoint).
> 
> I installed the latest firmware through the official Android Sony Headphones app: [play.google.com/store/apps/details?id=com.sony.songpal.mdr](https://play.google.com/store/apps/details?id=com.sony.songpal.mdr)
> 
> I don't think there is any other official way: [sony.com/electronics/support/wireless-headphones-bluetooth-headphones/wf-1000xm5/software/00296561](https://www.sony.com/electronics/support/wireless-headphones-bluetooth-headphones/wf-1000xm5/software/00296561) (there might be unofficial routes, but I have never tried that: [lzghzr/MDR_Proxy](https://github.com/lzghzr/MDR_Proxy) )
> 
> Once you pair/connect the headphones, it starts updating the firmware automatically in background (you can check in System/Update Software cogwheel/Update Status). After updating to the latest firmware (which is a requirement to enable LE Audio functionality on anything that is not an Xperia phone), enable BLEA in System/LE Audio connection setting for headphones/LE Audio Priority. That's it, the should buds now support BLEA - you can then even uninstall the app.

Ah now I realized why it doesn't show it for me, I have the WF-1000XM4 which I assume don't have LE Audio support yet, at least not officially.

drujd, 2024-04-10:
> I don't see this information for the left side though. @drujd did you capture the trace for both? It seems there is only one side on the btsnoop trace, also when you run btmon you can pass -p debug that shall capture the bluetoothd traces and put them together.

Well, it seems that I actually **cannot** create a trace that contains both 'Front Left' and 'Front Right'. Even though I _do_ connect to both through the CSIP powerdevil prompt (Trust & Authorize / Authorize). Only the one I select when pairing / connect to first is shown as Front Left/Right...

But at least here is the trace (with -p7) for the left one:
[btsnoop_left.log](https://github.com/bluez/bluez/files/14937710/btsnoop_left.log)

drujd, 2024-04-10:
> Ah now I realized why it doesn't show it for me, I have the WF-1000XM4 which I assume don't have LE Audio support yet, at least not officially.

I don't think it has any support at all, older hardware/chipset. That one is A2DP only I'm afraid...

drujd, 2024-04-10:
> Looks like bt_bap_select() thinks there is no suitable channel mapping, probably logic problem there. You can try this patch for BlueZ

See the (edited/updated) comment above, even with the patch, it seems to behave identically (at least from my point of view, maybe there is some change in the new btsnoop?)

pv, 2024-04-10:
@drujd: ok, too bad. Breakpoint at bt_bap_select() and stepping through in gdb should reveal what's going if you know how to do that.

> The general idea is that we stop allocating channels if we run out of locations, I understand you want to keep allocating until most locations are assigned while I think that would generally create problems at latter stage thus the I think going with the least amount channels is a better idea.

Agreed on that we can't set up arbitrary number of channels in practice. The use of PAC Supported_Audio_Channel_Counts still looks to me suspect though, as iiuc in spec it means something different.

BTW, I was looking at the Earfun Air Pro 3 device that needs Configuration 6(i) and it looks with this device BlueZ reads PACS Locations after the PACs, so they get zero rc->location, which triggers the 6(i) config; Galaxy Buds2 Pro on the other hand has the opposite read ordering so it gets Configuration 1. Earfun & one earbud of Gbuds2 have both 2 sink ASE, 2 sink locations, no ASE multiplexing so I was wondering how it picks different config for them...

> Hmm, Id say that is probably not a good idea when we are the server

Ok, yes, that's true. I guess this means it could be nice to have e.g. "Role" & "MaxStreams" fields in local PACs, so client wouldn't need to decide ahead of time which specific channels it's going to support, but on.

drujd, 2024-04-10:
> @drujd: ok, too bad. Breakpoint at bt_bap_select() and stepping through in gdb should reveal what's going if you know how to do that.

OK, I have recompiled bluez with -O0 and attached gdb to it, but what am I looking for, exactly? I know C, but bluez not at all. :-)

pv, 2024-04-11:
@drujd The question is why it doesn't call ops->select(), as it seems to be from logs? Bits set in lchan/rchan->count = Supported_Audio_Channel_Counts indicate how many channels can be multiplexed in a single stream, with BIT(0) = 1 channel, BIT(1) = 2 channels, BIT(0)|BIT(1) = 1 or 2 etc. according to the rpac (= remote audio capability information). Bits set in bt_bap_pac_get_locations(rpac) is a bitmask whose set bits indicate what channels the remote device has (e.g. FL = BIT(0), FR = BIT(1), ...). See [Assigned Numbers](https://www.bluetooth.com/specifications/assigned-numbers/). Each call to select() starts creating one stream, with the channels indicated by the bitmask in the second argument. Second argument is zero mean mono channel. In your single TWS earbud case there probably should be exactly one call to select().

drujd, 2024-04-11:
> @drujd The question is why it doesn't call ops->select(), as it seems to be from logs?
It does, though (`pac_select()`).

> In your single TWS earbud case there probably should be exactly one call to select() with one channel, which should lead to a DBus call to Pipewire visible in bluetoothd & wireplumber logs.
On connecting the buds, `pac_select()` is called two times - once in each `bt_bap_select()` call - which I assume is correct.
Going step-by-step in `pac_select()`, it does seem to call all the `dbus_*` functions just fine, enters the `if (qos && qos->phy)` branch and in the end calls `media_endpoint_async_call()` and seems to go through it just fine...

> Anyway, Sony Linkbuds S that I have also seems to have got a firmware update enabling some LE Audio features, so I can take a look with that device in case it is similar.
Yes! Those are basically a cheaper version of XM5s, they should have the same/similar chipset.

drujd, 2024-04-11:
Now thinking of it, all the btsnoops before were from the pairing process, whereas what I'm doing now is just connecting already paired devices.

[btsnoop_connect.log](https://github.com/bluez/bluez/files/14944345/btsnoop_connect.log)

drujd, 2024-04-11:
`dbus-monitor --system | grep -i blue -C50`:
[dbus-filtered.log](https://github.com/bluez/bluez/files/14944475/dbus-filtered.log)

drujd, 2024-04-11:
Here's a new wireplumber log with the buds already paired:
[wireplumber_connect.log](https://github.com/bluez/bluez/files/14946920/wireplumber_connect.log)

This ios the odler one from a comment above, there I actually paired the buds:
[wireplumber.log](https://github.com/bluez/bluez/files/14934890/wireplumber.log)

drujd, 2024-04-11:
Still the same behavior. I have tried https://github.com/tedd-an/bluez-upstream-test/commit/4c4b8879075ee7a2d6c6baa574a483ff9d8079fe alone and then that combined with https://github.com/tedd-an/bluez-upstream-test/commit/4f7a55c200894fd9913cbf7ef5084ea49797b1ca (in both cases cherrypicked on top of bluez master).
[btsnoop_connect_4c4b8879_6007b8df.log](https://github.com/bluez/bluez/files/14947785/btsnoop_connect_4c4b8879_6007b8df.log)
[wireplumber_connect_4c4b8879_6007b8df.log](https://github.com/bluez/bluez/files/14947786/wireplumber_connect_4c4b8879_6007b8df.log)

Vudentz, 2024-04-11:
> Still the same behavior. I have tried [tedd-an/bluez-upstream-test@4c4b887](https://github.com/tedd-an/bluez-upstream-test/commit/4c4b8879075ee7a2d6c6baa574a483ff9d8079fe) alone and then that combined with [tedd-an/bluez-upstream-test@4f7a55c](https://github.com/tedd-an/bluez-upstream-test/commit/4f7a55c200894fd9913cbf7ef5084ea49797b1ca) (in both cases cherrypicked on top of bluez master). [btsnoop_connect_4c4b8879_6007b8df.log](https://github.com/bluez/bluez/files/14947785/btsnoop_connect_4c4b8879_6007b8df.log) [wireplumber_connect_4c4b8879_6007b8df.log](https://github.com/bluez/bluez/files/14947786/wireplumber_connect_4c4b8879_6007b8df.log)

You need to enable debug at bluetoothd (-d) in order for it to appear at btsnoop logs, that said it really weird that you got no change, perhaps it is not a problem with bt_bap_select after all.

drujd, 2024-04-11:
i am running it with -d: `9445 ?        ts     0:00 /usr/lib/bluetooth/bluetoothd -d *`

See the comments above with @pv - the dbus message IS called, it seems. So maybe check wireplumber logs?

Debugging it, it indeed falls into the `if (!selected)` branch and then `pac_select()`, skips `if (loc)` and `if (location)`, enters `if (metadata)` and then out of 4 calls, it falls into `if (qos && qos->phy)` twice.

Vudentz, 2024-04-11:
> i am running it with -d: `9445 ? ts 0:00 /usr/lib/bluetooth/bluetoothd -d *` Looking into the btsnoop file, I'm pretty sure it contains the debug information, though. But I can provide separate bluetoothd logs, if needed (but again, are you sure you are looking at the correct file?)

Hmm, are you sure it is the right binary though? Try running it directly from source with -dn, it will show if it could not be run, etc, and all the messages would be printed into the terminal so it is much easier to forward here instead of using sound like journalctl, etc.

> See the comments above with @pv - the dbus message IS called, it seems. So maybe check wireplumber logs?
>
> Debugging it, it indeed falls into the `if (!selected)` branch and then `pac_select()`, [...]

Oh, so it is called, then perhaps SelectProperties does result in an error?

drujd, 2024-04-11:
Given that I am debugging it with gdb at the same time and it matches with the source I see perfectly, I'm pretty sure that yes, I am attaching to the right binary. But again, what exactly are you missing in the btsnoop file? A quick peek at journalctl -u bluetooth and the btsnoop file suggests that the file indeed contains what I see in the journal.

[image]

Vudentz, 2024-04-11:
@drujd my bad, while running btmon -r I also need to pass -p debug:

```
= bluetoothd: profiles/audio/bap.c:setup_new() ep 0x5925d81f7250 setup 0x5925d82687b0                                                                                                                                                                                               17.989739
= bluetoothd: profiles/audio/bap.c:select_cb() selecting 1                                                                                                                                                                                                                          17.989777
= bluetoothd: profiles/audio/bap.c:setup_config() setup 0x5925d82687b0 caps 0x5925d8232650 metadata 0x5925d8236f30                                                                                                                                                                  17.989796
= bluetoothd: #0: len 0x02 type 0x01                                                                                                                                                                                                                                                17.989812
= bluetoothd: Sampling Frequency: 16 Khz (0x03)                                                                                                                                                                                                                                     17.989825
= bluetoothd: #1: len 0x02 type 0x02                                                                                                                                                                                                                                                17.989837
= bluetoothd: Frame Duration: 7.5 ms (0x00)                                                                                                                                                                                                                                         17.989850
= bluetoothd: #2: len 0x03 type 0x04                                                                                                                                                                                                                                                17.989862
= bluetoothd: Frame Length: 30 (0x001e)                                                                                                                                                                                                                                             17.989874
= bluetoothd: #3: len 0x02 type 0x05                                                                                                                                                                                                                                                17.989886
= bluetoothd: Frame Blocks per SDU: 1 (0x01)                                                                                                                                                                                                                                        17.989898
= bluetoothd: src/shared/bap.c:bap_queue_req() req 0x5925d8230440 (op 0x01) queue 0x5925d82306a0                                                                                                                                                                                    17.989910
= bluetoothd: profiles/audio/bap.c:setup_new() ep 0x5925d8259fa0 setup 0x5925d8254960                                                                                                                                                                                               17.989931
= bluetoothd: profiles/audio/bap.c:select_cb() selecting 0                                                                                                                                                                                                                          17.989945
= bluetoothd: src/shared/bap.c:bap_process_queue()                                                                                                                                                                                                                                  18.000057
= bluetoothd: src/shared/bap.c:bap_send() req 0x5925d8230440 len 24
```

So looks like the fallback mechanism is working as intended, but we end up on configuring sources/input only:

`= bluetoothd: src/shared/bap.c:bt_bap_stream_io_get_qos() in 0x5925d8268870 out (nil)`

Perhaps this has to do with 16khz being selected, need to investigate why out QoS is not set.

pv, 2024-04-11:
@drujd From log it looks like it creates the sink. Try to see if you can see it in Helvum, it might be hidden from PA apps if the other device of the device set is not connected.

drujd, 2024-04-11:
[image]

drujd, 2024-04-11:
Just to be clear: I **do** see it in plasma-pa: as inactive device. I stated that in the comment yesterday. I was able to even attempt to run a test, I think the first logs on these issue contain that attempt.

Vudentz, 2024-04-11:
@drujd try with the v2: https://patchwork.kernel.org/project/bluetooth/list/?series=843708

drujd, 2024-04-11:
@Vudentz Adding that missing `+` certainly moves us in the right direction. However, playback is still not working and the channels are not set up correctly still (FL/mono).
[image]
[image]

The following logs are for connecting & test playback attempt:
[btsnoop_newpatch.log](https://github.com/bluez/bluez/files/14949146/btsnoop_newpatch.log)
[wireplumber_newpatch.log](https://github.com/bluez/bluez/files/14949147/wireplumber_newpatch.log)

Vudentz, 2024-04-11:
```
> ACL Data RX: Handle 2049 flags 0x02 dlen 15                                                                            #452 [hci0] 26.734909
      ATT: Handle Value Notification (0x1b) len 10
        Handle: 0x110e
          Data[8]: 0302030802010802
```

Which decodes to:

```
0x03 -> BT_ASCS_ENABLE
0x02 -> Number of ASE
0x03 -> ASE ID
0x08 -> BT_ASCS_RSP_CONF_REJECTED
0x02 -> BT_ASCS_REASON_CODEC_DATA
0x01 -> ASE ID
0x08 -> BT_ASCS_RSP_CONF_REJECTED
0x02 -> BT_ASCS_REASON_CODEC_DATA
```

So the device is not happy about the configuration, I see that in both direction btw, the actual configuration seems to be:

```
= bluetoothd: #0: len 0x02 type 0x01                                                                                                 13.594771
= bluetoothd: Sampling Frequency: 16 Khz (0x03)                                                                                      13.594788
= bluetoothd: #1: len 0x02 type 0x02                                                                                                 13.594805
= bluetoothd: Frame Duration: 7.5 ms (0x00)                                                                                          13.594822
= bluetoothd: #2: len 0x05 type 0x03                                                                                                 13.594839
= bluetoothd: Location: 0x00000001                                                                                                   13.594856
= bluetoothd: Location: Front Left (0x00000001)                                                                                      13.594872
= bluetoothd: #3: len 0x03 type 0x04                                                                                                 13.594900
= bluetoothd: Frame Length: 30 (0x001e)                                                                                              13.594938
= bluetoothd: #4: len 0x02 type 0x05                                                                                                 13.594955
= bluetoothd: Frame Blocks per SDU: 1 (0x01)                                                                                         13.594971
= bluetoothd: src/shared/bap.c:bap_queue_req() req 0x5c48acc4d7a0 (op 0x01) queue 0x5c48acc4cb30                                     13.594987
= bluetoothd: profiles/audio/bap.c:setup_config() setup 0x5c48acc1b8e0 caps 0x5c48acc186b0 metadata 0x5c48acc5ee70                   13.595004
= bluetoothd: #0: len 0x02 type 0x01                                                                                                 13.595020
= bluetoothd: Sampling Frequency: 48 Khz (0x08)                                                                                      13.595037
= bluetoothd: #1: len 0x02 type 0x02                                                                                                 13.595062
= bluetoothd: Frame Duration: 7.5 ms (0x00)                                                                                          13.595080
= bluetoothd: #2: len 0x05 type 0x03                                                                                                 13.595096
= bluetoothd: Location: 0x00000001                                                                                                   13.595113
= bluetoothd: Location: Front Left (0x00000001)                                                                                      13.595129
= bluetoothd: #3: len 0x03 type 0x04                                                                                                 13.595145
= bluetoothd: Frame Length: 75 (0x004b)                                                                                              13.595162
= bluetoothd: #4: len 0x02 type 0x05                                                                                                 13.595178
= bluetoothd: Frame Blocks per SDU: 1 (0x01)
```

So 16Khz/7.5ms input and 48Khz/7.5ms output, which in theory is valid, but perhaps it is picky about that being in the same CIS for example, anyway you can use bluetoothctl to confirm exactly what is the setup with transport.show <transport object>.

drujd, 2024-04-11:
```
list

Transport /org/bluez/hci0/dev_AC_80_0A_29_C1_C4/pac_source0/fd18 
Transport /org/bluez/hci0/dev_AC_80_0A_29_C1_C4/pac_sink0/fd19 
Transport /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_source0/fd20 
Transport /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_sink0/fd21 

show /org/bluez/hci0/dev_AC_80_0A_29_C1_C4/pac_source0/fd18c_source0/fd18

Transport /org/bluez/hci0/dev_AC_80_0A_29_C1_C4/pac_source0/fd18
        UUID: 00002bc9-0000-1000-8000-00805f9b34fb
        Codec: 0x06 (6)
        Configuration.#0: len 0x02 type 0x01
        Configuration.Sampling Frequency: 16 Khz (0x03)
        Configuration.#1: len 0x02 type 0x02
        Configuration.Frame Duration: 7.5 ms (0x00)
        Configuration.#2: len 0x03 type 0x04
        Configuration.Frame Length: 30 (0x001e)
        Configuration.#3: len 0x02 type 0x05
        Configuration.Frame Blocks per SDU: 1 (0x01)
        Device: /org/bluez/hci0/dev_AC_80_0A_29_C1_C4
        State: idle
        Endpoint: /org/bluez/hci0/dev_AC_80_0A_29_C1_C4/pac_source0
        QoS.CIG: 0x00 (0)
        QoS.CIS: 0x00 (0)
        QoS.Framing: 0x00 (0)
        QoS.PresentationDelay: 0x00009c40 (40000)
        QoS.Interval: 0x00001d4c (7500)
        QoS.Latency: 0x0008 (8)
        QoS.SDU: 0x001e (30)
        QoS.PHY: 0x02 (2)
        QoS.Retransmissions: 0x02 (2)
        Location: 0x00000000 (0)
        Links: /org/bluez/hci0/dev_AC_80_0A_29_C1_C4/pac_sink0/fd19

show /org/bluez/hci0/dev_AC_80_0A_29_C1_C4/pac_sink0/fd19pac_sink0/fd19

Transport /org/bluez/hci0/dev_AC_80_0A_29_C1_C4/pac_sink0/fd19
        UUID: 00002bcb-0000-1000-8000-00805f9b34fb
        Codec: 0x06 (6)
        Configuration.#0: len 0x02 type 0x01
        Configuration.Sampling Frequency: 48 Khz (0x08)
        Configuration.#1: len 0x02 type 0x02
        Configuration.Frame Duration: 7.5 ms (0x00)
        Configuration.#2: len 0x03 type 0x04
        Configuration.Frame Length: 75 (0x004b)
        Configuration.#3: len 0x02 type 0x05
        Configuration.Frame Blocks per SDU: 1 (0x01)
        Device: /org/bluez/hci0/dev_AC_80_0A_29_C1_C4
        State: idle
        Endpoint: /org/bluez/hci0/dev_AC_80_0A_29_C1_C4/pac_sink0
        QoS.CIG: 0x00 (0)
        QoS.CIS: 0x00 (0)
        QoS.Framing: 0x00 (0)
        QoS.PresentationDelay: 0x00009c40 (40000)
        QoS.Interval: 0x00001d4c (7500)
        QoS.Latency: 0x000f (15)
        QoS.SDU: 0x004b (75)
        QoS.PHY: 0x02 (2)
        QoS.Retransmissions: 0x05 (5)
        Location: 0x00000000 (0)
        Links: /org/bluez/hci0/dev_AC_80_0A_29_C1_C4/pac_source0/fd18

show /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_source0/fd20c_source0/fd20

Transport /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_source0/fd20
        UUID: 00002bc9-0000-1000-8000-00805f9b34fb
        Codec: 0x06 (6)
        Configuration.#0: len 0x02 type 0x01
        Configuration.Sampling Frequency: 16 Khz (0x03)
        Configuration.#1: len 0x02 type 0x02
        Configuration.Frame Duration: 7.5 ms (0x00)
        Configuration.#2: len 0x05 type 0x03
        Configuration.Location: 0x00000001
        Configuration.Location: Front Left (0x00000001)
        Configuration.#3: len 0x03 type 0x04
        Configuration.Frame Length: 30 (0x001e)
        Configuration.#4: len 0x02 type 0x05
        Configuration.Frame Blocks per SDU: 1 (0x01)
        Device: /org/bluez/hci0/dev_C6_FC_CD_99_12_86
        State: idle
        Endpoint: /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_source0
        QoS.CIG: 0x00 (0)
        QoS.CIS: 0x01 (1)
        QoS.Framing: 0x00 (0)
        QoS.PresentationDelay: 0x00009c40 (40000)
        QoS.Interval: 0x00001d4c (7500)
        QoS.Latency: 0x0008 (8)
        QoS.SDU: 0x001e (30)
        QoS.PHY: 0x02 (2)
        QoS.Retransmissions: 0x02 (2)
        Location: 0x00000001 (1)
        Links: /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_sink0/fd21

show /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_sink0/fd21pac_sink0/fd21

Transport /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_sink0/fd21
        UUID: 00002bcb-0000-1000-8000-00805f9b34fb
        Codec: 0x06 (6)
        Configuration.#0: len 0x02 type 0x01
        Configuration.Sampling Frequency: 48 Khz (0x08)
        Configuration.#1: len 0x02 type 0x02
        Configuration.Frame Duration: 7.5 ms (0x00)
        Configuration.#2: len 0x05 type 0x03
        Configuration.Location: 0x00000001
        Configuration.Location: Front Left (0x00000001)
        Configuration.#3: len 0x03 type 0x04
        Configuration.Frame Length: 75 (0x004b)
        Configuration.#4: len 0x02 type 0x05
        Configuration.Frame Blocks per SDU: 1 (0x01)
        Device: /org/bluez/hci0/dev_C6_FC_CD_99_12_86
        State: idle
        Endpoint: /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_sink0
        QoS.CIG: 0x00 (0)
        QoS.CIS: 0x01 (1)
        QoS.Framing: 0x00 (0)
        QoS.PresentationDelay: 0x00009c40 (40000)
        QoS.Interval: 0x00001d4c (7500)
        QoS.Latency: 0x000f (15)
        QoS.SDU: 0x004b (75)
        QoS.PHY: 0x02 (2)
        QoS.Retransmissions: 0x05 (5)
        Location: 0x00000001 (1)
        Links: /org/bluez/hci0/dev_C6_FC_CD_99_12_86/pac_source0/fd20
```

pv, 2024-04-11:
In the above bluetoothd.log it looks like might be reading Sink PACs before Sink PAC Location, so bt_bap_chan have 0 location for one of the devices, which is maybe it ends up in the fallback. @drujd the following patch adds some more debug to bluez, so we'd see this in bluetoothd.log: [loc-dbg.patch.txt](https://github.com/bluez/bluez/files/14949593/loc-dbg.patch.txt)

Vudentz, 2024-04-11:
Hmm, we don't have the location set for Front Right for example, perhaps because we couldn't read it or something which would explain why we couldn't do the channel allocation selection so left as 0 but Pipewire couldn't do it either, anyway both Left and Right earbuds return the same exact error so I would expect only the right bud to do return an error if that was the case.

Anther thing that looks odd, although supported by the spec, if the QoS configuration is not the same both direction, but the error code is about the codec configuration alone so I don't think it really matters.

pv, 2024-04-11:
In the Pipewire logs there is
```
$ grep -E 'SelectProperties|ep qos: Locations|ClearProperties' wireplumber_newpatch.log 
D 18:55:36.230476         spa.bluez5 ../pipewire/spa/plugins/bluez5/bluez5-dbus.c:4540:endpoint_handler: dbus: path=/MediaEndpointLE/BAPSink/lc3, interface=org.bluez.MediaEndpoint1, member=SelectProperties
D 18:55:36.230541         spa.bluez5 ../pipewire/spa/plugins/bluez5/bluez5-dbus.c:853:parse_endpoint_props: ep qos: Locations=1
D 18:55:36.231059         spa.bluez5 ../pipewire/spa/plugins/bluez5/bluez5-dbus.c:4540:endpoint_handler: dbus: path=/MediaEndpointLE/BAPSink/lc3, interface=org.bluez.MediaEndpoint1, member=SelectProperties
D 18:55:57.666686         spa.bluez5 ../pipewire/spa/plugins/bluez5/bluez5-dbus.c:4540:endpoint_handler: dbus: path=/MediaEndpointLE/BAPSink/lc3, interface=org.bluez.MediaEndpoint1, member=SelectProperties
D 18:55:57.666752         spa.bluez5 ../pipewire/spa/plugins/bluez5/bluez5-dbus.c:853:parse_endpoint_props: ep qos: Locations=1
D 18:55:57.666979         spa.bluez5 ../pipewire/spa/plugins/bluez5/bluez5-dbus.c:4540:endpoint_handler: dbus: path=/MediaEndpointLE/BAPSource/lc3, interface=org.bluez.MediaEndpoint1, member=SelectProperties
D 18:55:57.667117         spa.bluez5 ../pipewire/spa/plugins/bluez5/bluez5-dbus.c:853:parse_endpoint_props: ep qos: Locations=1
D 18:55:57.756625         spa.bluez5 ../pipewire/spa/plugins/bluez5/bluez5-dbus.c:4540:endpoint_handler: dbus: path=/MediaEndpointLE/BAPSink/lc3, interface=org.bluez.MediaEndpoint1, member=SelectProperties
D 18:55:57.756901         spa.bluez5 ../pipewire/spa/plugins/bluez5/bluez5-dbus.c:4540:endpoint_handler: dbus: path=/MediaEndpointLE/BAPSource/lc3, interface=org.bluez.MediaEndpoint1, member=SelectProperties
```
So it seems we are getting Locations= for FL but not for FR.

drujd, 2024-04-11:
[bluetoothd_adddebug.log](https://github.com/bluez/bluez/files/14949766/bluetoothd_adddebug.log)
[btsnoop_adddebug.log](https://github.com/bluez/bluez/files/14949767/btsnoop_adddebug.log)
[wireplumber_adddebug.log](https://github.com/bluez/bluez/files/14949770/wireplumber_adddebug.log)

Is there any particular reason to use syslog() instead of DBG() in that first case? syslog() entries are not present in thee btsnoop file (otherwise I'd omit the bluetooth service journal)

Vudentz, 2024-04-11:
Yeah, I think maybe it wasn't initialized during the first discovery for some reason, so Im planning to add the following change:

```
commit c66c2c2245cfed8b3b68de04331ee567d41ba34c (HEAD -> master)
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Thu Apr 11 14:05:47 2024 -0400

    shared/bap: Fix not resuming reading source/sink locations
    
    If there is and unexpected disconnect and the source/sink locations
    where left uninitialized this attempts to resume reading them once a
    new session is attached.

diff --git a/src/shared/bap.c b/src/shared/bap.c
index cc1fa1ffbe32..79d9345d9161 100644
--- a/src/shared/bap.c
+++ b/src/shared/bap.c
@@ -4703,6 +4703,18 @@ clone:
 			}
 		}
 
+		/* Resume reading sink locations if supported */
+		if (pacs->sink && pacs->sink_loc && !pacs->sink_loc_value) {
+			if (gatt_db_attribute_get_char_data(pacs->sink_loc,
+							NULL, &value_handle,
+							NULL, NULL, NULL)) {
+				bt_gatt_client_read_value(bap->client,
+							value_handle,
+							read_sink_pac_loc,
+							bap, NULL);
+			}
+		}
+
 		/* Resume reading sources if supported */
 		if (pacs->source && queue_isempty(bap->rdb->sources)) {
 			if (gatt_db_attribute_get_char_data(pacs->source,
@@ -4715,6 +4727,19 @@ clone:
 			}
 		}
 
+		/* Resume reading source locations if supported */
+		if (pacs->source && pacs->source_loc &&
+				!pacs->source_loc_value) {
+			if (gatt_db_attribute_get_char_data(pacs->source_loc,
+							NULL, &value_handle,
+							NULL, NULL, NULL)) {
+				bt_gatt_client_read_value(bap->client,
+							value_handle,
+							read_source_pac_loc,
+							bap, NULL);
+			}
+		}
+
 		queue_foreach(bap->remote_eps, bap_endpoint_foreach, bap);
 
 		bap_cp_attach(bap);
```

drujd, 2024-04-11:
Well, I now always get FL/FR (after a few reconnects/re-pairs). Although I was not able to replicate the FL/mono situation after I reconnected before even without the patch.

pv, 2024-04-11:
Does it still give `update pacs 0x60c000043600 pac 0x608000018a20 location 0` in logs though? I seem to get those on Linkbuds S (which indeed now does have LE Audio).

drujd, 2024-04-11:
I get a bunch of `update pacs`, but only on bluetoothd start and pairing, not on connection/playback, and no `location 1`

On start
```
dub 11 20:32:12 box bluetoothd[3854]: update pacs 0x569f8d705e70 pac 0x569f8d758680 location fffffff
dub 11 20:32:12 box bluetoothd[3854]: update pacs 0x569f8d705e70 pac 0x569f8d757f50 location fffffff
```

Pairing the 1st one:
```
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d773790 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d773790 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d773790 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d773790 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d773790 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d773790 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d773790 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d773790 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d773790 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d773790 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d773790 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d773790 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d770ae0 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d770ae0 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d770ae0 location 0
dub 11 20:33:23 box bluetoothd[3854]: update pacs 0x569f8d765a40 pac 0x569f8d770ae0 location 0
```
...and the 2nd one:
```
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d7a0210 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d7a0210 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d7a0210 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d7a0210 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d7a0210 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d7a0210 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d7a0210 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d7a0210 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d7a0210 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d7a0210 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d7a0210 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d7a0210 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d781530 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d781530 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d781530 location 0
dub 11 20:33:29 box bluetoothd[3854]: update pacs 0x569f8d7898d0 pac 0x569f8d781530 location 0
```

pv, 2024-04-11:
Ok, so it's still reading PACs before PACS Locations so bt_bap_chan.location is not properly initialized. 

For the device rejecting configuration issue, at least Linkbuds S works in sink-only config, make file `~/.config/wireplumber/wireplumber.conf.d/bap-sink.conf` with content `monitor.bluez.properties = { bluez5.roles = [ bap_source ] }`  + restart WP.

drujd, 2024-04-11:
Indeed it works! This is the very first time I'm getting BLEA/LC3 out of my PC!
Interestingly, from time to time, the buds become **very** unresponsive. Switching between ANC off/on/ambient takes 10-20 s. Then they work perfectly fine and then problem appears again.
But it works!

drujd, 2024-04-11:
Here's log from 1s of successful playback - then it stopped working:
[btsnoop_playback.log](https://github.com/bluez/bluez/files/14950751/btsnoop_playback.log)

[17 more replies]

closed 2024-04-12 by BluezTestBot