bluez/bluez #799: "BT socket not connected" journal spam

chriscjs, 2024-03-31:
The full error is:

bluetoothd[1463]: profiles/input/device.c:hidp_send_message() BT socket not connected

I think this started after update from 5.72 to 5.73 and it happens even when no bluetooth devices are connected.

I'm running Arch Linux with kernel 6.6.23 LTS.

YoinkerBoinker, 2024-04-01:
Can verify that as well. Seems to happen here after a device has been connected, then disconnected and disable bluetooth (via KDE). Then i'll get the same message every 30 seconds.

wescode, 2024-04-06:
Same issue occurs on 6.8.4 fedora 39 bluez 5.73. I am happy to provide any further info needed.

ryanmusante, 2024-04-08:
```
~ > bluetoothctl --version                                                                                                                       8s
bluetoothctl: 5.73
```
` bluetoothd[562]: profiles/input/device.c:hidp_send_message() BT socket not connected`

olegon, 2024-10-19:
I'm receiving this error from times to times:

```
Oct 19 09:37:31 olegonsys bluetoothd[36384]: profiles/input/device.c:hidp_send_message() BT socket not connected
Oct 19 09:37:31 olegonsys kernel: bluetoothd[36384]: segfault at 625032866133 ip 000062560e93444c sp 00007ffdc301d9b0 error 4 in bluetoothd[62560e882000+ea000] likely on CPU 3 (core 3, socke
t 0)
Oct 19 09:37:31 olegonsys systemd-coredump[44582]: Process 36384 (bluetoothd) of user 0 terminated abnormally with signal 11/SEGV, processing...
Oct 19 09:37:31 olegonsys systemd[1]: Started Process Core Dump (PID 44582/UID 0).
Oct 19 09:37:31 olegonsys systemd-coredump[44585]: Process 36384 (bluetoothd) of user 0 dumped core.
                                                   
                                                   Stack trace of thread 36384:
                                                   #0  0x000062560e93444c n/a (bluetoothd + 0xba44c)
                                                   #1  0x000062560e955bf9 n/a (bluetoothd + 0xdbbf9)
                                                   #2  0x00007aaa1aa89559 n/a (libglib-2.0.so.0 + 0x5d559)
                                                   #3  0x00007aaa1aaec157 n/a (libglib-2.0.so.0 + 0xc0157)
                                                   #4  0x00007aaa1aa8a287 g_main_loop_run (libglib-2.0.so.0 + 0x5e287)
                                                   #5  0x000062560e884005 n/a (bluetoothd + 0xa005)
                                                   #6  0x00007aaa1a6e5e08 n/a (libc.so.6 + 0x25e08)
                                                   #7  0x00007aaa1a6e5ecc __libc_start_main (libc.so.6 + 0x25ecc)
                                                   #8  0x000062560e885305 n/a (bluetoothd + 0xb305)
                                                   ELF object binary architecture: AMD x86-64
```

Version:

```bash
> bluetoothctl --version         
bluetoothctl: 5.78
```

LordChaos73, 2024-10-29:
Same here (CachyOS - Plasma 6.2.2). Happens every hour or so. I'm happy to provide more details if needed.

```
okt 29 16:19:18 yoga7pro bluetoothd[35025]: profiles/input/device.c:hidp_send_message() BT socket not connected
okt 29 16:19:18 yoga7pro kernel: bluetoothd[35025]: segfault at 618d2eabbfb0 ip 0000618b040df7c9 sp 00007ffca50c2a70 error 4 in bluetoothd[da7c9,618b0400d000+11b000] likely on CPU 12 (core 6>
okt 29 16:19:18 yoga7pro kernel: Code: 48 8b 43 08 48 85 c0 74 4b f0 ff 03 eb 06 0f 1f 00 48 89 e8 48 83 7b 08 00 74 2b 83 3b 01 7e 26 48 8b 68 08 48 8b 00 8b 0c 24 <39> 48 04 75 12 48 8b 50>
okt 29 16:19:18 yoga7pro systemd-coredump[45562]: Process 35025 (bluetoothd) of user 0 terminated abnormally with signal 11/SEGV, processing...
okt 29 16:19:18 yoga7pro systemd[1]: Started Process Core Dump (PID 45562/UID 0).
okt 29 16:19:18 yoga7pro systemd-coredump[45563]: [🡕] Process 35025 (bluetoothd) of user 0 dumped core.

                                                  Stack trace of thread 35025:
                                                  #0  0x0000618b040df7c9 n/a (bluetoothd + 0xda7c9)
                                                  #1  0x0000618b04113a99 n/a (bluetoothd + 0x10ea99)
                                                  #2  0x0000737675d738d7 n/a (libglib-2.0.so.0 + 0x628d7)
                                                  #3  0x0000737675de91d8 n/a (libglib-2.0.so.0 + 0xd81d8)
                                                  #4  0x0000737675d7b8af g_main_loop_run (libglib-2.0.so.0 + 0x6a8af)
                                                  #5  0x0000618b0400ec77 n/a (bluetoothd + 0x9c77)
                                                  #6  0x000073767599f18e n/a (libc.so.6 + 0x2618e)
                                                  #7  0x000073767599f24a __libc_start_main (libc.so.6 + 0x2624a)
                                                  #8  0x0000618b0400fef5 n/a (bluetoothd + 0xaef5)
                                                  ELF object binary architecture: AMD x86-64
```

belenos, 2024-11-02:
Happening to me as well, every couple of hours. Pretty annoying.

`bluetoothctl: 5.78`

```
Nov 02 03:23:08 valkyrie bluetoothd[8688]: profiles/input/device.c:hidp_send_message() BT socket not connected
Nov 02 03:23:08 valkyrie systemd-coredump[24164]: Process 8688 (bluetoothd) of user 0 terminated abnormally with signal 11/SEGV, processing...
Nov 02 03:23:08 valkyrie kernel: bluetoothd[8688]: segfault at 5712412c7187 ip 000057172a5e744c sp 00007fff85084310 error 4 in bluetoothd[ba44c,57172a535000+ea000] likely on CPU 1 (core 1, socket 0)
Nov 02 03:23:08 valkyrie systemd-coredump[24165]: Process 8688 (bluetoothd) of user 0 dumped core.
                                                  #0  0x000057172a5e744c n/a (bluetoothd + 0xba44c)
                                                  #1  0x000057172a608bf9 n/a (bluetoothd + 0xdbbf9)
                                                  #5  0x000057172a537005 n/a (bluetoothd + 0xa005)
                                                  #8  0x000057172a538305 n/a (bluetoothd + 0xb305)
Nov 02 03:23:08 valkyrie systemd[1]: bluetooth.service: Main process exited, code=dumped, status=11/SEGV
Nov 02 03:23:08 valkyrie systemd[1]: bluetooth.service: Failed with result 'core-dump'.
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Bluetooth daemon 5.78
Nov 02 03:23:09 valkyrie bluetoothd[24178]: src/main.c:check_config() Unknown group ConnectionParameters in /etc/bluetooth/main.conf
Nov 02 03:23:09 valkyrie bluetoothd[24178]: src/main.c:check_options() Unknown key IdleTimeout for group General in /etc/bluetooth/main.conf
Nov 02 03:23:09 valkyrie bluetoothd[24178]: src/main.c:check_options() Unknown key MultiProfilePolicy for group Policy in /etc/bluetooth/main.conf
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Starting SDP server
Nov 02 03:23:09 valkyrie bluetoothd[24178]: src/plugin.c:init_plugin() System does not support bap plugin
Nov 02 03:23:09 valkyrie bluetoothd[24178]: src/plugin.c:init_plugin() System does not support bass plugin
Nov 02 03:23:09 valkyrie bluetoothd[24178]: src/plugin.c:init_plugin() System does not support mcp plugin
Nov 02 03:23:09 valkyrie bluetoothd[24178]: src/plugin.c:init_plugin() System does not support vcp plugin
Nov 02 03:23:09 valkyrie bluetoothd[24178]: profiles/audio/micp.c:micp_init() D-Bus experimental not enabled
Nov 02 03:23:09 valkyrie bluetoothd[24178]: src/plugin.c:init_plugin() System does not support micp plugin
Nov 02 03:23:09 valkyrie bluetoothd[24178]: src/plugin.c:init_plugin() System does not support ccp plugin
Nov 02 03:23:09 valkyrie bluetoothd[24178]: src/plugin.c:init_plugin() System does not support csip plugin
Nov 02 03:23:09 valkyrie bluetoothd[24178]: src/plugin.c:init_plugin() System does not support asha plugin
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Bluetooth management interface 1.23 initialized
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Battery Provider Manager created
Nov 02 03:23:09 valkyrie bluetoothd[24178]: profiles/audio/avctp.c:avctp_server_socket() setsockopt(L2CAP_OPTIONS): Invalid argument (22)
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Failed to set privacy: Rejected (0x0b)
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/ldac
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSink/aptx_hd
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/aptx_hd
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSink/aptx
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/aptx
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSink/aac
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/aac
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSink/opus_g
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/opus_g
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSink/sbc
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/sbc
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/faststream
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/faststream_duplex
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSink/opus_05
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/opus_05
Nov 02 03:23:09 valkyrie bluetoothd[24178]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSink/opus_05_duplex
[... cut ...] 78]: Endpoint registered: sender=:1.85 path=/MediaEndpoint/A2DPSource/opus_05_duplex
```

```
OS: EndeavourOS Linux x86_64 
Host: B550M AORUS ELITE -CF 
Kernel: 6.11.5-arch1-1 
Uptime: 2 hours, 14 mins 
Packages: 1149 (pacman) 
Shell: bash 5.2.37 
Resolution: 1920x1080 
DE: GNOME 47.1 
WM: Mutter 
WM Theme: Adwaita 
Theme: adw-gtk3 [GTK2/3] 
Icons: Adwaita [GTK2/3] 
Terminal: kgx 
CPU: AMD Ryzen 5 5600 (12) @ 4.4GHz 
GPU: AMD ATI Radeon RX 550 640SP / RX 56 
Memory: 5.73GiB / 15.54GiB (36%) 
```

closed 2025-07-27 by github-actions[bot]