bluez/bluez #803: Mouse scroll wheel not working after reconnecting to bluetoothd (regression between 5.72 and 5.73)

fadein, 2024-04-04:
Symptoms: after upgrading to BlueZ 5.73, the scroll wheel on my Bluetooth mouse stopped working after reconnecting to `bluetoothd` (i.e. turning the mouse off and on, or waking the laptop up from suspend).  All other functions of the mouse continue to work as before; only the scroll wheel is affected.  I can easily reproduce this issue on demand.

`git bisect` identifies this commit as the culprit:

```
commit 247ae8524 (HEAD, tag: bisect-bad)
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   2024-02-09 07:55:24 -0700

    hog-lib: Don't destroy UHID device on detach

    This makes bt_hog_detach not to destroy UHID device which means the
    device node don't need to be recreated in case of reconnections which
    speeds up the process.

    Fixes: https://github.com/bluez/bluez/issues/737
```

When the problem occurs, X11 apps do not respond to mouse button 4 or 5.  From `xev`'s perspective, the ButtonPress events never happen.

On the bad commit (247ae8524), `btmon` shows these events when I scroll the wheel downward after re-connecting the mouse.  This tells me that Bluez perceives every slight touch of the mouse, though they are not passed on to X11.

```
> ACL Data RX: Handle 3585 flags 0x02 dlen 16                       #2816 [hci0] 550.118037
      ATT: Handle Value Notification (0x1b) len 11
        Handle: 0x0016 Type: Report (0x2a4d)
          Data: 0000000000ffff0000
> ACL Data RX: Handle 3585 flags 0x02 dlen 16                       #2817 [hci0] 550.169992
      ATT: Handle Value Notification (0x1b) len 11
        Handle: 0x0016 Type: Report (0x2a4d)
          Data: 0000000000ffff0000
```

Correct me if I'm wrong, but the velocity of the scroll wheel appears in the rightmost 32-bits of the Data field.  Notice the 2nd word in the Data field is `0xffff0000` when I scroll the mouse wheel down when this problem occurs.

Every scroll wheel event looks the same after re-connecting the mouse, no matter how fast or slow I flick the wheel.  When I scroll the wheel up, this value is always `0x01000000.`  These are the *only* two values that `btmon` shows for the scroll wheel.

When the mouse is functioning normally, other values appear in the data field which change with how fast or slow I move the scroll wheel.  The following examples are seen in `btmon` under these circumstances:

*   In commits before the bad commit, regardless of how many times the mouse has been disconnected/reconnected
*   On the bad commit after the mouse initially connects to bluetoothd, but before it is disconnected for the first time

Scrolling down:

```
> ACL Data RX: Handle 3585 flags 0x02 dlen 16                        #1297 [hci0] 96.115838
      ATT: Handle Value Notification (0x1b) len 11
        Handle: 0x0016 Type: Report (0x2a4d)
          Data: 0000000000f7ff0000
> ACL Data RX: Handle 3585 flags 0x02 dlen 16                        #1298 [hci0] 96.121969
      ATT: Handle Value Notification (0x1b) len 11
        Handle: 0x0016 Type: Report (0x2a4d)
          Data: 0000000000f8ff0000
```

This is what a good "scroll up" event looks like:

```
> ACL Data RX: Handle 3585 flags 0x02 dlen 16                       #1362 [hci0] 131.312962
      ATT: Handle Value Notification (0x1b) len 11
        Handle: 0x0016 Type: Report (0x2a4d)
          Data: 000000000008000000
> ACL Data RX: Handle 3585 flags 0x02 dlen 16                       #1363 [hci0] 131.364980
      ATT: Handle Value Notification (0x1b) len 11
        Handle: 0x0016 Type: Report (0x2a4d)
          Data: 000000000009000000
```

Thank you for looking into this.  Please reach out if you need more information.  As I said above, I can easily reproduce this issue.

ShadwDrgn, 2024-04-07:
Is it that it stops working entirely or do you have to scroll a lot and fast for it to catch even one scroll? My MX Master 3 has a problem where if it goes into power saving or i turn it off on the back of the mouse and back on then mouse scrolling becomes unbearably insensitive to the point i have to scroll like 360 degrees or more for one line worth of scroll. If I restart bluetooth it seems to go back to normal. downgrading to 5.72 also fixed it permanently (until i upgrade again of course)

lazarosgogos, 2024-04-07:
> Is it that it stops working entirely or do you have to scroll a lot and fast for it to catch even one scroll? My MX Master 3 has a problem where if it goes into power saving or i turn it off on the [...]

Same problem here, I don't know how to look into it further though. Downgraded to 5.69 and it is resolved for now (Fedora 39)

fadein, 2024-04-07:
> Is it that it stops working entirely or do you have to scroll a lot and fast for it to catch even one scroll? My MX Master 3 has a problem where if it goes into power saving or i turn it off on the [...]

Yes, I experience the exact the same issue.  I have a Microsoft Arc Mouse with a touchpad instead of a real wheel.  It is difficult to "spin" it that much, but it will give me maybe one or two lines of scroll for my efforts.

> If I restart bluetooth it seems to go back to normal. downgrading to 5.72 also fixed it permanently (until i upgrade again of course)

This sounds exactly like my situation.

rburcham, 2024-04-08:
Exact same device (MS Arc Mouse), observations (BT low power -> active transition results in scrolling requiring Olympic effort to move a single line), action (downgrade 5.73 -> 5.72) and bypass as @fadein.

AdamWill, 2024-04-09:
the 'slow wheel scrolling after inactivity' bug for logitech mice is https://github.com/bluez/bluez/issues/778 . lots of people are affected by it. it sounds like this may possibly be a dupe.

fadein, 2024-04-10:
> the 'slow wheel scrolling after inactivity' bug for logitech mice is #778 . lots of people are affected by it. it sounds like this may possibly be a dupe.

Thanks for the heads up, @AdamWill.  The symptoms sure sound the same.  I'm reading through the comments to get a better understanding of the underlying issue.  Would it be helpful to bring up the commit I identified with `git bisect`?

AdamWill, 2024-04-10:
It can't hurt!

Vudentz, 2024-04-11:
Looks like we have yet another manufacturer that can't make a mouse that remembers users configuration then... anyway there is a replay logic on BlueZ git, please give it a try before we release 5.74.

fadein, 2024-04-14:
I can confirm that this problem is fixed as of f1a7ab0ef.

closed 2024-04-14 by fadein