bluez/bluez #804: 5.73: Devices who autoconnect aren't shown in either KDE nor bluetoothctl

NyaomiDEV, 2024-04-07:
Taking for example a DualSense controller, which also suffers from #777 in my system; I can power it on and it'll be usable, but bluez seems to have no control over it. It won't get reported as connected in KDE and even if I type `disconnect <mac>` in `bluetoothctl` it would tell me it isn't connected.

This behaviour also is the same with my DualShock 4 and my pair of Sony WH-1000XM4.

Hardware Info
- ASRock X570 Taichi with builtin Intel AX200 Bluetooth/WiFi combo card
- Artix Linux rolling
- bluez version 5.73-4.1
- Kernel version 6.8.4 (just moved from 6.8.2 which would bug my Bluetooth stack altogether)

Mershl, 2024-04-07:
Same goes for GNOME. The DualSense does not show the BT-connected icon in the quick settings menu for GNOME 46 (tested on Fedora 40 - kernel 6.8.4 - bluez 5.73).

lilxandrias, 2024-04-09:
Same issue here.
As you can see, my headphones are connected(shows even battery level), but shows like it's not connected.  But device is completely operational though.

[image]
The issue started few weeks ago if i remember correctly.

The same issue remains with blueman, bluetuith and others, so i don't think it's KDE specific bug.

I noticed this after applications started to automatically switch HSP profile to A2DP profile... I would also like to disable this behaviour somehow.

lilxandrias, 2024-04-09:
Figured out something: downgrading to older kernel works, i'm now trying to figure out which version caused the issue

lilxandrias, 2024-04-09:
https://bugzilla.kernel.org/show_bug.cgi?id=218680

d-air1, 2024-04-14:
I no longer seem to be having this issue on Arch

lilxandrias, 2024-04-15:
Bluetooth fix made it into 6.9-rc4 mainline: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/diff/net/bluetooth/l2cap_core.c?id=v6.9-rc4&id2=v6.9-rc3

But it seems the fix isn't available in stable 6.8.6 yet.

NyaomiDEV, 2024-04-15:
... hope they'll also backport it...