bluez/bluez #806: bluez-5.73 only installs org.bluez.obex.service when systemd support is enabled

pacho2, 2024-04-07:
Even after applying:
https://github.com/bluez/bluez/commit/b16b19885c5383cd0499503617b98ceb188c898e

Current build system installs dbussessionbus_DATA = obexd/src/org.bluez.obex.service only when systemd is enabled. The following changes would be needed:
```
--- bluez-5.73.orig/Makefile.obexd	2024-04-02 10:38:48.743419656 +0200
+++ bluez-5.73/Makefile.obexd	2024-04-02 10:40:53.300086281 +0200
@@ -1,13 +1,13 @@
 # SPDX-License-Identifier: GPL-2.0
 if OBEX
 
+dbussessionbusdir = $(DBUS_SESSIONBUSDIR)
+dbussessionbus_DATA = obexd/src/org.bluez.obex.service
+
 if SYSTEMD
 systemduserunitdir = $(SYSTEMD_USERUNITDIR)
 systemduserunit_DATA = obexd/src/obex.service
 
-dbussessionbusdir = $(DBUS_SESSIONBUSDIR)
-dbussessionbus_DATA = obexd/src/org.bluez.obex.service
-
 obexd-add-service-symlink:
 	$(LN_S) -f obex.service $(DESTDIR)$(SYSTEMD_USERUNITDIR)/dbus-org.bluez.obex.service
 ```

Thanks

joakim-tjernlund, 2024-04-09:
I think this was working until commit "build: obexd: Fix make distcheck" which moved it back under systemd.
@Vudentz ?

Vudentz, 2024-04-09:
> I think this was working until commit "build: obexd: Fix make distcheck" which moved it back under systemd. @Vudentz ?

Haven't found a way to resolve the build error though, so fill free to propose a change.

joakim-tjernlund, 2024-04-09:
> > I think this was working until commit "build: obexd: Fix make distcheck" which moved it back under systemd. @Vudentz ?
> 
> Haven't found a way to resolve the build error though, so fill free to propose a change.

Just tried make distcheck on a clean checkout and I get a bunch of:
```
n file included from ../../profiles/audio/bap.c:47:
../../src/shared/bap.h:13:10: fatal error: src/shared/bap-defs.h: No such file or directory
   13 | #include "src/shared/bap-defs.h"
```
Should I do something else before make distcheck ?

Vudentz, 2024-04-09:
> Just tried make distcheck on a clean checkout and I get a bunch of:
>
> ```
> n file included from ../../profiles/audio/bap.c:47:
> ../../src/shared/bap.h:13:10: fatal error: src/shared/bap-defs.h: No such [...]

```
git clone https://github.com/bluez/bluez.git bluez-github
./bootstrap-configure && make -j23 distcheck
...
============================================
bluez-5.73 archives ready for distribution: 
bluez-5.73.tar.xz
============================================
```

joakim-tjernlund, 2024-04-09:
> ```
> git clone https://github.com/bluez/bluez.git bluez-github
> ./bootstrap-configure && make -j23 distcheck
> ...
> ============================================
> bluez-5.73 archives ready for [...]

....
configure: error: ELL source is required or use --enable-external-ell

....
./bootstrap-configure --enable-external-ell
...
make --no-print-directory distdir-am
make[2]: *** No rule to make target 'ell/util.h', needed by 'distdir-am'.  Stop.
make[1]: *** [Makefile:12019: distdir] Error 2
make: *** [Makefile:12099: dist] Error 2

gudnimg, 2024-06-30:
Just to add to the discussion here, this is also a problem for the Yocto project. On BlueZ 5.72 this patch is being used: https://git.openembedded.org/openembedded-core/tree/meta/recipes-connectivity/bluez5/bluez5/0001-Allow-using-obexd-without-systemd-in-the-user-sessio.patch and it seems to me it is a really old patch (~8 years)

While much of the patch is not relevant anymore I believe  this section is still relevant (just needs to be changed for BlueZ 5.73 and newer):

```
---
 Makefile.obexd                                                | 4 ++--
 .../src/{org.bluez.obex.service => org.bluez.obex.service.in} | 2 +-
 2 files changed, 3 insertions(+), 3 deletions(-)
 rename obexd/src/{org.bluez.obex.service => org.bluez.obex.service.in} (76%)

diff --git a/Makefile.obexd b/Makefile.obexd
index de59d29..73004a3 100644
--- a/Makefile.obexd
+++ b/Makefile.obexd
@@ -1,12 +1,12 @@
 if SYSTEMD
 systemduserunitdir = $(SYSTEMD_USERUNITDIR)
 systemduserunit_DATA = obexd/src/obex.service
+endif
 
 dbussessionbusdir = $(DBUS_SESSIONBUSDIR)
 dbussessionbus_DATA = obexd/src/org.bluez.obex.service
-endif
 
-EXTRA_DIST += obexd/src/obex.service.in obexd/src/org.bluez.obex.service
+EXTRA_DIST += obexd/src/obex.service.in obexd/src/org.bluez.obex.service.in
 
 if OBEX
```

pacho2, 2024-11-08:
```
--- a/Makefile.obexd
+++ b/Makefile.obexd
@@ -4,15 +4,14 @@
 if SYSTEMD
 systemduserunit_DATA += obexd/src/obex.service
 
-dbussessionbusdir = $(DBUS_SESSIONBUSDIR)
-dbussessionbus_DATA = obexd/src/org.bluez.obex.service
-
 obexd-add-service-symlink:
 	$(LN_S) -f obex.service $(DESTDIR)$(SYSTEMD_USERUNITDIR)/dbus-org.bluez.obex.service
 
 obexd-remove-service-symlink:
 	rm -f $(DESTDIR)$(SYSTEMD_USERUNITDIR)/dbus-org.bluez.obex.service
 else
+dbussessionbusdir = $(DBUS_SESSIONBUSDIR)
+dbussessionbus_DATA = obexd/src/org.bluez.obex.service
 obexd-add-service-symlink:
 obexd-remove-service-symlink:
 endif
```

Updated patch for 5.79

tagunil, 2025-04-18:
> ```
> --- a/Makefile.obexd
> +++ b/Makefile.obexd
> @@ -4,15 +4,14 @@
>  if SYSTEMD
>  systemduserunit_DATA += obexd/src/obex.service
>
> -dbussessionbusdir = $(DBUS_SESSIONBUSDIR)
> -dbussessionbus_DATA = [...]

This one breaks the systemd case. I believe the dbus service file should be installed in both cases, shouldn't it?

seamusdemora, 2025-05-18:
> This one breaks the systemd case. I believe the dbus service file should be installed in both cases, shouldn't it?

Well - it certainly appears to be BROKEN. 

```
~/bluez $ make
...

obexd/src/logind.c:23:10: fatal error: systemd/sd-login.h: No such file or directory
   23 | #include <systemd/sd-login.h>
      |          ^~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile:9779: obexd/src/obexd-logind.o] Error 1
make: *** [Makefile:4693: all] Error 2
``` 

Does anyone have a fix for this??