bluez/bluez #808: how to be notified when a Bluetooth device disconnects

pfps, 2024-04-09:
With bluez 5.72 it was easy to determine when a device disconnected, because the device nodes disappeared.  But with bluez 5.73 this no longer works.

How can a program now determine when a Bluetooth device they are communicating with via the HID RAW interface disconnects?

Vudentz, 2024-04-09:
@pfps nothing has changed with respect to the D-Bus interface, you still can use Connected event:

https://github.com/bluez/bluez/blob/master/doc/org.bluez.Device.rst#boolean-connected-readonly

On top of it, we shall also be destroying the device node if the driver attempts to communicate while disconnected.

pfps, 2024-04-10:
@Vudentz Thanks.  

Solaar https://github.com/pwr-Solaar/Solaar/ needs to set up Logitech devices when they reconnect, even if no communication attempts happened while the device was disconnected.  So Solaar can't depend on a write failure to find out that a device disconnected.   Solaar does listen for new data from the device though.  Will 5.74 cancel existing reads when a device disconnects?

Currently Solaar uses udev to keep track of Logitech devices, which can use USB or Bluetooth.   My guess is that Solaar now has to also use D-Bus as well as it appears to me that udev device creation events do not all show up in D-Bus.  Can you confirm that?

Vudentz, 2024-04-10:
> @Vudentz Thanks.
> 
> Solaar [pwr-Solaar/Solaar](https://github.com/pwr-Solaar/Solaar/) needs to set up Logitech devices when they reconnect, even if no communication attempts happened while the device was disconnected. So Solaar can't depend on a write failure to find out that a device disconnected. Solaar does listen for new data from the device though. Will 5.74 cancel existing reads when a device disconnects?

You can probably figure that out yourself by just attempting to connect with current master, is it supposed to work with MX Anywhere 3?

> Currently Solaar uses udev to keep track of Logitech devices, which can use USB or Bluetooth. My guess is that Solaar now has to also use D-Bus as well as it appears to me that udev device creation events do not all show up in D-Bus. Can you confirm that?

Well the user session is not really supposed to configure devices like that, since then each user has to configure the device separately which I bet would create conflicts if there are more than one user, also having the hability to capture user input by a third party application is sort of security breach, this is why we have dedicate plugin in bluetoothd and driver in kernel so the applications cannot eavesdrop what the users are typing.

pfps, 2024-04-10:
@Vudentz Solaar is used to modify device behaviour, for example by changing the behaviour of the configurable scroll wheel ratchet on Logitech MX Master 3 mice.  It seems to me that this is something that should only happen in the user session, and only for devices that are associated with the user's seat, as Solaar is limited to.

Vudentz, 2024-04-10:
@pfps looks like it is working just fine here with my MX Anywhere 3, that said is that supposed to remember the setting everytime the device is reconnected? Which the changes introduce to replay on reconnection now it does remember everything the user has set, so I guess it is an improvement in that respect.

pfps, 2024-04-10:
@Vudentz What happens if the user changes something during the session?  It seems to me that either these in-session changes will not be replayed, or extra messages will be sent.  Is there a limit to the number of messages that are remembered?  It is possible that a user session could last for days, which quite a few changes to the device behaviour happening over these days.

Vudentz, 2024-04-10:
> @Vudentz What happens if the user changes something during the session? It seems to me that either these in-session changes will not be replayed, or extra messages will be sent. Is there a limit to [...]

All GET/SET_REPORT are recorded during the connection, regardless if they were generated by the kernel driver or not. As to limit the number of events recorded this way, there is no limit currently, we could attempt to set a upper limit but I didn't feel it was necessary since the number of messages is quite limited and in any case they are not persisted across daemon reboots.

pfps, 2024-04-10:
@Vudentz 

I am now convinced that replay is not a suitable solution to reinitializing devices.  Replay depends on device commands not being sensitive to device state and also being idempotent.  There are setup commands for Bluetooth Logitech devices that violate these dependencies - including the command that the Linux driver for HID++ devices https://github.com/torvalds/linux/blob/master/drivers/hid/hid-logitech-hidpp.c uses.  (Perhaps fortunately the driver is very ham-fisted and uses the command in a way that does not depend on context but an improved implementation should use the command in a way that does depend on context.)

pfps, 2024-04-10:
@Vudentz Is there a better place to discuss replay?

Vudentz, 2024-04-10:
> > > [...] Is there an easy way to temporarily install the new version of bluez on a Fedora/XFCE system?
> > 
> > 
> > I usually recommend building and running it from source if possible:
> > [`master`/HACKING#L38](https://github.com/bluez/bluez/blob/master/HACKING?rgh-link-date=2024-04-10T15%3A01%3A50Z#L38)
> 
> That's failing in Fedora because
> 
> ```
> configure: error: ELL source is required or use --enable-external-ell
> ```

Install libell in the parent directory:

https://git.kernel.org/pub/scm/libs/ell/ell.git/

pfps, 2024-04-10:
@Vudentz I don't understand how the Linux driver is a suitable place for doing things like setting the characteristics of the scroll wheel ratchet of Logitech mice.  That is a user preference.  I certainly don't understand how the Linux driver is a suitable place to switch Logitech mice between computers.

pfps, 2024-04-10:
There are two issues intertwined here.

First, there are devices that need have messages sent to them to set them up when they reconnect via Bluetooth.  These devices include many Logitech devices.   The reason that these devices need to be set up is that they lose some of their local state when they are turned off or go into a power-saving mode.  So some system has to know when these devices reconnect and has to send the appropriate messages to change their local state to the desired one.

Second, recording all messages sent to the device and replaying them at reconnect time to set up devices is problematic.  There are lots of reasons that a message might be sent to the device.  For example, a message might be sent to the device to find out part of its current state.  Resending these mssages and then ignoring the response is inefficient.  Worse, the response from the device might affect a subsequent message.  Blindly replying such subsequent messages ignores the current state of the device.  Even worse, a message might be sent to the device to change its local state in a way that is undesirable at the time of reconnection.  

It doesn't matter whether the program sending the messages to be replayed is the driver, some other system program, or even a user program.   In all cases blind replaying of messages at reconnect time can lead to bad results.

Vudentz, 2024-04-10:
> There are two issues intertwined here.
>
> First, there are devices that need have messages sent to them to set them up when they reconnect via Bluetooth. These devices include many Logitech devices. [...]

I understand that, but the vast majority of drivers will just send a fixed sequence to setup the device, if that doesn't work for Logitech than we need something different for it, but that is the exception not the rule and to be honest that is sort of a bad desing from Logitech, how come it doesn't remember its last settings? No wonder it would fall flat to support different OS, because each and every OS would need to implement their vendor protocol to reprogram the user settings on every reconnection.

pfps, 2024-04-10:
There are good reasons for small devices to not remember their settings, at least until recently.  Persistent storage can be expensive, unreliable, or have limited lifetime so it may be better to use memory that is cleared whenever the device shuts down.  Logitech is a well-known vendor of such devices, but there may be others.

It may be that the only Linux driver that does device setup is the one for Logitech HID++ devices.  But there are several Linux programs that configure devices, including Solaar and https://github.com/libratbag/libratbag.

Vudentz, 2024-04-10:
> There are good reasons for small devices to not remember their settings, at least until recently. Persistent storage can be expensive, unreliable, or have limited lifetime so it may be better to use memory that is cleared whenever the device shuts down. Logitech is a well-known vendor of such devices, but there may be others.
> 
> It may be that the only Linux driver that does device setup is the one for Logitech HID++ devices. But there are several Linux programs that configure devices, including Solaar and [libratbag/libratbag](https://github.com/libratbag/libratbag).

Well I though the days of not having a persistent storage was over with LE, because all the GATT notification subscriptions needs to be stored and if you have storage for that you sure have storage for user settings since those are actually propagated over GATT, anyway if you don't have a reliable storage then GATT itself may not work properly, so you will not be able to convince me that LE devices can't remember their settings for any reason other than they are leaving it up to the Windows driver because they only care about Windows in the end.

pfps, 2024-04-10:
Well, yes, this may now just be a matter of inertia and only caring about Windows, particularly for Bluetooth devices.