bluez/bluez #809: Failed to add UUID: Failed (0x03)

Opendreas, 2024-04-10:
I have Intel Wireless-AC 9260 Bluetooth Adapter and following errors when booting the system.

```
bluetoothd	src/adapter.c:reset_adv_monitors_complete() Failed to reset Adv Monitors: Failed (0x03)
bluetoothd	Failed to clear UUIDs: Failed (0x03)
bluetoothd	Failed to add UUID: Failed (0x03)
bluetoothd	Failed to add UUID: Failed (0x03)
bluetoothd	Failed to add UUID: Failed (0x03)
bluetoothd	Failed to add UUID: Failed (0x03)
bluetoothd	Failed to add UUID: Failed (0x03)
bluetoothd	Failed to add UUID: Failed (0x03)
```

These errors are present on kernels 6.4, 6.5, 6.8, 6.9-rc1 and bluez 5.71.

[hwinfo.txt](https://github.com/bluez/bluez/files/14925901/hwinfo.txt)
[journald-6.4.0.txt](https://github.com/bluez/bluez/files/14925903/journald-6.4.0.txt)
[journald-6.5.9.txt](https://github.com/bluez/bluez/files/14925904/journald-6.5.9.txt)
[journald-6.8.2.txt](https://github.com/bluez/bluez/files/14925906/journald-6.8.2.txt)
[journald-6.9.0-rc1.txt](https://github.com/bluez/bluez/files/14925908/journald-6.9.0-rc1.txt)

Opendreas, 2024-04-26:
After the fix patch was backported into the distribution kernel by the maintainer, the UID errors disappeared, but one more remained
```
bluetoothd src/adapter.c:reset_adv_monitors_complete() Failed to reset Adv Monitors: Failed (0x03)
```

Strange warnings also appeared
```
bluetoothd[1513]: src/plugin.c:plugin_init() System does not support micp plugin
bluetoothd[1513]: src/plugin.c:plugin_init() System does not support vcp plugin
bluetoothd[1513]: src/plugin.c:plugin_init() System does not support mcp plugin
bluetoothd[1513]: src/plugin.c:plugin_init() System does not support bass plugin
bluetoothd[1513]: src/plugin.c:plugin_init() System does not support bap plugin
```
[journald-6.4.0-150600.txt](https://github.com/bluez/bluez/files/15133802/journald-6.4.0-150600.txt)

Opendreas, 2024-04-26:
> bluetoothd src/adapter.c:reset_adv_monitors_complete() Failed to reset Adv Monitors: Failed (0x03)
I just found out that there is already a fix for this error.

YoinkerBoinker, 2024-05-11:
I got 3 Systems (all are using different TP-Link UB400 Adapters) and all of them show the very same messages in the Journal. I'm not sure if it is because i usually have BT disabled and just enable it when needed. 

```
Mai 11 08:34:29 linux kernel: Bluetooth: hci0: unexpected event for opcode 0x0000
Mai 11 08:34:30 linux bluetoothd[620]: Failed to set mode: Failed (0x03)
Mai 11 08:34:30 linux bluetoothd[620]: Too small Add Device complete event
Mai 11 08:34:30 linux bluetoothd[620]: Too small Add Device complete event
Mai 11 08:34:34 linux bluetoothd[620]: src/gatt-database.c:database_add_chrc() Failed to create characteristic entry in database
Mai 11 08:34:34 linux bluetoothd[620]: src/gatt-database.c:database_add_service() Failed to add characteristic
Mai 11 08:34:34 linux bluetoothd[620]: src/gatt-database.c:database_add_app() Failed to add service
Mai 11 08:34:34 linux bluetoothd[620]: src/gatt-database.c:client_ready_cb() Failed to create GATT service entry in local database
Mai 11 08:34:34 linux bluetoothd[620]: Failed to set mode: Failed (0x03)
```

erfan-star-1999, 2024-05-11:
Hi i have the same error but it's been at least a two months i was thinking they fix it because last week i got update for bluez and bluez-qt on arch but still there

[image]

Vudentz, 2024-05-13:
> Hi i have the same error but it's been at least a two months i was thinking they fix it because last week i got update for bluez and bluez-qt on arch but still there
> 
> ![Screenshot_20240511_201916](https://private-user-images.githubusercontent.com/73936623/329801423-552c9542-77bd-43ba-b496-8054abccb1e2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2MzY0OTMsIm5iZiI6MTcxNTYzNjE5MywicGF0aCI6Ii83MzkzNjYyMy8zMjk4MDE0MjMtNTUyYzk1NDItNzdiZC00M2JhLWI0OTYtODA1NGFiY2NiMWUyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDIxMzYzM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJmMDhiMGFjM2NiYzM4ODgwNmQ5Njc4Y2JmN2MwYzNmMzRkZDIwZWE4MWYwOTkzY2ZiNzI4MmQwZmYwMTljN2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.2QXS4I5i1e4LI0nYjm6MGQmfLCs9C-NeYAf3_WIVmqM)

This is a different error though, looks like something is trying to register services in the gatt database and that end up failing.

Vudentz, 2024-05-13:
@Opendreas if this was fixed for you please close it.

erfan-star-1999, 2024-05-14:
> This is a different error though, looks like something is trying to register services in the gatt database and that end up failing.

so what should i do? i make new issue?

Opendreas, 2024-05-15:
fixed.

closed 2024-05-15 by Opendreas