bluez/bluez #810: Connect to both LE and BR/EDR Bluetooth devices simultaneously Authentication Failure

dabeixiaomibujiatang, 2024-04-10:
buildroot+kernel 6.7.0-rc1+bluez5.70（board）
Description of Problem:
I used the nRF Connect app on my phone to simultaneously connect to Bluetooth devices over LE and BR/EDR. While the BLE connection was successful, the BR/EDR connection failed. The BR/EDR disconnection occurred due to an Authentication Failure.
hcidump log : 
[btsnoop_log.log](https://github.com/bluez/bluez/files/14930045/btsnoop_log.log)
and 
```
Bluetooth: [hci_encrypt_change_evt:3591]hci0: status 0x00
Bluetooth: [hci_conn_check_link_mode:2362]hcon 00000000cdd17cc6
Bluetooth: hci0: Invalid security: Missing AES-CCM usage
Bluetooth: [l2cap_security_cfm:8332]conn 00000000829f7e18 status 0x05 encrypt 1
Bluetooth: [hci_disconnect:242]hcon 00000000cdd17cc6
Bluetooth: [hci_abort_conn:2970]hci0: handle 0x33 reason 0x05
```

Vudentz, 2024-04-10:
BR/EDR controller needs to encrypt the link properly, using AES-CCM:

```
commit 8746f135bb01872ff412d408ea1aa9ebd328c1f5
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Wed May 20 14:20:14 2020 -0700

    Bluetooth: Disconnect if E0 is used for Level 4
    
    E0 is not allowed with Level 4:
    
    BLUETOOTH CORE SPECIFICATION Version 5.2 | Vol 3, Part C page 1319:
    
      '128-bit equivalent strength for link and encryption keys
       required using FIPS approved algorithms (E0 not allowed,
       SAFER+ not allowed, and P-192 not allowed; encryption key
       not shortened'
    
    SC enabled:
    
    > HCI Event: Read Remote Extended Features (0x23) plen 13
            Status: Success (0x00)
            Handle: 256
            Page: 1/2
            Features: 0x0b 0x00 0x00 0x00 0x00 0x00 0x00 0x00
              Secure Simple Pairing (Host Support)
              LE Supported (Host)
              Secure Connections (Host Support)
    > HCI Event: Encryption Change (0x08) plen 4
            Status: Success (0x00)
            Handle: 256
            Encryption: Enabled with AES-CCM (0x02)
    
    SC disabled:
    
    > HCI Event: Read Remote Extended Features (0x23) plen 13
            Status: Success (0x00)
            Handle: 256
            Page: 1/2
            Features: 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x00
              Secure Simple Pairing (Host Support)
              LE Supported (Host)
    > HCI Event: Encryption Change (0x08) plen 4
            Status: Success (0x00)
            Handle: 256
            Encryption: Enabled with E0 (0x01)
    [May 8 20:23] Bluetooth: hci0: Invalid security: expect AES but E0 was used
    < HCI Command: Disconnect (0x01|0x0006) plen 3
            Handle: 256
            Reason: Authentication Failure (0x05)
    
    Signed-off-by: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
    Signed-off-by: Marcel Holtmann <marcel@holtmann.org>

```

closed 2024-04-10 by Vudentz

dabeixiaomibujiatang, 2024-04-11:
BLUETOOTH CORE SPECIFICATION Version 5.2 | Vol 3, Part C page 1401:
14.1 CROSS-TRANSPORT KEY DERIVATION
If both the local and remote devices support Secure Connections over the LE 
transport but not over the BR/EDR transport, then the devices may optionally 
generate the BR/EDR keys of identical strength and the same MITM protection 
as the LE keys as part of the LE pairing procedure (see [Vol 3] Part H, Section 
2.3.5.7).

It looks like bluez defaults to generating the BR/EDR keys with the same strength and MITM protection as the LE keys during the LE pairing procedure.

Vudentz, 2024-04-11:
> Yes, it used the P-256 type of key from LE generate the P-256 type key for BR/EDR, so it needs to use secure connections and AES-CCM encryption. But my controller BR/EDR does not seem to support secure connections. Is this a design issue with the controller?

Yep, the spec is pretty strick regarding it:

`E0 not allowed, SAFER+ not allowed, and P-192 not allowed; encryption key not shortened'`

So don't you cannot use SC if BR/EDR doesn't support it and you intend to connect it as well.