bluez/bluez #811: Encryption Change Event v2 not being used on bluetooth v5.3 controllers

jpdclerk, 2024-04-13:
When pairing a device with a bluetooth 5.3 controller it fails:
```
> HCI Event: Encryption Change (0x08) plen 4                                                                                     #1340 [hci0] 843.807535
        Status: Success (0x00)
        Handle: 2048 Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
        Encryption: Enabled with E0 (0x01)
< HCI Command: Read Encryption Key Size (0x05|0x0008) plen 2                                                                     #1341 [hci0] 843.807558
        Handle: 2048 Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
> HCI Event: Command Status (0x0f) plen 4                                                                                        #1342 [hci0] 843.808535
      Read Encryption Key Size (0x05|0x0008) ncmd 1
        Status: Unknown HCI Command (0x01)
> HCI Event: Encryption Key Refresh Complete (0x30) plen 3                                                                       #1343 [hci0] 843.867538
        Status: Success (0x00)
        Handle: 2048 Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
```
        
When pairing the same device with a bluetooth 5.0 controller it succeeds:
```
> HCI Event: Encryption Change (0x08) plen 4                                                                                       #64 [hci1] 121.510775
        Status: Success (0x00)
        Handle: 12 Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
        Encryption: Enabled with E0 (0x01)
< HCI Command: Read Encryption Key Size (0x05|0x0008) plen 2                                                                       #65 [hci1] 121.510795
        Handle: 12 Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
> HCI Event: Command Complete (0x0e) plen 7                                                                                        #66 [hci1] 121.511284
      Read Encryption Key Size (0x05|0x0008) ncmd 1																																		
        Status: Success (0x00)
        Handle: 12 Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
        Key size: 16
```
        
From the Bluetooth core specifications for version 5.3 (https://www.bluetooth.com/wp-content/uploads/2021/01/Bluetooth_5.3_Feature_Enhancements_Update.pdf), section 2.0 specifies a new version of the Encryption Change Event interface.
From the extracts above the new interface is not implemented.

The usb controller works on windows with the generic driver, so the hardware can mostly be eliminated as cause.

```
hciconfig -a
hci1:   Type: Primary  Bus: UART
        BD Address: D8:3A:DD:B8:E1:64  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:4531 acl:0 sco:0 events:443 errors:0
        TX bytes:69887 acl:0 sco:0 commands:443 errors:0
        Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH SNIFF
        Link mode: PERIPHERAL ACCEPT
        Name: 'raspberrypi'
        Class: 0x6c0000
        Service Classes: Rendering, Capturing, Audio, Telephony
        Device Class: Miscellaneous,
        HCI Version: 5.0 (0x9)  Revision: 0x17e
        LMP Version: 5.0 (0x9)  Subversion: 0x6119
        Manufacturer: Cypress Semiconductor (305)

hci0:   Type: Primary  Bus: USB
        BD Address: F4:4E:FC:BE:E2:5F  ACL MTU: 679:6  SCO MTU: 240:20
        UP RUNNING
        RX bytes:13238 acl:1 sco:0 events:390 errors:0
        TX bytes:5694 acl:0 sco:0 commands:120 errors:0
        Features: 0xbf 0xfe 0x0d 0xfe 0xdb 0xfd 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH SNIFF
        Link mode: PERIPHERAL ACCEPT
        Name: 'raspberrypi #1'
        Class: 0x6c0000
        Service Classes: Rendering, Capturing, Audio, Telephony
        Device Class: Miscellaneous,
        HCI Version:  (0xc)  Revision: 0x201
        LMP Version:  (0xc)  Subversion: 0x201
        Manufacturer: Actions (Zhuhai) Technology Co., Limited (992)
```

jpdclerk, 2024-04-14:
When switching off SSP the pairing process follows the working logic used by windows, but it still fails to pair since the host is not responding to the IO Capability Request of the controller.

```
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                                                                           #1401 [hci0] 845.373548
      Write Scan Enable (0x03|0x001a) ncmd 1
        Status: Success (0x00)
< HCI Command: Read Remote Extended Features (0x01|0x001c) plen 3                                                                                                                                                                                                     #1402 [hci0] 845.373565
        Handle: 2048 Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
        Page: 1
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                                                                                             #1403 [hci0] 845.374550
      Read Remote Extended Features (0x01|0x001c) ncmd 1
        Status: Success (0x00)
> HCI Event: Read Remote Extended Features (0x23) plen 13                                                                                                                                                                                                             #1404 [hci0] 845.375555
        Status: Success (0x00)
        Handle: 2048 Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
        Page: 1/1
        Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
          Secure Simple Pairing (Host Support)
< HCI Command: Remote Name Request (0x01|0x0019) plen 10                                                                                                                                                                                                              #1405 [hci0] 845.375568
        Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                                                                                             #1406 [hci0] 845.376547
      Remote Name Request (0x01|0x0019) ncmd 1
        Status: Success (0x00)
> HCI Event: Max Slots Change (0x1b) plen 3                                                                                                                                                                                                                           #1407 [hci0] 845.432561
        Handle: 2048 Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
        Max slots: 5
> HCI Event: Remote Name Req Complete (0x07) plen 255                                                                                                                                                                                                                 #1408 [hci0] 845.452547
        Status: Success (0x00)
        Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
        Name: DualSense Wireless Controller
@ MGMT Event: Device Connected (0x000b) plen 49                                                                                                                                                                                                                    {0x0002} [hci0] 845.452558
        BR/EDR Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
        Flags: 0x00000008
          Connection Locally Initiated
        Data length: 36
        Name (complete): DualSense Wireless Controller
        Class: 0x002508
          Major class: Peripheral (mouse, joystick, keyboards)
          Minor class: 0x02
          Limited Discoverable Mode
@ MGMT Event: Device Connected (0x000b) plen 49                                                                                                                                                                                                                    {0x0001} [hci0] 845.452558
        BR/EDR Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
        Flags: 0x00000008
          Connection Locally Initiated
        Data length: 36
        Name (complete): DualSense Wireless Controller
        Class: 0x002508
          Major class: Peripheral (mouse, joystick, keyboards)
          Minor class: 0x02
          Limited Discoverable Mode
< HCI Command: Authentication Requested (0x01|0x0011) plen 2                                                                                                                                                                                                          #1409 [hci0] 845.452569
        Handle: 2048 Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                                                                                             #1410 [hci0] 845.453555
[... cut ...]
        BR/EDR Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
@ MGMT Event: Command Complete (0x0001) plen 10                                                                                                                                                                                                                    {0x0001} [hci0] 965.414738
      Remove Device (0x0034) plen 7
        Status: Success (0x00)
        BR/EDR Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD)
```

Vudentz, 2024-04-15:
> When pairing a device with a bluetooth 5.3 controller it fails:
>
> ```
>         Status: Success (0x00)
>         Handle: 2048 Address: D0:BC:C1:CC:18:69 (WEIFANG GOERTEK ELECTRONICS CO.,LTD) [...]

Well this is typically the controller fails, it marks the command as supported but when we attempt to use it respond with Unknown command as seem above, so yes it is a controller fault and most likely you will need to use the following quirk:

```
	/*
	 * When this quirk is set, the HCI_OP_READ_ENC_KEY_SIZE command is
	 * skipped during an HCI_EV_ENCRYPT_CHANGE event. This is required
	 * for Actions Semiconductor ATS2851 based controllers, which erroneously
	 * claim to support it.
	 */
	HCI_QUIRK_BROKEN_READ_ENC_KEY_SIZE,
```