bluez/bluez #814: Can't use l2ping

Qazianwar222, 2024-04-15:
Hello sir I don't know if it's a kernel issue
Everything is working fine but I can't use l2ping from my phone nexus 6p (Kali nethunter Kernel) in chroot
Anything related to the issue might help
Or let me know where I'm doing wrong
[image]

Vudentz, 2024-04-15:
> Hello sir I don't know if it's a kernel issue Everything is working fine but I can't use l2ping from my phone nexus 6p (Kali nethunter Kernel) in chroot Anything related to the issue might help Or let me know where I'm doing wrong ![Screenshot_20240415-131133_NetHunter_Terminal](https://private-user-images.githubusercontent.com/88237249/322403242-aede756c-25bf-4058-884d-d9da984a2ae9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMxODc4MTksIm5iZiI6MTcxMzE4NzUxOSwicGF0aCI6Ii84ODIzNzI0OS8zMjI0MDMyNDItYWVkZTc1NmMtMjViZi00MDU4LTg4NGQtZDlkYTk4NGEyYWU5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE1VDEzMjUxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVkN2NmMDIxZGRjYWUyYTQzOGNkYWQzM2U4NjE3NDVjZjgxMjY5Y2NkYzcxMTNmZjNlZWIxZjQ0YTViMDYyNTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ZIZSaTZY5RSvMvzC3RtOa4raC_suNwDQKvpRKTCBGkY)

What BlueZ has to do with Android? What is l2flood? It looks like it is a modified version of l2ping but then if you are using the Android stack I doubt that it would work anyway.

Qazianwar222, 2024-04-18:
No it's not working
I can't use l2ping at any concern
It just doesn't give me the permission to do
Just saying "permission denied"

Qazianwar222, 2024-04-18:
l2flood is a tool used to disconnect 2 Bluetooth paired devices by sending ping requests 
It uses bluez tools (l2ping) to do so but I guess it's not working

closed 2025-06-12 by Qazianwar222