bluez/bluez #815: Random crash on device reconnect

serfreeman1337, 2024-04-15:
```bash
$ /usr/lib/bluetooth/bluetoothd -v
5.74
```

<details>
<summary>backtrace 1</summary>

```
Core was generated by `/usr/lib/bluetooth/bluetoothd'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x000062e772fcc6ff in queue_pop_head (queue=0x30) at src/shared/queue.c:155
155             if (!queue || !queue->head)                                                                                                                                                                                                 
(gdb) bt
#0  0x000062e772fcc6ff in queue_pop_head (queue=0x30) at src/shared/queue.c:155
#1  bt_uhid_record (uhid=uhid@entry=0x62e77357c650, input=input@entry=true, ev=ev@entry=0x7ffd19918b10) at src/shared/uhid.c:117
#2  0x000062e772fcca10 in bt_uhid_get_report_reply (uhid=0x62e77357c650, id=<optimized out>, number=<optimized out>, status=<optimized out>, data=<optimized out>, size=<optimized out>) at src/shared/uhid.c:484
#3  0x000062e773001e39 in uhid_send_get_report_reply.isra.0 (idev=<optimized out>, data=<optimized out>, size=8, id=<optimized out>, err=<optimized out>) at profiles/input/device.c:285
#4  0x000062e772f51670 in hidp_recv_ctrl_data (size=9, data=0x7ffd19919c70 "\241\003d\334,&C\035m", param=1 '\001', idev=0x62e77357c220) at profiles/input/device.c:517
#5  hidp_recv_ctrl_message (chan=<optimized out>, idev=0x62e77357c220) at profiles/input/device.c:576
#6  0x000062e772f516e4 in ctrl_watch_cb (chan=0x62e773573900, cond=G_IO_IN, data=0x62e77357c220) at profiles/input/device.c:594
#7  0x00007d1f1af63199 in g_main_dispatch (context=0x62e773568800) at ../glib/glib/gmain.c:3344
#8  0x00007d1f1afc23bf in g_main_context_dispatch_unlocked (context=0x62e773568800) at ../glib/glib/gmain.c:4152
#9  g_main_context_iterate_unlocked.isra.0 (context=0x62e773568800, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/glib/gmain.c:4217
#10 0x00007d1f1af63dc7 in g_main_loop_run (loop=0x62e7735689f0) at ../glib/glib/gmain.c:4419
#11 0x000062e772f23ef3 in mainloop_run () at src/shared/mainloop-glib.c:66
#12 mainloop_run_with_signal (func=0x62e772f774e0 <signal_callback>, user_data=0x0) at src/shared/mainloop-notify.c:188
#13 main (argc=<optimized out>, argv=<optimized out>) at src/main.c:1456
```
</details>

<details>
<summary>backtrace 2</summary>

```
Core was generated by `/usr/lib/bluetooth/bluetoothd'.
Program terminated with signal SIGABRT, Aborted.
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44            return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;                                                                                                                                                     
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x0000752cc3014393 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x0000752cc2fc36c8 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x0000752cc2fab4b8 in __GI_abort () at abort.c:79
#4  0x0000752cc33000f4 in g_assertion_message (domain=domain@entry=0x752cc33fe3a1 "GLib", file=file@entry=0x752cc33ffd1a "../glib/glib/gmain.c", line=line@entry=3993, 
    func=func@entry=0x752cc33a4f10 <__func__.21.lto_priv.8> "g_main_context_check_unlocked", message=message@entry=0x5c132f74ac90 "assertion failed: (i <= 0 || fds[i - 1].fd < fds[i].fd)") at ../glib/glib/gtestutils.c:3331
#5  0x0000752cc3367d90 in g_assertion_message_expr (domain=0x752cc33fe3a1 "GLib", file=0x752cc33ffd1a "../glib/glib/gmain.c", line=3993, func=0x752cc33a4f10 <__func__.21.lto_priv.8> "g_main_context_check_unlocked", 
    expr=<optimized out>) at ../glib/glib/gtestutils.c:3357
#6  0x0000752cc333df8d in g_main_context_check_unlocked (context=0x5c132f709800, max_priority=2147483647, fds=<optimized out>, n_fds=7) at ../glib/glib/gmain.c:3993
#7  0x0000752cc339c21e in g_main_context_iterate_unlocked.isra.0 (context=0x5c132f709800, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/glib/gmain.c:4214
#8  0x0000752cc333ddc7 in g_main_loop_run (loop=0x5c132f7099f0) at ../glib/glib/gmain.c:4419
#9  0x00005c132d9b3ef3 in mainloop_run () at src/shared/mainloop-glib.c:66
#10 mainloop_run_with_signal (func=0x5c132da074e0 <signal_callback>, user_data=0x0) at src/shared/mainloop-notify.c:188
#11 main (argc=<optimized out>, argv=<optimized out>) at src/main.c:1456
```
</details>

<details>
<summary>backtrace 3</summary>

```
Core was generated by `/usr/lib/bluetooth/bluetoothd'.
Program terminated with signal SIGABRT, Aborted.
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44            return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;                                                                                                                                                     
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x0000796022ddd393 in __pthread_kill_internal (signo=6, threadid=<optimized out>) at pthread_kill.c:78
#2  0x0000796022d8c6c8 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x0000796022d744b8 in __GI_abort () at abort.c:79
[... 37 lines cut ...]
    expr=<optimized out>) at ../glib/glib/gtestutils.c:3357
#6  0x000071bff25e7f8d in g_main_context_check_unlocked (context=0x5ba9e1b95800, max_priority=2147483647, fds=<optimized out>, n_fds=7) at ../glib/glib/gmain.c:3993
#7  0x000071bff264621e in g_main_context_iterate_unlocked.isra.0 (context=0x5ba9e1b95800, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/glib/gmain.c:4214
#8  0x000071bff25e7dc7 in g_main_loop_run (loop=0x5ba9e1b959f0) at ../glib/glib/gmain.c:4419
#9  0x00005ba9e0d6aef3 in mainloop_run () at src/shared/mainloop-glib.c:66
#10 mainloop_run_with_signal (func=0x5ba9e0dbe4e0 <signal_callback>, user_data=0x0) at src/shared/mainloop-notify.c:188
#11 main (argc=<optimized out>, argv=<optimized out>) at src/main.c:145
```
</details>

backtrace 4 related log:
```
bluetoothd[586]: profiles/input/device.c:hidp_send_message() BT socket not connected
kernel: apple XXXX:XXXX:XXXX.XXXX: unknown main item tag 0x0
kernel: apple XXXX:XXXX:XXXX.XXXX: Non-apple keyboard detected; function keys will default to fnmode=2 behavior
kernel: input: Keychron K1 SE as /devices/virtual/misc/uhid/XXXX:XXXX:XXXX.XXXX/input/input30
kernel: apple XXXX:XXXX:XXXX.XXXX: input,hidraw5: BLUETOOTH HID v1.03 Keyboard [Keychron K1 SE] on xx:xx:xx:xx:xx:xx
bluetoothd[586]: **
bluetoothd[586]: GLib:ERROR:../glib/glib/gmain.c:3993:g_main_context_check_unlocked: assertion failed: (i <= 0 || fds[i - 1].fd < fds[i].fd)
bluetoothd[586]: Bail out! GLib:ERROR:../glib/glib/gmain.c:3993:g_main_context_check_unlocked: assertion failed: (i <= 0 || fds[i - 1].fd < fds[i].fd)
systemd[1]: Created slice Slice /system/systemd-coredump.
systemd[1]: Started Process Core Dump (PID 2421/UID 0).
systemd-logind[413]: Watching system buttons on /dev/input/event21 (Keychron K1 SE)
systemd-coredump[2429]: [🡕] Process 586 (bluetoothd) of user 0 dumped core.
```

Vudentz, 2024-04-15:
> ```shell
> $ /usr/lib/bluetooth/bluetoothd -v
> 5.74
> ```
> 
> backtrace 1

This one seems that is crashing on a NULL pointer in queue_pop_head but the if statement is exactly checking for NULL pointer so it really beats me what is going on, can you try running it under valgrind?

> backtrace 2
> backtrace 3
> backtrace 4
> 
> backtrace 4 related log:
> 
> ```
> bluetoothd[586]: profiles/input/device.c:hidp_send_message() BT socket not connected
> kernel: apple XXXX:XXXX:XXXX.XXXX: unknown main item tag 0x0
> kernel: apple XXXX:XXXX:XXXX.XXXX: Non-apple keyboard detected; function keys will default to fnmode=2 behavior
> kernel: input: Keychron K1 SE as /devices/virtual/misc/uhid/XXXX:XXXX:XXXX.XXXX/input/input30
> kernel: apple XXXX:XXXX:XXXX.XXXX: input,hidraw5: BLUETOOTH HID v1.03 Keyboard [Keychron K1 SE] on xx:xx:xx:xx:xx:xx
> bluetoothd[586]: **
> bluetoothd[586]: GLib:ERROR:../glib/glib/gmain.c:3993:g_main_context_check_unlocked: assertion failed: (i <= 0 || fds[i - 1].fd < fds[i].fd)
> bluetoothd[586]: Bail out! GLib:ERROR:../glib/glib/gmain.c:3993:g_main_context_check_unlocked: assertion failed: (i <= 0 || fds[i - 1].fd < fds[i].fd)
> systemd[1]: Created slice Slice /system/systemd-coredump.
> systemd[1]: Started Process Core Dump (PID 2421/UID 0).
> systemd-logind[413]: Watching system buttons on /dev/input/event21 (Keychron K1 SE)
> systemd-coredump[2429]: [🡕] Process 586 (bluetoothd) of user 0 dumped core.
> ```

Well these other logs are crashing inside glib so I wonder what is going on here since I never seem anything like that, we don't even enable threading so the presence of pthread in the traces is really not expected.

serfreeman1337, 2024-04-15:
[valgrind.txt](https://github.com/bluez/bluez/files/14980529/valgrind.txt)

<details>
<summary>Also another backtrace</summary>

```
Core was generated by `/usr/lib/bluetooth/bluetoothd'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  queue_push_tail (queue=0x591305100110 <ctrl_watch_cb>, data=0x591306143eb0) at src/shared/queue.c:85
85                      queue->tail->next = entry;
(gdb) bt
#0  queue_push_tail (queue=0x591305100110 <ctrl_watch_cb>, data=0x591306143eb0) at src/shared/queue.c:85
#1  0x000059130517dfef in bt_uhid_record (uhid=uhid@entry=0x5913061183e0, input=input@entry=false, ev=ev@entry=0x7ffcde473de0) at src/shared/uhid.c:129
#2  0x000059130517e12d in uhid_read_handler (io=<optimized out>, user_data=0x5913061183e0) at src/shared/uhid.c:158
#3  0x00005913051a6a65 in watch_callback (channel=<optimized out>, cond=<optimized out>, user_data=<optimized out>) at src/shared/io-glib.c:157
#4  0x000075aca7f89199 in g_main_dispatch (context=0x5913060f8800) at ../glib/glib/gmain.c:3344
#5  0x000075aca7fe83bf in g_main_context_dispatch_unlocked (context=0x5913060f8800) at ../glib/glib/gmain.c:4152
#6  g_main_context_iterate_unlocked.isra.0 (context=0x5913060f8800, block=block@entry=1, dispatch=dispatch@entry=1, self=<optimized out>) at ../glib/glib/gmain.c:4217
#7  0x000075aca7f89dc7 in g_main_loop_run (loop=0x5913060f89f0) at ../glib/glib/gmain.c:4419
#8  0x00005913051a70f5 in mainloop_run () at src/shared/mainloop-glib.c:66
#9  0x00005913051a754c in mainloop_run_with_signal (func=func@entry=0x59130511bfa0 <signal_callback>, user_data=user_data@entry=0x0) at src/shared/mainloop-notify.c:188
#10 0x00005913050d26d5 in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:1456
```
</details>

Vudentz, 2024-04-15:
@serfreeman1337 any chance to test the changes above?

closed 2024-04-15 by BluezTestBot

serfreeman1337, 2024-04-16:
Looks good.  bluetoothd no longer crashes