bluez/bluez #818: No CPE entries in NIST NVD since 5.63

wisteso, 2024-04-16:
Searching for bluez 5.6* (wildcard) shows that BlueZ has not been added to the NIST CVE database since 3.63
https://nvd.nist.gov/products/cpe/search/results?namingFormat=2.3&keyword=cpe%3A2.3%3Aa%3Abluez%3Abluez%3A5.6*

NIST is critical for informing products of important security vulnerabilities through the CVE system.  Without proper entries in NIST, end-users cannot be reliably informed of critical security issues.  BlueZ is a highly used software component that is exposed outside of end-user devices.

Fix is to add missing versions to the NIST database.

Vudentz, 2024-04-16:
@wisteso Ive don't think we have been doing anything in this regard, so it might be something the downstream has been doing, that said I don't oppose to have some sort of automation that updates this database. How does it works for things like the Linux kernel?

wisteso, 2024-04-19:
@Vudentz from what I've heard from colleagues that work in the vulnerability management space, the data that goes into NVD can't just be submitted by any user.  It would need to be someone from the organization that maintains the product.

https://nvd.nist.gov/products/cpe
https://nvd.nist.gov/developers

I suppose it is possible that someone downstream (OS level,  perhaps) was entering the new versions.    The link provided above does speak about an API based solution to submitting new data, so your idea of automation may work.

closed 2025-07-27 by github-actions[bot]