bluez/bluez #819: How to display GATT services correctly?

TIANJIANGYE, 2024-04-17:
Hello,I want to create a GATT server on my development board and use it as a peripheral device, referring to the implementation in the peripheral section. However, I don't want to use static MAC and names, but I will still use the original configuration of the Bluetooth adapter. I am unable to correctly discover my customized services on a Bluetooth 5.1 phone. How should I solve this problem?
The service I want to showcase is 9000, but the services that the Bluetooth debugging assistant has been seeing on my phone are 1800 and 1801.

TIANJIANGYE, 2024-04-17:
[main.txt](https://github.com/bluez/bluez/files/15004584/main.txt)

Vudentz, 2024-04-17:
@TIANJIANGYE you are not supposed to copy the code like that to create your own services, just use bluetoothctl> gatt.register* commands to register the services, that way you just use the D-Bus APIs to generate the service, the alternative method is to have it as a plugin to be loaded by bluetoothd but that is normally only required for services that require extra permission or deal with sensitive data like HID. If you are looking into develop your own GATT server then you are in the wrong place, BlueZ is only meant for Linux systems running bluetoothd anything else you are in your own.

closed 2024-04-17 by Vudentz