bluez/bluez #821: example-gatt-server: "Failed to create entry in database"

valgusk, 2024-04-18:
Starting with 5.74 [gatt server example ](https://github.com/bluez/bluez/blob/master/test/example-gatt-server) seems to fail.
Is this example not updated to reflect some new changes?

journal contains:
```
Apr 18 20:49:03 arch-linus bluetoothd[736]: src/gatt-database.c:database_add_chrc() Failed to create characteristic entry in database
Apr 18 20:49:03 arch-linus bluetoothd[736]: src/gatt-database.c:database_add_service() Failed to add characteristic
Apr 18 20:49:03 arch-linus bluetoothd[736]: src/gatt-database.c:database_add_app() Failed to add service
Apr 18 20:49:03 arch-linus bluetoothd[736]: src/gatt-database.c:client_ready_cb() Failed to create GATT service entry in local database
```

99linesofcode, 2024-04-19:
[image]

```
Apr 19 13:18:15 archlinux bluetoothd[2301]: Failed to add UUID: Failed (0x03)
Apr 19 13:18:15 archlinux bluetoothd[2301]: Failed to add UUID: Failed (0x03)
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/ldac
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSink/aptx_hd
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/aptx_hd
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSink/aptx
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/aptx
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSink/aac
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/aac
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSink/sbc
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/sbc
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSink/sbc_xq
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/sbc_xq
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/aptx_ll_1
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/aptx_ll_0
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/faststream
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/faststream_duplex
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSink/opus_05
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/opus_05
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSink/opus_05_duplex
Apr 19 13:18:15 archlinux bluetoothd[2301]: Endpoint registered: sender=:1.19 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Apr 19 13:18:15 archlinux bluetoothd[2301]: Failed to add UUID: Failed (0x03)
Apr 19 13:18:15 archlinux bluetoothd[2301]: Failed to add UUID: Failed (0x03)
Apr 19 13:18:15 archlinux bluetoothd[2301]: src/gatt-database.c:database_add_chrc() Failed to create characteristic entry in database
Apr 19 13:18:15 archlinux bluetoothd[2301]: src/gatt-database.c:database_add_service() Failed to add characteristic
Apr 19 13:18:15 archlinux bluetoothd[2301]: src/gatt-database.c:database_add_app() Failed to add service
Apr 19 13:18:15 archlinux bluetoothd[2301]: src/gatt-database.c:client_ready_cb() Failed to create GATT service entry in local database
```

cy6ernaut, 2024-04-22:
Here too. I'm on Fedora 40

DavidNorena, 2024-04-23:
could that reason I randomly lose connection of my Bluetooth devices ?

Thanks !

hdm9527, 2024-04-24:
same problem

edt-xx, 2024-04-26:
also happening on arch

aykevl, 2024-04-27:
Also found on Fedora 39 with BlueZ 5.75. See: https://github.com/tinygo-org/bluetooth/issues/265#issuecomment-2080417310

VilotStar, 2024-04-28:
[image]
Issue found on 5.75-1 arch linux

Girtholomew, 2024-05-01:
I'm trying to setup a gatt server using Bluez, but it just doesn't work. all ive gotten is tons of errors, now im faced with this error as well. is there any way to create a custom ble service and advertise it? maybe im just dumb and can't figure it out. Please if anyone has a solution for this, let me know asap

zavorra, 2024-05-03:
Same on Fedora 40, bluez bluez-5.75-1 

I am trying to use a ble_midi controller. I can connect it within the bluetooth applet of KDE Plasma 6 and it is shown on qjackctl / qpwgraph as a jack midi port (both in and out), but as I try to open it with python (or wathever other software) it disappear and I get this error on journald

```
May 03 10:17:28 Ulmo bluetoothd[38839]: src/gatt-database.c:database_add_chrc() Failed to create characteristic entry in database
May 03 10:17:28 Ulmo bluetoothd[38839]: src/gatt-database.c:database_add_service() Failed to add characteristic
May 03 10:17:28 Ulmo bluetoothd[38839]: src/gatt-database.c:database_add_app() Failed to add service
May 03 10:17:28 Ulmo bluetoothd[38839]: src/gatt-database.c:client_ready_cb() Failed to create GATT service entry in local database

```

ssorensenroku, 2024-05-03:
@Vudentz This is caused by 7604a577c9d7bf15c9144b8154842ce277afec90 - registering a GATT characteristic no longer works when the `Handle` property is missing or set to 0.

aykevl, 2024-05-03:
I tried setting `Handle` to 0 but it didn't seem to help (same error). Is there any workaround that we can use in the meantime?

Vudentz, 2024-05-03:
@aykevl how about the above change?

ricolxwz, 2024-05-04:
+1 on fedora40, when I download things in Chrome, the connection always drops.

ssorensenroku, 2024-05-06:
@Vudentz this changes fixes registering GATT characteristics in my application.

Girtholomew, 2024-05-06:
i did try using https://github.com/noble/bleno to setup, but that also resulted in errors so i had to use @abandonware/bleno for my work. i was able to get the test code to execute, but no message shows up saying that it has started to advertise. id like some help, so if anyone would like to lend a hand, my discord is girtholomew420
i seriously need some help

aykevl, 2024-05-06:
> @aykevl how about the above change?

Which above change?
If you mean https://github.com/bluez/bluez/commit/7604a577c9d7bf15c9144b8154842ce277afec90, I don't see how that's supposed to help me? What I need is a workaround for _current_ shipping BlueZ versions, not a patch I can apply on top of BlueZ to fix a bug. (I'm not sure whether that patch is meant as a patch to BlueZ or as a suggestion how to fix my own library).

I've also tried other values than 0, in each case I get the same rather useless error (it would be great if BlueZ reported an actually useful error here, but it doesn't).

Vudentz, 2024-05-06:
@aykevl you can try removing the Handle altogether but I think you might still hit this problem, so either you have to rollback to a working version or wait for 5.76 to be release with the fix.

closed 2024-05-06 by BluezTestBot

aykevl, 2024-05-25:
> you can try removing the Handle altogether but I think you might still hit this problem

Indeed, this was the behavior when I hit this bug initially.

> so either you have to rollback to a working version or wait for 5.76 to be release with the fix.

Not an option for me: I'm targeting the BlueZ version installed on common Linux distros.
But I guess there's no other option than to tell users of the library "wait until a fix is released for your distro".

cy6ernaut, 2024-05-25:
It is resolved now, in Fedora 40 with bluez 5.76.

aykevl, 2024-05-25:
Confirmed fixed in 5.75. Thanks!