bluez/bluez #823: Issue with the interval and latency parameters of bt_iso_qos configuration

vlad-pruteanu, 2024-04-22:
The bluetoothctl command transport.send uses the interval and latency parameters stored in iso_qos.bcast.out (or iso_qos.ucast.out) to determine the number of packets sent and the interval between these packets.
 
The iso_qos.bcast.out.latency field is first set by the Host to contain the Max_Transport_Latency. This value is then used by the kernel as a parameter for the LE Create BIG command.
For the source, the controller sends the LE Create BIG Complete event after the creation of the BIG. This event contains the Transport_Latency_BIG value. The kernel does not use this event to update the iso_qos.bcast.out.latency. iso_qos.bcast.out.latency remains set to Max_Transport_Latency.
 
For the sink, the controller will send the LE BIG Sync Established event which contains the Transport_Latency_BIG value. The kernel sets iso_qos.bcast.in.interval (which holds the SDU Interval) to the latency reported by the event, and then iso_qos.bcast.in.latency is set to the latency reported by the event * 1.25:
```
__le32 interval;
...
memset(&interval, 0, sizeof(interval));
memcpy(&interval, ev->latency, sizeof(ev->latency));
bis->iso_qos.bcast.in.interval = le32_to_cpu(interval);
/* Convert ISO Interval (1.25 ms slots) to latency (ms) */
bis->iso_qos.bcast.in.latency = le16_to_cpu(ev->interval) * 125 / 100;
```

The event reports the latency expressed in us, no additional calculation is necessary. iso_qos.bcast.in.latency should be set to ev->latency, after the conversion to ms.
iso_qos.bcast.in.interval should not be set here since the SDU interval is not reported by this event and BlueZ does not need to know the SDU Interval value on the receiver side.
 
The same applies for unicast, on the LE CIS Established event, the kernel sets both iso_qos.ucast.out.interval and iso_qos.ucast.in.interval to the ISO_Interval (reported by the event) * 1.25. This formula is incorrect, and the Core Spec doesn't mention a formula that directly links the SDU_Interval and ISO_Interval. In any case, for the source, the SDU Interval (iso_qos.ucast.out.interval) has already been set and for the sink (iso_qos.ucast.in.interval) it does not need to be known.

The issue with the out.latency field was observed while working on the transport.send command where it was seen that the value doesn't match the one reported by the controller. The rest of the issues were seen while searching for other such cases.

Vudentz, 2024-04-22:
```
BLUETOOTH CORE SPECIFICATION Version 5.4 | Vol 6, Part G
page 3075:

Transport_Latency_C_To_P = CIG_Sync_Delay + (FT_C_To_P) ×
ISO_Interval + SDU_Interval_C_To_P
Transport_Latency_P_To_C = CIG_Sync_Delay + (FT_P_To_C) ×
ISO_Interval + SDU_Interval_P_To_C

```
So if I'm not mistaken the SDU Interval can be calculated as:

SDU_Interval = Transport_Latency - (CIG_Sync_Delay + (FT) x ISO_Interval)

And for BIS:

```
Transport_Latency_BIG = BIG_Sync_Delay + PTO × (NSE ÷ BN –
IRC) * ISO_Interval + ISO_Interval + SDU_Interval
```

SDU_Interval = Transport_Latency_BIG - (BIG_Sync_Delay + PTO × (NSE ÷ BN –
IRC) * ISO_Interval + ISO_Interval)

In terms of bt_qos_io, the latency field shall represent the Transport_Latancy and interval the SDU_Interval, I hope that makes it clearer.

pv, 2024-05-11:
> BlueZ does not need to know the SDU Interval value on the receiver side

BAP Server may also be a Source (e.g. a microphone peripheral), and in this case the SDU_Interval is not known a priori by BAP Server Host.

As the spec doesn't outright tell how to get SDU_Interval of the Established event parameters (is it clear these formulas are normative statements?), one wonders if it's intended to be known from the upper protocol layer. For BAP it is always known from the codec configuration.

Vudentz, 2024-06-10:
Fixed by https://github.com/bluez/bluetooth-next/commit/9b70755997d372a80928faf90fd5ea4a637b90ac

closed 2024-06-10 by Vudentz