bluez/bluez #824: Switch pro controller and joycons won't pair / connect properly on rtl8852ce

coinkillerl, 2024-04-23:
Switch pro controllers and joycons will not pair correctly on rtl8852ce. If you do the pairing process, the device will be shown as paired, but will disconnect shortly after. After trying to reconnect to it, this error is returned : 
Failed to connect: org.bluez.Error.Failed br-connection-create-socket
If there is a nintendo switch powered on nearby in which the controller was previously paired, pressing any button on the controller will just make the controller reconnect to the switch. The controller still shows up as paired in bluetoothctl and in bluedevil.

Connecting without pairing the device will keep it connected, but it will be unusable and the LEDs on the pro controller / joycon will be stuck in the pairing animation.

MaxBaktBrood, 2024-04-30:
This is a problem with the last update to Bluez specifically. See: https://forums.linuxmint.com/viewtopic.php?t=410356#p2411078

unicornsandcandies, 2024-05-10:
> This is a problem with the last update to Bluez specifically. See: https://forums.linuxmint.com/viewtopic.php?t=410356#p2411078

Will it be solved with the next release? What exactly is the issue here?

Reual4412, 2024-05-21:
I have the same problem, with  RTL8852BE

except worse. connecting does nothing. Pairing does nothing. Dualsense controller works fine. Anything involving Nintendo Switch controllers such as
Switch Pro Controller
NSO N64 controller (with the needed NSO dkms module)
And NSO SNES controller (same as above)
Claim it pairs, fails, doesn't connect.

MaxBaktBrood, 2024-05-21:
I meant the versions in the package manager for Ubuntu. My apologies.

For me, with version 5.64-0ubuntu1, my Pro Controller connects properly. In 5.64-0ubuntu1.1, I can't deliver input anymore.

Reual4412, 2024-05-21:
I've been using arch, so that's the version I'm referring to

coinkillerl, 2024-05-21:
> I have the same problem, with RTL8852BE
> 
> Pairing does nothing. Dualsense controller works fine. Anything involving Nintendo Switch controllers such as Switch Pro Controller NSO N64 controller (with the needed NSO dkms module) And NSO SNES controller (same as above) Claim it pairs, fails, doesn't connect.
> 
> And this issue goes beyond seemingly 5.75. As it's broken in 5.72-2 as well.

Does it pair according to bluetoothctl? it does for me, but using KDE's bluetooth applet, it actually says the pairing has failed.  Actual result doesn't differ between trying to pair via GUI or the command line. Also, in the very short time it's connected, input doesn't work, just specifying.

nfp0, 2024-07-10:
I'm having the same issue, on a RTL8852AE.
My Nintendo Switch controllers (Pro, SNES, N64, etc) fail to pair through booth `bluetoothctl` and the KDE interface.

The only way to sometimes get it working is doing `connect` through `bluetoothctl` without pairing. But this is very inconsistent. It sometimes works, and others it doesn't.
It only works when `bluetoothctl` shows this when doing `connect`:

```
[Pro Controller]# [CHG] Device 64:B5:C6:41:CB:1F ServicesResolved: yes
```

Otherwise, it looks connected but no input works and it eventually disconnects.
I can't find a pattern.

Trying to pair results in this output:
```
[CHG] Device 64:B5:C6:41:CB:1F Connected: yes
[Pro Controller]# [CHG] Device 64:B5:C6:41:CB:1F Bonded: yes
[Pro Controller]# [CHG] Device 64:B5:C6:41:CB:1F Paired: yes
[Pro Controller]# Pairing successful
[CHG] Device 64:B5:C6:41:CB:1F Connected: no
```
It pairs, but then the controller disconnects and is unusable.

This is on Arch on BlueZ 5.77 released yesterday.
On my desktop on the same version, with a TP-Link USB bluetooth dongle, all controllers work perfectly specially with `FastConnectable = true` to make reconnection always work.

queler, 2024-07-10:
See https://github.com/bluez/bluez/commit/25a471a83e02e1effb15d5a488b3f0085eaeb675

There are some other issues think it's a bug that this change exacerbates, but I have no idea. If you want it to work, security be damned, add `ClassicBondedOnly=false` to /etc/bluetooth/input.conf
You might be able to get away with just restarting bluez and the Bluetooth service, but I've had luck with a restart. Just don't forget to re modprobe hid-nintendo

Ymmv but I've had the most luck searching, double-clicking, and then trusting after its connected.

nfp0, 2024-07-21:
@queler I was already using `ClassicBondedOnly=false` and it does not seem to fix it.

One strange behavior is that the Pro controller pairs successfully, but then it never connects  (hid-nintendo doesn't even get loaded), it just disconnects immediately when trying to connect or reconnect.
On the PS4 controller, it pairs, but disconnects right away, BUT subsequent reconnects work.

These behaviors make me feel like this is a timing problem.
Both controllers pair and connect perfectly fine on my TP-Link USB dongle, so perhaps it might be a driver problem on the Realtek side? I know for a fact that this RTL8852AE card also has WiFi problems on some Thinkpad laptops, mine included (P14s gen 2)

> Ymmv but I've had the most luck searching, double-clicking, and then trusting after its connected.

On KDE it trusts the device when you double-click on the pairing window.

teinturman, 2024-07-24:
hi, i have the same issue on my Wiiu Pro Controller : 
1) i have added the options : no security and ClassicBondedOnly=false
2) if i ensure that there is no paired device by removing the ones that are paired, then i connect : i  am able to connect to the controller;
but if i pair first ( i get a successfull message ) .then , the same connect method fails and i cannot connect. ( i get a prompt for codepin, instead of being connected silently ) .

This was working in previous versions ( i do not have the exact version number, but i was using Bullseyes version in raspberry pi os. Since the latest os version which includes  

	/* encrypt connection if device is bonded */
	if (input_device_bonded(idev))
		sec_level = BT_IO_SEC_MEDIUM;
	else
		sec_level = BT_IO_SEC_LOW;

Reual4412, 2024-08-11:
> One strange behavior is that the Pro controller pairs successfully, but then it never connects (hid-nintendo doesn't even get loaded), it just disconnects immediately when trying to connect or reconnect.
> On the PS4 controller, it pairs, but disconnects right away, BUT subsequent reconnects work.

This  happens to my dualsense controller. as of me retesting this problem On plasma (latest version on arch as of the time writing this), I get a result saying the wizard failed, but added to the list of devices. When turning the controller back on (as the controller immediately shuts off after paring) it works just fine. 

Which heavily implies to me at least this might be an issue with the driver, which means that to get it fixed we'll likely have to move over to another github.

nfp0, 2024-08-11:
I have changed my laptop's wi-fi/bluetooth card from the Realtek RTL8852AE to an Intel Dual Band Wireless-AC 8265.
All my Pro Controller pairing and connection problems went away immediately.

Considering it also works fine on my USB Bluetooth dongle and on my other PC, I guess I have to assume that the Bluetooth on that Realtek's card is buggy.

Reual4412, 2024-08-14:
I have manually downgraded bluez to some of the oldest (while remaining compatibleish) libraries possible. On version 5.55-3 the switch pro controller works just fine as intended. Connecting correctly via connect.

HOWEVER, pairing is still broken.

PS5 controller (dualsense) still abruptly shuts off when being paired, and requires a second turn on. While the pro controller refuses to pair what so ever.

Reual4412, 2024-08-14:
I can now confirm that the nintendo switch online controllers, ( at least the ones I have, snes, and n64) also exhibits the same behavior. Connect functions, pairing breaks.

The only other notable behavior is what happens when multiple controllers are connected, in which it doesn't quite work. But I'm going to chock that up to using a old package, and hopefully something fixed in later releases.

Reual4412, 2024-08-15:
I've deleted two of my posts. Since on latest version changing settings in input.conf, changing UserspaceHID to false. And ClassicBondedOnly to false. Allows the pro controller to connect. Pairing however, is still broken

Sylverena, 2024-09-27:
Would like to chime in here and say that this issue is also affecting me with a switch pro controller and a realtek wireless card on my laptop. Anecdotally speaking,  before I switched distros to arch months ago, I was using pop!_os which might be a few versions behind. At this time I think it is unclear when exactly this issue was introduced, however.

Generally speaking, when connecting after attempting to pair it looks like this:

```
connect DC:CD:18:A9:37:9B
Attempting to connect to DC:CD:18:A9:37:9B
[CHG] Device DC:CD:18:A9:37:9B Connected: yes
[Pro Controller]# Failed to connect: org.bluez.Error.Failed br-connection-create-socket
[CHG] Device DC:CD:18:A9:37:9B Connected: no
```

I made a post on this on the Arch forums, and a workaround was presented to me: 
Forget device -> trust MAC -> pair MAC -> connect MAC

This workaround I can confirm works for me, but does take a few tries.

infidelus, 2024-10-20:
Just to add to this after trying @Sylverena's workaround,  I tried that and it didn't work but instead just did scan on and then connect MAC and it worked and allowed me to play a game.  After it goes to sleep and disconnects I did run into the same problem again though so it is only a workaround.

sickdaflip, 2024-11-03:
For me the same thing with rtl8852be no matter which controller 8bitdo, PS4 etc. pairing and trust works, connect doesn't work. When will this finally be fixed, please! 🥲

bootsector, 2025-01-18:
I'm having similar issues when re-connecting a Nintendo Switch Online NES Controller to a MiSTer FPGA:

bootsector, 2025-01-18:
[btmon dumps.zip](https://github.com/user-attachments/files/18465858/btmon.dumps.zip)

Attaching the btmon logs

Sylverena, 2025-02-18:
> I made a post on this on the Arch forums, and a workaround was presented to me: Forget device -> trust MAC -> pair MAC -> connect MAC
> 
> This workaround I can confirm works for me, but does take a few tries.

Update on the workaround I presented: you need game-devices-udev to use this workaround. I tried the workaround on a new, different machine running 6.13.1-arch1-1 and I was getting br-connection-profile-unavailable until I realized my laptop had this package and my desktop didn't. You can find game-devices-udev at https://codeberg.org/fabiscafe/game-devices-udev , but I installed it through an AUR package.

game-devices-udev makes these devices available to user-grade permissions, perhaps the issues lies somewhere with bluez trying to access device files that the user running the daemon is not able to access? I am not fully educated on this however, so this is pure speculation. Either way, this definitely seems to be a multifaceted issue.

BeeFox-sys, 2025-03-24:
from what i understand, joycons only remember one pairing, so they need to be re-paired each time they connect if they connected to another device (such as the switch) between connections. my current work around (that is sadly inconsitent right now) is a bash script specifically removing and then reconnecting my joycons so they are re-paired each time i run the script.

ChrisAtJAM, 2025-05-14:
I'll need to check if this is still an issue. On CachyOS.

1337haXXor, 2025-05-16:
This is still an issue with me, as well. UB500 BT Adapter, and with two different controllers; a GuliKit KK2 (NS09) and an 8BitDo Pro 2. Initial pairing and connection works, but after that, it never connects again without fully removing and unpairing it.

herrzettpunkt, 2025-06-23:
Same here on ChimeraOS. XBOX and Stadia controller works just fine after reconnecting - even after reboot the system. Switch Pro Controller fails to reconnect once system is rebooted.

doktormerlin, 2025-06-23:
Nobara in the meantime updated to bluez 5.81 I think, I am now able to pair the controller. However, irregularly but often (every 20s to 2 minutes) the controller disconnects. Playing via bluetooth stays completely impossible. Even though the controller is paired, it doesnt show up in bluetoothctl for some reason

nfp0, 2025-08-03:
Just to reiterate, for the people that can pair, but cannot reconnect afterwards, try using `FastConnectable = true`

@1337haXXor @herrzettpunkt

Sylverena, 2025-08-08:
I did get a chance to try this just now, and I still am not able to connect. This time, I'm getting:

```
Failed to connect: org.bluez.Error.Failed br-connection-create-socket
[SIGNAL] org.bluez.Device1.Disconnected org.bluez.Reason.Local Connection terminated by local host
```

The `br-connection-create-socket` error has been showing up every single time in the many months I have been experiencing this issue. At this rate, I'm wondering if there's something going on with realtek bluetooth cards and bluez not agreeing with each other

1337haXXor, 2025-08-13:
I'm just using a TP-Link USB one, but looking around it seems like all USB ones use RealTek? So is my only option a PCIe/M.2 thing? I have an internal one on my MoBo, an Asus TUF Z690 WiFi, which seems to have the AX200, but stopped using it because it would constantly disconnect. I assumed that was because the computer was a bit far from my controller, though I was only ~8 feet. Would a dedicated one like an Intel Wireless AC 7260 work (or work better?).

nfp0, 2025-08-14:
@1337haXXor Hmm. I am not sure that would help, because I have a TP-Link UB500 (same as you, it seems), and it's been working flawlessly for me on Arch Linux with a bluetooth mouse and multiple Nintendo controllers.

1337haXXor, 2025-08-15:
Sheesh, I don't understand.. I have some XBox controllers, a Nintendo one, and a Gulikit. And I've tried an 8BitDo Pro 2. All of them disconnect constantly or immediately, except the XBox controllers, which "only" disconnect every 5 or 10 minutes. >.>
I just did the FastConnectable change, I'll test and see if that works.

nfp0, 2025-08-15:
@1337haXXor That is really strange, considering that TP-Link UB500 dongle has never failed for me after setting `FastConnectable = true`. What distro and Kernel version are you on?
And also, do you have the chance to test the TP-Link UB500 on another computer? Could be a faulty USB hub perhaps, or the even the dongle itself.

nfp0, 2025-08-19:
@1337haXXor I am out of ideas for your specific case.
I think what I would do is try the dongle on another Linux distro perhaps, and see the result.

berrydiaboli, 2025-11-01:
still an issue, I have this 
`bluetoothd[762]: No matching connection for device`