bluez/bluez #827: GATT server: AcquireWrite error path not handled properly

beku-epitome, 2024-04-26:
Follow-up on #820

My application rejects AcquireWrite calls if there's no corresponding AcquireNotify call done beforehand. This is done to ensure that a response channel is available when the first request comes in, and a requirement of ACS control point. The ACS spec requires returning 0xFD ATT error code, which is BT_ERROR_CCC_IMPROPERLY_CONFIGURED in bluez terms, however, `dbus_error_to_att_ecode` doesn't accept this error code, and bluez proceeds to try writing through WriteValue DBus API instead (which is not implemented on this CP characteristic's DBus bindings). On the client device I get a success returned on the write request.

If I change the error code to something that is accepted in `dbus_error_to_att_ecode`, I still don't get that error back on the client device. When I try the second time consecutively, the write request doesn't even reach my application, bluez and the phone just drop the connection instead.

So in general, the error path of the AcquireWrite call seems to be faulty and in need of review/redesign.

Vudentz, 2024-04-26:
@beku-epitome could you please test the above changes?

closed 2024-04-29 by BluezTestBot

beku-epitome, 2024-04-30:
That was fast. Too fast in fact. I left a comment [here](https://github.com/BluezTestBot/bluez/commit/6573090b5a208eb9bb51bed6a054709fa665a6c7#r141462877) which was not noticed. I will also test the change on Thursday, but the issue I reported is about more than just adding support to this error code, see the second paragraph.

Vudentz, 2024-04-30:
@beku-epitome https://github.com/bluez/bluez/commit/7e6079219b2218f4c3f52521f4a2c521049d961a and https://github.com/bluez/bluez/commit/6fc599e334cead9cbd014f7830753bc28ee5868b

beku-epitome, 2024-05-02:
One more thing related to this, it just hit me while going through the function again, I believe there is another comparison logic swap here, shouldn't it be `*endptr == '\0'`?
https://github.com/bluez/bluez/blob/6fc599e334cead9cbd014f7830753bc28ee5868b/src/gatt-database.c#L2224-L2226