bluez/bluez #828: Bluez does not with Ubuntu 24.04

bogdan-mitrea-ds, 2024-04-26:
Bluez library is not able to connect to a USB bluetooh dongle and it throws and error.

Steps to reproduce the issue:
* connect the USB bluetooth dongle to a Ubuntu 24.04 machine 
* install bluez package 
* Start bluetoothctl and make the BT dongle discoverable using `discoverable on`
* Find the machine via regular BT manager available on an  Android phone
* Device is discovered 
* Tap on the screen and try to connect
* Confirm the passkey
* Device gets connected and imediately gets disconnected
* Error message `hci0 74:9E:F5:28:3C:4C type BR/EDR disconnected with reason 2`

Bluetoothctl  logs
`
 bluetoothctl
Waiting to connect to bluetoothd...[bluetooth]# hci0 new_settings: powered connectable discoverable bondable ssp br/edr le secure-conn
[bluetooth]# Agent registered
[bluetooth]# [CHG] Controller 00:1A:7D:10:73:63 Pairable: yes
[bluetooth]# show
Controller 00:1A:7D:10:73:63 (public)
        Manufacturer: 0x005d (93)
        Version: 0x0a (10)
        Name: bomi-edge
        Alias: bomi-edge
        Class: 0x0000010c (268)
        Powered: yes
        Discoverable: yes
        DiscoverableTimeout: 0x000000b4 (180)
        Pairable: yes
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0548
        Discovering: no
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x04 (4)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
        SupportedSecondaryChannels: 1M
        SupportedSecondaryChannels: 2M
        SupportedSecondaryChannels: Coded
[bluetooth]# discoverable on
[bluetooth]# Changing discoverable on succeeded
[bluetooth]# hci0 74:9E:F5:28:3C:4C type BR/EDR connected eir_len 15
[CHG] Device 74:9E:F5:28:3C:4C Connected: yes
[C's S10+]# Request confirmation
[C's S10+]#   1;39m[agent] Confirm passkey 520835 (yes/no): yes
[C's S10+]# hci0 new_link_key 74:9E:F5:28:3C:4C type 0x05 pin_len 0 store_hint 1
[C's S10+]# hci0 74:9E:F5:28:3C:4C type BR/EDR disconnected with reason 2
[CHG] Device 74:9E:F5:28:3C:4C Connected: no
`
hciconfig details
`
hciconfig hci0
hci0:   Type: Primary  Bus: USB
        BD Address: 00:1A:7D:10:73:63  ACL MTU: 1021:6  SCO MTU: 255:12
        UP RUNNING PSCAN
        RX bytes:5631 acl:36 sco:0 events:297 errors:0
        TX bytes:35701 acl:36 sco:0 commands:233 errors:0
`
bluez logs
`
Apr 18 09:22:36 bomi-edge bluetoothd[2413]: src/adapter.c:set_mode() sending set mode command for index 0
Apr 18 09:22:36 bomi-edge bluetoothd[2413]: src/adapter.c:property_set_mode() sending Set Discoverable command for index 0
Apr 18 09:22:36 bomi-edge bluetoothd[2413]: src/shared/mgmt.c:send_request() [0x0000] command 0x0007
Apr 18 09:22:36 bomi-edge bluetoothd[2413]: src/shared/mgmt.c:can_read_data() [0x0000] command 0x0007 complete: 0x00
Apr 18 09:22:36 bomi-edge bluetoothd[2413]: src/shared/mgmt.c:send_request() [0x0000] command 0x0006
Apr 18 09:22:37 bomi-edge bluetoothd[2413]: src/shared/mgmt.c:can_read_data() [0x0000] command 0x0006 complete: 0x00
Apr 18 09:22:37 bomi-edge bluetoothd[2413]: src/adapter.c:property_set_mode_complete() Success (0x00)
Apr 18 09:22:37 bomi-edge bluetoothd[2413]: src/adapter.c:new_settings_callback() Settings: 0x00000adb
Apr 18 09:22:37 bomi-edge bluetoothd[2413]: src/adapter.c:settings_changed() Changed settings: 0x00000008
Apr 18 09:22:37 bomi-edge bluetoothd[2413]: src/adapter.c:settings_changed() Pending settings: 0x00000002
Apr 18 09:23:07 bomi-edge bluetoothd[2413]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000b
Apr 18 09:23:07 bomi-edge bluetoothd[2413]: src/adapter.c:connected_callback() hci0 device 74:9E:F5:28:3C:4C connected eir_len 15
Apr 18 09:23:07 bomi-edge bluetoothd[2413]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000f
Apr 18 09:23:07 bomi-edge bluetoothd[2413]: src/adapter.c:user_confirm_request_callback() hci0 74:9E:F5:28:3C:4C confirm_hint 0
Apr 18 09:23:07 bomi-edge bluetoothd[2413]: src/device.c:new_auth() Requesting agent authentication for 74:9E:F5:28:3C:4C
Apr 18 09:23:07 bomi-edge bluetoothd[2413]: src/agent.c:agent_ref() 0x5e96f8c12780: ref=2
Apr 18 09:23:07 bomi-edge bluetoothd[2413]: src/agent.c:agent_request_confirmation() Calling Agent.RequestConfirmation: name=:1.49, path=/org/bluez/agent, passkey=639324
Apr 18 09:23:14 bomi-edge bluetoothd[2413]: src/agent.c:agent_ref() 0x5e96f8c12780: ref=3
Apr 18 09:23:14 bomi-edge bluetoothd[2413]: src/adapter.c:btd_adapter_confirm_reply() hci0 addr 74:9E:F5:28:3C:4C success 1
Apr 18 09:23:14 bomi-edge bluetoothd[2413]: src/agent.c:agent_unref() 0x5e96f8c12780: ref=2
Apr 18 09:23:14 bomi-edge bluetoothd[2413]: src/agent.c:agent_unref() 0x5e96f8c12780: ref=1
Apr 18 09:23:14 bomi-edge bluetoothd[2413]: src/shared/mgmt.c:send_request() [0x0000] command 0x001c
Apr 18 09:23:14 bomi-edge bluetoothd[2413]: src/shared/mgmt.c:can_read_data() [0x0000] command 0x001c complete: 0x00
Apr 18 09:23:14 bomi-edge bluetoothd[2413]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x0009
Apr 18 09:23:14 bomi-edge bluetoothd[2413]: src/adapter.c:new_link_key_callback() hci0 new key for 74:9E:F5:28:3C:4C type 5 pin_len 0 store_hint 1
Apr 18 09:23:14 bomi-edge bluetoothd[2413]: src/device.c:device_bonding_complete() bonding (nil) status 0x00
Apr 18 09:23:14 bomi-edge bluetoothd[2413]: src/adapter.c:resume_discovery()
Apr 18 09:23:19 bomi-edge bluetoothd[2413]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000c
Apr 18 09:23:19 bomi-edge bluetoothd[2413]: src/adapter.c:dev_disconnected() Device 74:9E:F5:28:3C:4C disconnected, reason 3
Apr 18 09:23:19 bomi-edge bluetoothd[2413]: src/adapter.c:adapter_remove_connection()
Apr 18 09:23:19 bomi-edge bluetoothd[2413]: plugins/policy.c:disconnect_cb() reason 3
Apr 18 09:23:19 bomi-edge bluetoothd[2413]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 74:9E:F5:28:3C:4C type 0 status 0xe
Apr 18 09:23:19 bomi-edge bluetoothd[2413]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Apr 18 09:23:19 bomi-edge bluetoothd[2413]: src/device.c:device_bonding_failed() status 14
Apr 18 09:23:19 bomi-edge bluetoothd[2413]: src/adapter.c:resume_discovery()
`

Vudentz, 2024-04-26:
> Bluez library is not able to connect to a USB bluetooh dongle and it throws and error.
>
> Steps to reproduce the issue:
>
>     * connect the USB bluetooth dongle to a Ubuntu 24.04 machine
>
>     * install [...]

Looks like these logs were not pasted properly, they seem to be pasted as one-liners when in fact they are not.

Vudentz, 2024-04-26:
@bogdan-mitrea-ds please add the HCI traces with use of btmon.

bogdan-mitrea-ds, 2024-04-26:
> > Bluez library is not able to connect to a USB bluetooh dongle and it throws and error.
> > Steps to reproduce the issue:
> > ```
> > * connect the USB bluetooth dongle to a Ubuntu 24.04 machine
> > 
> > * install bluez package
> > 
> > * Start bluetoothctl and make the BT dongle discoverable using `discoverable on`
> > 
> > * Find the machine via regular BT manager available on an  Android phone
> > 
> > * Device is discovered
> > 
> > * Tap on the screen and try to connect
> > 
> > * Confirm the passkey
> > 
> > * Device gets connected and imediately gets disconnected
> > 
> > * Error message `hci0 74:9E:F5:28:3C:4C type BR/EDR disconnected with reason 2`
> > ```
> 
> Looks like these logs were not pasted properly, they seem to be pasted as one-liners when in fact they are not. (Fixed it myself)

thanks

bogdan-mitrea-ds, 2024-05-03:
I am getting a btmon error when trying to dump the traces. i have updated the description

Vudentz, 2024-05-03:
> I am getting a btmon error when trying to dump the traces. i have updated the description

Hmm we might need to fix it, actually it might be that 5.72 has a problem any chance to update it to 5.75?

bogdan-mitrea-ds, 2024-05-10:
[btmon_traces_dump.log](https://github.com/bluez/bluez/files/15271211/btmon_traces_dump.log)
@Vudentz i have managed to dump he traces to a file and attached to the issue. I used the same 5.72 version for btmon.

bogdan-mitrea-ds, 2024-05-14:
any chance to check the logs @Vudentz

giner, 2024-07-15:
I've compiled a package from Ubuntu 24.10 sources, installed on my Ubuntu 24.04 and 22.04 machines and all bluetooth connectivities issues have gone. I'm not sure what the maintenaners plan to do as package versions have been already freezed for 24.04.

See also:
- https://bugs.launchpad.net/ubuntu/+source/gnome-bluetooth/+bug/2071415
- https://bugs.launchpad.net/ubuntu/+source/bluez/+bug/2064783

ByerRA, 2024-07-18:
Having the same issue.  My machine never had an issue with bluetooth with 22.04 but with 24.04 I can pair bluetooth speakers just fine and system sounds do go through the speakers, but anything else like a YouTube page in Chrome, no sound.

wasim7raja10, 2024-08-07:
[image]
Suddenly the Bluetooth disappears after disconnecting a device or during sleep.

2opremio, 2024-09-29:
I am also getting the btmon core dump in 5.72 (also in Ubuntu 24.0.4). Here is the core dump:

[_usr_bin_btmon.0.crash.gz](https://github.com/user-attachments/files/17180873/_usr_bin_btmon.0.crash.gz)

Otherwise the system works fine with my peripherals.

2opremio, 2024-09-29:
I am getting the same error when compiling btmon 5.78 from sources:

```
sudo ./btmon 
Bluetooth monitor ver 5.78
*** buffer overflow detected ***: terminated
Aborted
```

RahulSinghGulia, 2024-11-08:
For me, this post worked perfectly: https://akyriako.medium.com/solve-bluetooth-problems-on-ubuntu-377d7419c839

giner, 2024-11-27:
I've created a PPA with a Bluez backport from Ubuntu 25.04 to be used until the issue if fixed in 24.04 upstream https://launchpad.net/~giner/+archive/ubuntu/bluez