bluez/bluez #831: DBus connected property changed to false but bluetooth connection still valid

david-bouthillier, 2024-04-30:
I have a strange situation where I receive a `PropertiesChanged` event for the 'connected' property on `/org/bluez/hci0/dev_A0_56_F3_97_94_4C`, interface `org.bluez.MediaControl1`, which gives me the value of `false`. When this occurs, the media player (spotify) pauses the song on my phone (although I think the song pauses and then I get the property changed event). When I look at the bluetooth connection screen on my phone, it says I'm still connected. When I inspect the embedded device with `bluetoothctl`, it also says the phone is still connected. And when I use `gdbus` to introspect the object, it also says `connected = true`. So what's going on here?

Here's the basic flow. I'm using DBus API calls in C

- Call StartDiscovery on the bluetooth adapter
- Call Connect with device address (address to my phone)
- Subscribe to PropertiesChanged on the phone device address
- Use PulseAudio loopback to stream audio from phone to my embedded device
- Unload PulseAudio loopback to stop audio streaming
- Use PulseAudio loopbacks to perform a hands free phone call
- Unload PulseAudio loopbacks after call is finished
- Load PulseAudio loopback for streaming audio again

After a random amount of time (usually a few minutes), I'll get the properties changed callback and the audio will pause on my phone.

Environment:
```
bluez v5.72
pulseaudio v16.1
```

Phone:
```
iPhone 8
Software v15.5
```

Toradex Embedded device info:
```
Bootloader:               U-Boot
Kernel version:           5.15.129-6.4.0-devel+git.67c3153d20ff #1 SMP PREEMPT Wed Sep 27 12:30:36 UTC 2023
Kernel command line:      root=PARTUUID=8ed95557-02 ro rootwait console=tty1 console=ttymxc2,115200 consoleblank=0 earlycon
Distro name:              NAME="TDX Wayland with XWayland"
Distro version:           VERSION_ID=6.4.0-devel-20231220205908-build.0
Hostname:                 verdin-imx8mp-15006329
```

Vudentz, 2024-04-30:
> I have a strange situation where I receive a `PropertiesChanged` event for the 'connected' property on `/org/bluez/hci0/dev_A0_56_F3_97_94_4C`, interface `org.bluez.MediaControl1`, which gives me [...]

that usually means that L2CAP AVCTP/AVRCP channel is disconnected while the ACL Links is still connected, so I guess AVRCP is disconnecting for some reason which is probably not intended, or perhaps you didn't ask it to be connected? Sometimes this can happens if A2DP is disconnected as well since AVRCP and A2DP are kind tied together in some implementations, so perhaps A2DP is the problem.

david-bouthillier, 2024-04-30:
@Vudentz is there something that can be done to fix the issue?

Vudentz, 2024-04-30:
> @Vudentz is there something that can be done to fix the issue?

Try collecting the HCI trace with btmon so we can see what is going on over there.

david-bouthillier, 2024-05-02:
@Vudentz I ran btmon and I believe I have captured the failure. For some unknown reason, I get a Connection Timeout event. Any idea what could be causing this?

```
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                       #5583 [hci0] 63.547751
        Status: Success (0x00)
        Handle: 1
        Reason: Connection Timeout (0x08)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                 {0x0001} [hci0] 63.547793
        BR/EDR Address: A0:56:F3:97:94:4C (Apple, Inc.)
        Reason: Connection timeout (0x01)
= bluetoothd: ../../../../../../workspace/sources/bluez5/src/profile.c:ext_io_disconnected() Unable to get io data for hfp_hf: getpeername: Transport endpoint is not connected (107)             63.555730
> HCI Event: Vendor (0xff) plen 67                                                                                                                                                   #5584 [hci0] 63.547762
        72 01 00 4e 0d 00 00 72 0d 00 00 84 0c 00 00 11  r..N...r........
        0c 00 00 ef 0b 00 00 d8 b9 0c 00 e4 b9 0c 00 8c  ................
        d9 0c 00 e4 c1 0c 00 f0 c0 0c 00 e4 b9 0c 00 b4  ................
        b9 0c 00 74 d9 0c 00 d8 c1 0c 00 14 c1 0c 00 01  ...t............
        02 00 00                                         ...             
< HCI Command: Write Scan Enable (0x03|0x001a) plen 1                                                                                                                                #5585 [hci0] 63.580150
        Scan enable: Page Scan (0x02)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                          #5586 [hci0] 63.581142
      Write Scan Enable (0x03|0x001a) ncmd 1
        Status: Success (0x00)
< HCI Command: Create Connection (0x01|0x0005) plen 13                                                                                                                               #5587 [hci0] 65.269908
        Address: A0:56:F3:97:94:4C (Apple, Inc.)
        Packet type: 0xcc18
          DM1 may be used
          DH1 may be used
          DM3 may be used
          DH3 may be used
          DM5 may be used
          DH5 may be used
        Page scan repetition mode: R2 (0x02)
        Page scan mode: Mandatory (0x00)
        Clock offset: 0x0000
        Role switch: Allow peripheral (0x01)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                            #5588 [hci0] 65.271064
      Create Connection (0x01|0x0005) ncmd 1
        Status: Success (0x00)
> HCI Event: Connect Complete (0x03) plen 11                                                                                                                                         #5589 [hci0] 65.811190
        Status: Success (0x00)
        Handle: 1
        Address: A0:56:F3:97:94:4C (Apple, Inc.)
        Link type: ACL (0x01)
        Encryption: Disabled (0x00)
```

[bt.log](https://github.com/bluez/bluez/files/15189233/bt.log)