bluez/bluez #837: Unexpected result from build option '--enable-nfc'

timoknoop, 2024-05-08:
Hello,

I was looking for possibilities to pair my device using Out-Of-Band (OOB) pairing, when searching through this repo I came across the push_oob and request_oob DBusMessages in the neard.c plugin. After looking at the README file I activated the neard plugin using the build option '--enable-nfc' by building a new BlueZ image. After starting the org.neard DBus service I was expecting to be able to access the push_oob and request_oob methods through the org.neard.HandoverAgent interface but when running the "busctl introspect org.neard /org/neard" command I got an unexpected result:
[image]

I don't know where the PHDC or NDEF functionalities came from as I was only expecting these interfaces as defined in the neard.c file:

[image]

Can anyone explain this unexpected behavior?

Thank you

Timo

Vudentz, 2024-05-13:
> Hello,
> 
> I was looking for possibilities to pair my device using Out-Of-Band (OOB) pairing, when searching through this repo I came across the push_oob and request_oob DBusMessages in the neard.c plugin. After looking at the README file I activated the neard plugin using the build option '--enable-nfc' by building a new BlueZ image. After starting the org.neard DBus service using the command "systemctl start neard" I was expecting to be able to access the push_oob and request_oob methods through the org.neard.HandoverAgent interface but when running the "busctl introspect org.neard /org/neard" command I got an unexpected result: ![image](https://private-user-images.githubusercontent.com/90353716/328908954-08b031fd-4ee0-4dc7-a2fa-60719b19bccd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2MzQyOTAsIm5iZiI6MTcxNTYzMzk5MCwicGF0aCI6Ii85MDM1MzcxNi8zMjg5MDg5NTQtMDhiMDMxZmQtNGVlMC00ZGM3LWEyZmEtNjA3MTliMTliY2NkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDIwNTk1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE2NmJmN2JiZjQwZTAwZDlkZjVkZTc3NzcxMDRlMDQ2OTE1Nzk1OTIxMjIwZDczNWMzNDhmYTAwMWI5ZGViYTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.32ezH8u1Yhwe3H_RbDaxzHdtsJbrREoaJRgAf2HOdPQ)
> 
> I don't know where the PHDC or NDEF functionalities came from as I was only expecting these interfaces as defined in the neard.c file:

Not sure they matter to Bluetooth actually, the expected method is there which is RegisterHandoverAgent

> ![image](https://private-user-images.githubusercontent.com/90353716/328908102-faddd844-5f96-42ce-8e3e-992fac6863c1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2MzQyOTAsIm5iZiI6MTcxNTYzMzk5MCwicGF0aCI6Ii85MDM1MzcxNi8zMjg5MDgxMDItZmFkZGQ4NDQtNWY5Ni00MmNlLThlM2UtOTkyZmFjNjg2M2MxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDIwNTk1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE5NmUxMmU5ZTYwZWMzMjU3OGVmMGExNGEwZDM0YjAyODMyZDg4MzE2NWMwZTYxYzU2MGQwYmNiOGFjODdlZDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.B-7VafFCj3YVpuQ2uBVsEosRsUuLnEZohmdMn28A3z0)
> 
> I am using Bluez version 5.65
> 
> Can anyone explain this unexpected behavior or provide a step-by-step explanation on how to implement OOB into BlueZ?

What is unexpected? Or are you talking about org.neard.Manager vs org.neard.AgentManager? I think that has been broken ever since:

https://github.com/linux-nfc/neard/commit/e17ec5222707768b5f039b224077ec98c2d5f23f

timoknoop, 2024-05-14:
> > Hello,
> > I was looking for possibilities to pair my device using Out-Of-Band (OOB) pairing, when searching through this repo I came across the push_oob and request_oob DBusMessages in the neard.c plugin. After looking at the README file I activated the neard plugin using the build option '--enable-nfc' by building a new BlueZ image. After starting the org.neard DBus service using the command "systemctl start neard" I was expecting to be able to access the push_oob and request_oob methods through the org.neard.HandoverAgent interface but when running the "busctl introspect org.neard /org/neard" command I got an unexpected result: ![image](https://private-user-images.githubusercontent.com/90353716/328908954-08b031fd-4ee0-4dc7-a2fa-60719b19bccd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2MzQyOTAsIm5iZiI6MTcxNTYzMzk5MCwicGF0aCI6Ii85MDM1MzcxNi8zMjg5MDg5NTQtMDhiMDMxZmQtNGVlMC00ZGM3LWEyZmEtNjA3MTliMTliY2NkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDIwNTk1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE2NmJmN2JiZjQwZTAwZDlkZjVkZTc3NzcxMDRlMDQ2OTE1Nzk1OTIxMjIwZDczNWMzNDhmYTAwMWI5ZGViYTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.32ezH8u1Yhwe3H_RbDaxzHdtsJbrREoaJRgAf2HOdPQ)
> > I don't know where the PHDC or NDEF functionalities came from as I was only expecting these interfaces as defined in the neard.c file:
> 
> Not sure they matter to Bluetooth actually, the expected method is there which is RegisterHandoverAgent
> 
> > ![image](https://private-user-images.githubusercontent.com/90353716/328908102-faddd844-5f96-42ce-8e3e-992fac6863c1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2MzQyOTAsIm5iZiI6MTcxNTYzMzk5MCwicGF0aCI6Ii85MDM1MzcxNi8zMjg5MDgxMDItZmFkZGQ4NDQtNWY5Ni00MmNlLThlM2UtOTkyZmFjNjg2M2MxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTEzVDIwNTk1MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE5NmUxMmU5ZTYwZWMzMjU3OGVmMGExNGEwZDM0YjAyODMyZDg4MzE2NWMwZTYxYzU2MGQwYmNiOGFjODdlZDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.B-7VafFCj3YVpuQ2uBVsEosRsUuLnEZohmdMn28A3z0)
> > I am using Bluez version 5.65
> > Can anyone explain this unexpected behavior or provide a step-by-step explanation on how to implement OOB into BlueZ?
> 
> What is unexpected? Or are you talking about org.neard.Manager vs org.neard.AgentManager? I think that has been broken ever since:
> 
> [linux-nfc/neard@e17ec52](https://github.com/linux-nfc/neard/commit/e17ec5222707768b5f039b224077ec98c2d5f23f)

Thank you for your reply, I wasn't aware the neard plugin referred to a different git project as I thought it only referred to the local [neard.c](https://github.com/bluez/bluez/blob/master/plugins/neard.c) file found in this git project. This explains a lot of my confusion. Am I correct to assume that by using the functionalities within linux-nfc I can request or push Out-Of-Band values without being forced to use NFC as I plan on using a wired connection to transfer OOB values between my devices?

Vudentz, 2024-05-14:
> Thank you for your reply, I wasn't aware the neard plugin referred to a different git project as I thought it only referred to the local [...]

Other types of transport will probably require their own plugin to be able to communicate the OOB data, etc, or you can do it yourself with use of MGMT socket but that requires root access thus why we normally recommend doing a plugin to exchange data with other entities doing the OOB communication, or we can actually do a more generic interface of OOB data exchange but Im afraid there is a risk of it being misused since that can be used to bypass pairing procedure.

timoknoop, 2024-05-15:
> Other types of transport will probably require their own plugin to be able to communicate the OOB data, etc, or you can do it yourself with use of MGMT socket but that requires root access thus why [...]

Thanks for your help! Seems like I have more work to do that I initially thought, creating a custom plugin seems like a fun challenge :)

closed 2024-05-15 by timoknoop