bluez/bluez #838: bluetoothd crashes sometimes when reconnecting my Bluetooth keyboard

X-m7, 2024-05-10:
Recently, I've noticed that in some situations (so far appearing to be when either resuming from suspend or after idling for a few hours with no activity on the BT keyboard), bluetoothd crashes after I press a key on the BT keyboard to make it reconnect, so I can only use the BT keyboard after bluetoothd restarts. The BT keyboard will then work fine, until the next time the system wakes up from suspend or after idling.

Just turning the BT keyboard off and on again doesn't seem to cause the problem, and neither does turning the BT adapter off and on again.

I only saw this problem starting from a few weeks or so ago, prior to that everything was working fine, not sure when exactly it started happening.

Going through the core dumps tend to be mostly SIGABRT or SIGSEGV, with a couple of SIGBUS ones thrown in. Here are some of the back traces I got:

[bt.log](https://github.com/bluez/bluez/files/15270822/bt.log)
[bt2.log](https://github.com/bluez/bluez/files/15270823/bt2.log)
[bt3.log](https://github.com/bluez/bluez/files/15270824/bt3.log)
[bt4.log](https://github.com/bluez/bluez/files/15270826/bt4.log)
[bt5.log](https://github.com/bluez/bluez/files/15270827/bt5.log)
[bt6.log](https://github.com/bluez/bluez/files/15270844/bt6.log)

The system is running Arch Linux with the Intel AX200 WiFi+Bluetooth adapter and BlueZ 5.75, and the Bluetooth keyboard is a Logitech K480.

snow-nene, 2024-05-11:
I have  the same problem, bro. My keyboard is HHKB BT. 
           PID: 4642 (bluetoothd)
           UID: 0 (root)
           GID: 0 (root)
        Signal: 11 (SEGV)
     Timestamp: Sat 2024-05-11 20:09:47 CST (4min 4s ago)
  Command Line: /usr/lib/bluetooth/bluetoothd
    Executable: /usr/lib/bluetooth/bluetoothd
 Control Group: /system.slice/bluetooth.service
          Unit: bluetooth.service
         Slice: system.slice
       Boot ID: a15d627d1a40499d9c71a1e7e69ffdc7
    Machine ID: 37d1e8428dab442ea6b700c433288181
      Hostname: archlinux
       Storage: /var/lib/systemd/coredump/core.bluetoothd.0.a15d627d1a40499d9c71a1e7e69ffdc7.4642.1715429387000000.zst (present)
  Size on Disk: 140.5K
       Message: Process 4642 (bluetoothd) of user 0 dumped core.
                
                Stack trace of thread 4642:
                #0  0x0000780dbb60deb7 n/a (libglib-2.0.so.0 + 0x58eb7)
                #1  0x0000780dbb60e1f0 n/a (libglib-2.0.so.0 + 0x591f0)
                #2  0x0000780dbb611c58 n/a (libglib-2.0.so.0 + 0x5cc58)
                #3  0x0000780dbb6739b7 n/a (libglib-2.0.so.0 + 0xbe9b7)
                #4  0x0000780dbb612787 g_main_loop_run (libglib-2.0.so.0 + 0x5d787)
                #5  0x00006232aef16ef3 n/a (bluetoothd + 0x9ef3)
                #6  0x0000780dbb274c88 n/a (libc.so.6 + 0x25c88)
                #7  0x0000780dbb274d4c __libc_start_main (libc.so.6 + 0x25d4c)
                #8  0x00006232aef18155 n/a (bluetoothd + 0xb155)
                ELF object binary architecture: AMD x86-64

RodoMa92, 2024-05-18:
I can reproduce the same here, same Logitech K480 keyboard as OP, same bluez version (5.75). I'll post logs shortly.

ImReallyAdi, 2026-01-31:
Any fixes yall?