bluez/bluez #839: BAP Unicast issue

vlad-pruteanu, 2024-05-10:
As discussed, in order to fix the issue where the qos latency field would take a value smaller than the SDU interval, for Unicast, the latency parameter will refer to the maximum latency set by the host, as is the case for Broadcast.
 
I was investigating the impact of this change, which requires modifying the kernel so that the `hci_le_cis_established_evt` does not overwrite the latency parameter. While doing so, I discovered a bug in the Unicast setup. If the Source is an Unicast Server and the Sink an Unicast Client, then on the Source,  `iso_qos.ucast.out` won't be set by BAP. This is because the created stream's endpoint direction on Source is set to Source instead of Sink.
 
`setup_create_ucast_io` calls `bt_bap_stream_io_get_qos`
`bt_bap_stream_io_get_qos` uses the stream's qos to set `qos[0]` if the `ep->dir` is SOURCE, otherwise, if it is SINK, it sets `qos[1]`
`setup_create_ucast_io` then sets `iso_qos.ucast.in` to `qos[0]` and `iso_qos.ucast.out` to `qos[1]`
 
So, on the Source, with `ep->dir` being Source, `qos[1]` will remain unset, and therefore `iso_qos.ucast.out` will also be empty.
 
Adding the following lines to the `bt_bap_stream_io_get_qos` function yields the attached logs. There you can see that EP_DIR is SOURCE for both devices.
```
	DBG(stream->bap,"STREAM EP IS %p\n", stream->ep);
	DBG(stream->bap,"STREAM EP_DIR IS %d\n", stream->ep->dir);
	if(stream->link)
		DBG(stream->bap,"STREAM LINK IS %d\n", stream->link);
```
 
This issue is not currently visible as the kernel sets the iso_qos.ucast.out latency and interval on the `hci_le_cis_established_evt`. However, we shouldn't rely on the kernel to do this and the qos should be set by the Host appropriately.

Logs:
[log_bluez_sink_client.txt](https://github.com/bluez/bluez/files/15276136/log_bluez_sink_client.txt)
[log_bluez_source_server.txt](https://github.com/bluez/bluez/files/15276139/log_bluez_source_server.txt)

Vudentz, 2024-05-10:
You can't be both unicast client and source at the same time, BAP doesn't allow that, but perhaps you are talking a source + sink setup started by the client which the server doesn't setup the source qos?

pv, 2024-05-10:
> the latency parameter will refer to the maximum latency set by the host

User applications should still be able to find the actual transport latency (not maximum), as it's needed for knowing the SDU synchronizarion reference (See Core Vol 6 Part G Fig 3.1/3.2 and BAP Sec. 7).

closed 2025-07-26 by github-actions[bot]