bluez/bluez #840: bluez-daemon crashed, segment error

snow-nene, 2024-05-11:
My keyboard is HHKB BT, and the bluetooth adapter is MT7921. I notice that sometimes the keyboard connection is unstable. When I connect the keyboard to the computer, it may soon disconnect and the bluez-daemon process ended with the core-dump blow and it is segment error.

This is the coredumpinfo:
(base) [clifford@archlinux ~]$ sudo coredumpctl info 
[sudo] password for clifford: 
           PID: 4642 (bluetoothd)
           UID: 0 (root)
           GID: 0 (root)
        Signal: 11 (SEGV)
     Timestamp: Sat 2024-05-11 20:09:47 CST (4min 4s ago)
  Command Line: /usr/lib/bluetooth/bluetoothd
    Executable: /usr/lib/bluetooth/bluetoothd
 Control Group: /system.slice/bluetooth.service
          Unit: bluetooth.service
         Slice: system.slice
       Boot ID: a15d627d1a40499d9c71a1e7e69ffdc7
    Machine ID: 37d1e8428dab442ea6b700c433288181
      Hostname: archlinux
       Storage: /var/lib/systemd/coredump/core.bluetoothd.0.a15d627d1a40499d9c71a1e7e69ffdc7.4642.1715429387000000.zst (present)
  Size on Disk: 140.5K
       Message: Process 4642 (bluetoothd) of user 0 dumped core.
                
                Stack trace of thread 4642:
                #0  0x0000780dbb60deb7 n/a (libglib-2.0.so.0 + 0x58eb7)
                #1  0x0000780dbb60e1f0 n/a (libglib-2.0.so.0 + 0x591f0)
                #2  0x0000780dbb611c58 n/a (libglib-2.0.so.0 + 0x5cc58)
                #3  0x0000780dbb6739b7 n/a (libglib-2.0.so.0 + 0xbe9b7)
                #4  0x0000780dbb612787 g_main_loop_run (libglib-2.0.so.0 + 0x5d787)
                #5  0x00006232aef16ef3 n/a (bluetoothd + 0x9ef3)
                #6  0x0000780dbb274c88 n/a (libc.so.6 + 0x25c88)
                #7  0x0000780dbb274d4c __libc_start_main (libc.so.6 + 0x25d4c)
                #8  0x00006232aef18155 n/a (bluetoothd + 0xb155)
                ELF object binary architecture: AMD x86-64

Vudentz, 2024-05-13:
Version?

snow-nene, 2024-05-14:
> Version?

local/bluez 5.75-1
    Daemons for the bluetooth protocol stack
local/bluez-libs 5.75-1
    Deprecated libraries for the bluetooth protocol stack
local/bluez-qt 6.2.0-1 (kf6)
    Qt wrapper for Bluez 5 DBus API
local/bluez-utils 5.75-1
    Development and debugging utilities for the bluetooth protocol stack

closed 2025-07-26 by github-actions[bot]