bluez/bluez #841: Remove of remote LE OOB data

SFrohn, 2024-05-11:
Hello,

we are trying to remove LE remote OOB using the "Remove Remote Out Of Band Data Command" (0x0022) of the Mgmt API.
However, when setting address_type to 1 or 2 the command always returns an error ("INVALID_PARAMETER").

Looking at the implementation in mgmt.c, the function seems not to accept address_type LE_PUBLIC or LE_RANDOM: https://github.com/bluez/bluetooth-next/blob/93e31170f4d098a9c07c5e389091ea45cb10e72a/net/bluetooth/mgmt.c#L5814

Is this intended behavior or is there any other way to remove LE OOB data for a certain device?

By the way, calling this function with BDADDR_ANY and BREDR address_type works as documented and removes _all_ remote OOB data.

Best regards
Simon

Vudentz, 2024-05-13:
@SFrohn might be a bug actually since the documentation does say it is allowed:

```
	Possible values for the Address_Type parameter:
		0	BR/EDR
		1	LE Public
		2	LE Random

```

SFrohn, 2024-05-23:
@Vudentz Yes, exactly. According to this documentation, we expected address_type LE_PUBLIC or LE_RANDOM to be valid as well. As the opposite command "Add Remote Out Of Band Data" accepts these address types, it is probably a bug.

closed 2025-07-25 by github-actions[bot]