bluez/bluez #843: >=bluez-5.72 fail to build with musl-1.2.5: tools/hex2hcd.c:305:18: error: call to undeclared function 'basename'

ernsteiswuerfel, 2024-05-13:
bluez-5.72 and current 5.75 fail to build on musl-1.2.5 systems:
```
[...]
clang -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib/dbus-1.0/include -I/usr/include/elogind -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -pthread -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I./lib  -ffunction-sections -fdata-sections      -Os -march=x86-64-v2 -pipe -c -o tools/hex2hcd.o tools/hex2hcd.c
tools/hex2hcd.c:305:18: error: call to undeclared function 'basename'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  305 |                 ver_parse_file(basename(pathname));
      |                                ^
tools/hex2hcd.c:305:18: error: incompatible integer to pointer conversion passing 'int' to parameter of type 'const char *' [-Wint-conversion]
  305 |                 ver_parse_file(basename(pathname));
      |                                ^~~~~~~~~~~~~~~~~~
tools/hex2hcd.c:234:40: note: passing argument to parameter 'pathname' here
  234 | static void ver_parse_file(const char *pathname)
      |                                        ^
2 errors generated.
make[1]: *** [Makefile:7817: tools/hex2hcd.o] Error 1
make: *** [Makefile:4654: all] Error 2
```
This is Gentoo downstream bugreport #[926344](https://bugs.gentoo.org/926344).

Build log attached.
[bluez-5.75:20240513-203450.log](https://github.com/bluez/bluez/files/15299850/bluez-5.75.20240513-203450.log)

ernsteiswuerfel, 2024-08-14:
This specific issue gets fixed with the patch above.

But there are yet more similar issues remaining it seems:
```
[...]
clang -DHAVE_CONFIG_H -I.   -I/usr/include/dbus-1.0 -I/usr/lib/dbus-1.0/include -I/usr/include/elogind -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -pthread -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I./lib  -ffunction-sections -fdata-sections      -Os -march=x86-64-v2 -pipe -c -o mesh/rpl.o mesh/rpl.c
mesh/rpl.c:149:11: error: call to undeclared function 'basename'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  149 |         iv_txt = basename(iv_path);
      |                  ^
mesh/rpl.c:149:9: error: incompatible integer to pointer conversion assigning to 'const char *' from 'int' [-Wint-conversion]
  149 |         iv_txt = basename(iv_path);
      |                ^ ~~~~~~~~~~~~~~~~~
2 errors generated.
make[1]: *** [Makefile:7845: mesh/rpl.o] Error 1
make: *** [Makefile:4669: all] Error 2
[...]
```
[bluez-5.77:20240814-162838.log](https://github.com/user-attachments/files/16616259/bluez-5.77.20240814-162838.log)

cctfubeet, 2024-08-16:
Implementing the same function on mesh/rpl.c provided on the commit above makes it compile again under musl.
Tested on Gentoo Musl (6.10.3-gentoo-dist-hardened).
[Patch for reference.](https://github.com/user-attachments/files/16631356/rpl.patch.txt)

ernsteiswuerfel, 2024-08-19:
Thanks @cctfubeet! I can confirm your patch works and bluez successfully builds on musl-1.2.5 with both patches in place.

Vudentz, 2024-08-19:
> Implementing the same function on mesh/rpl.c provided on the commit above makes it compile again under musl. Tested on Gentoo Musl (6.10.3-gentoo-dist-hardened). [Patch for reference.](https://github.com/user-attachments/files/16631356/rpl.patch.txt)

Perhaps having this as part of shared/util.h would have been better (e.g. util_basename)

alyssais, 2024-09-16:
The fix doesn't seem right — mesh/mesh-config-json.cc includes both mesh/missing.h and libgen.h, so it gets two conflicting definitions for basename, which is causes a compiler error:

```
       > In file included from mesh/mesh-config-json.c:31:
       > ./mesh/missing.h:15:27: error: conflicting types for ‘basename’; have ‘const char *(const char *)’
       >    15 | static inline const char *basename(const char *path)
       >       |                           ^~~~~~~~
       > In file included from mesh/mesh-config-json.c:20:
       > /nix/store/xym3znxirxk462a4cjdcvn1p2zybj3v0-musl-1.2.5-dev/include/libgen.h:9:7: note: previous declaration of ‘basename’ with type ‘char *(char *)’
       >     9 | char *basename(char *);
       >       |       ^~~~~~~~
       >   CC       mesh/agent.o
```

kraj, 2024-09-16:
@Vudentz does your musl build include - https://git.musl-libc.org/cgit/musl/commit/?id=725e17ed6dff4d0cd22487bb64470881e86a92e7 ?

alyssais, 2024-09-16:
Assuming you meant to ask me.  Yes it does.  It's musl 1.2.5.

alyssais, 2024-09-16:
[config.log](https://github.com/user-attachments/files/17016832/config.log)

kraj, 2024-09-16:
> [config.log](https://github.com/user-attachments/files/17016832/config.log)

@alyssais thanks, I think the detection seems to work fine which I wanted to check, if you could upload the pre-processed output of `mesh/mesh-config-json.c` that could help in next step.

alyssais, 2024-09-16:
[mesh-config-json.preprocessed.c.txt](https://github.com/user-attachments/files/17018041/mesh-config-json.preprocessed.c.txt)

kraj, 2024-09-16:
> But it seems pretty straightforward to me? There is a basename defined in libgen.h, and there is a basename defined in missing.h. These two definitions conflict. mesh-config-json.c includes both of them.

yeah thanks for adding this info, it helped a lot, I thought that removing `libgen.h` include would do it but this file also uses `dirname` API which should require including `libgen.h`,  so a way around would be to rename local basename to something like bluez_basename and use it everywhere to avoid namespace clash

kraj, 2024-09-16:
@alyssais Can you test out this patch in your environment for me please ?

[0001-mesh-Move-local-basename-into-utility-file.patch](https://github.com/user-attachments/files/17020818/0001-mesh-Move-local-basename-into-utility-file.patch)

macmpi, 2024-11-02:
fixed with https://github.com/bluez/bluez/commit/7104f37735cd74ef57a4e18d61e6f55a99edd7a8 ?

alyssais, 2024-11-02:
Yes, it seems to have been!  The latest release has a new failure, but I see [a patch](https://lore.kernel.org/linux-bluetooth/20241102161018.2518100-1-gudni.m.g@gmail.com/) has recently been submitted to fix it.

closed 2025-07-25 by github-actions[bot]