bluez/bluez #844: No Bluetooth devices connecting

kode54, 2024-05-14:
After leaving my system booted for so long, no devices will connect to the Bluetooth radio. From the kernel log:

```
May 14 11:20:06 laharl bluetoothd[3189716]: Failed to set mode: Authentication Failed (0x05)
May 14 11:20:06 laharl bluetoothd[3189716]: Too small Add Device complete event
May 14 11:20:55 laharl bluetoothd[3189716]: src/device.c:device_connect_le() ATT bt_io_connect(F4:6A:D7:10:6A:F4): connect to F4:6A:D7:10:6A:F4: No route to host (113)
May 14 11:21:12 laharl bluetoothd[3189716]: src/device.c:device_connect_le() ATT bt_io_connect(F4:6A:D7:10:6A:F4): connect to F4:6A:D7:10:6A:F4: No route to host (113)
```

This one was for an Xbox Wireless Controller, but I can't connect a DualShock 4 contoller, either. I've restarted bluetoothd several times.

kode54, 2024-05-15:
More logs:

[journal.log.txt](https://github.com/bluez/bluez/files/15326704/journal.log.txt)

And just the bluetooth parts:
[journal.log.bluetooth.txt](https://github.com/bluez/bluez/files/15326705/journal.log.bluetooth.txt)

Machion, 2024-09-14:
It seems, I have a similar issue on Manjaro Stable/bluez 5.77

I have bluetooth turned off by default. Only messages in the logs are: 
```
    Failed to set mode: Failed (0x03)
    Too small Add Device complete event
```

Now I tried connecting a Trust Bosi GXT 590 Gamepad. It always connects/disconnects with following messages: 
```
profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
profiles/input/hog-lib.c:external_report_reference_cb() Read External Report Reference descriptor failed: Request attribute has encountered an unlikely error
profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
```

closed 2025-07-25 by github-actions[bot]