bluez/bluez #847: [5.72] BlueZ disconnects from peripheral after repeatedly rejecting Connection Parameter Update

charlesnicholson, 2024-05-17:
I'm using BlueZ 5.72 on Ubuntu 24.04:
```
charles@fi-nuc-charles:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 24.04 LTS
Release:	24.04
Codename:	noble
```

I'm using BlueZ as an LE Central / GATT Client to connect to an IoT device acting as LE Peripheral / GATT Server. BlueZ connects with an initial connection interval range of 30-50msec, and my peripheral accepts with 45msec:

```
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 42             #58 [hci0] 66.609420
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Random (0x01)
        Peer address: E2:96:74:76:59:4B (Static)
        Initiating PHYs: 0x03
        Entry 0: LE 1M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 30.00 msec (0x0018)
          Max connection interval: 50.00 msec (0x0028)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
        Entry 1: LE 2M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 30.00 msec (0x0018)
          Max connection interval: 50.00 msec (0x0028)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                      #59 [hci0] 66.611274
      LE Extended Create Connection (0x08|0x0043) ncmd 2
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 31                                      #60 [hci0] 67.152317
      LE Enhanced Connection Complete (0x0a)
        Status: Success (0x00)
        Handle: 3585
        Role: Central (0x00)
        Peer address type: Random (0x01)
        Peer address: E2:96:74:76:59:4B (Static)
        Local resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Peer resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Connection interval: 45.00 msec (0x0024)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Central clock accuracy: 0x00
@ MGMT Event: Device Connected (0x000b) plen 38                           {0x0001} [hci0] 67.152352
        LE Address: E2:96:74:76:59:4B (Static)
        Flags: 0x00000008
          Connection Locally Initiated
        Data length: 25
        Appearance: Unknown (0x0000)
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        128-bit Service UUIDs (complete): 1 entry
          Vendor specific
```

The connection is established, the services + characteristics + descriptors are enumerated, data transfer continues and all seems well. No pairing or bonding takes place.

But, my peripheral needs a higher connection interval, between 100-200msec (80-160 time units), with a 4s connection supervisory timeout. It attempts repeatedly to request this, but is rejected by BlueZ every time:

```
> ACL Data RX: Handle 3585 flags 0x02 dlen 16                                 #151 [hci0] 72.159343
      LE L2CAP: Connection Parameter Update Request (0x12) ident 3 len 8
        Min interval: 80
        Max interval: 160
        Peripheral latency: 0
        Timeout multiplier: 400
< ACL Data TX: Handle 3585 flags 0x00 dlen 10                                 #152 [hci0] 72.159457
      LE L2CAP: Connection Parameter Update Response (0x13) ident 3 len 2
        Result: Connection Parameters rejected (0x0001)
```

This pattern of request / reject continues until eventually my peripheral gives up and disconnects with the "Unacceptable Connection Parameters" code:
```
> HCI Event: Disconnect Complete (0x05) plen 4                               #871 [hci0] 162.383723
        Status: Success (0x00)
        Handle: 3585 Address: E2:96:74:76:59:4B (Static)
        Reason: Unacceptable Connection Parameters (0x3b)
@ MGMT Event: Device Disconnected (0x000c) plen 8                        {0x0001} [hci0] 162.383816
        LE Address: E2:96:74:76:59:4B (Static)
        Reason: Unspecified (0x00)
```

I've attached my btmon session saved as a snoop file. I've also confirmed this behavior externally with an Ellisys BTR-1:
[image]
[image]

My BlueZ conf, which on my system lives in `/etc/bluetooth/main.conf`, has no values set for any of the defaults:
```
# LE default connection parameters.  These values are superceeded by any
# specific values provided via the Load Connection Parameters interface
#MinConnectionInterval=
#MaxConnectionInterval=
#ConnectionLatency=
#ConnectionSupervisionTimeout=
#Autoconnecttimeout=
```

But, when I change them to 6 and 3200, the problem persists.

[connection-interval.btsnoop.zip](https://github.com/bluez/bluez/files/15351878/connection-interval.btsnoop.zip)

I have also tried changing the supervisory timeout down from 4s to ~800ms, but that made no difference either.

Any help that anyone can offer me would be greatly appreciated! I'm having trouble figuring out why BlueZ is rejecting me here; I'm unsure if it's a bug or a low-level radio driver limitation or .... ?

Thanks in advance for reading.
Charles

charlesnicholson, 2024-05-17:
Re-running the same test code from my Macbook, an Ellisys capture reveals that the Connection Parameter Update Request does succeed against Apple's CoreBluetooth system:
[image]

Vudentz, 2024-05-17:
Looks like it is failing in either of these cases:

```
	if (max > hcon->le_conn_max_interval) {
		BT_DBG("requested connection interval exceeds current bounds.");
		err = -EINVAL;
	} else {
		err = hci_check_conn_params(min, max, latency, to_multiplier);
	}

static inline int hci_check_conn_params(u16 min, u16 max, u16 latency,
					u16 to_multiplier)
{
	u16 max_latency;

	if (min > max || min < 6 || max > 3200)
		return -EINVAL;

	if (to_multiplier < 10 || to_multiplier > 3200)
		return -EINVAL;

	if (max >= to_multiplier * 8)
		return -EINVAL;

	max_latency = (to_multiplier * 4 / max) - 1;
	if (latency > 499 || latency > max_latency)
		return -EINVAL;

	return 0;
}
```

I believe the problem might be that max > hcon->le_conn_max_interval which was introduced by:

commit e4b019515f950b4e6e5b74b2e1bb03a90cb33039
Author: Kai-Heng Feng <kai.heng.feng@canonical.com>
Date:   Thu Jan 25 14:50:28 2024 +0800

    Bluetooth: Enforce validation on max value of connection interval
    
    Right now Linux BT stack cannot pass test case "GAP/CONN/CPUP/BV-05-C
    'Connection Parameter Update Procedure Invalid Parameters Central
    Responder'" in Bluetooth Test Suite revision GAP.TS.p44. [0]
    
    That was revoled by commit c49a8682fc5d ("Bluetooth: validate BLE
    connection interval updates"), but later got reverted due to devices
    like keyboards and mice may require low connection interval.
    
    So only validate the max value connection interval to pass the Test
    Suite, and let devices to request low connection interval if needed.
    
    [0] https://www.bluetooth.org/docman/handlers/DownloadDoc.ashx?doc_id=229869
    
    Fixes: 68d19d7d9957 ("Revert "Bluetooth: validate BLE connection interval updates"")
    Signed-off-by: Kai-Heng Feng <kai.heng.feng@canonical.com>
    Signed-off-by: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>

The current range is:
```

	hdev->le_conn_min_interval = 0x0018;
	hdev->le_conn_max_interval = 0x0028;

```

I think this range is actually incorrect, it is actually the default range in case there is no device specific (hci_conn_params) but shall probably allow something over that, also note that we recently introduced support for PPCP which shall create a valid hci_conn_params if the device supports it.

charlesnicholson, 2024-05-17:
Thanks for investigating! I would (naively) assume that the Central should be able to accept any legal ranges in [7.5ms 4000ms], barred only by the limitations of the radio hardware + firmware. I'm curious what the actual spec allows here.

If it's interesting / useful, it's quite common to have a connection interval > 1 second in ultra-low-power IoT devices.

Vudentz, 2024-05-17:
[image]

It looks like we are using TGAP(initial_conn_interval) but that is just a recommended value for the initiating a connection over LE 1M PHY, perhaps we shall have a separated set of field for the maximum we could accept and then probably use it for triggering the reject response.

charlesnicholson, 2024-05-17:
It occurred to me to run an experiment where my peripheral attempts to negotiate something in that range other than the 45ms default, just to see if BlueZ would accept it. I suspect given what you wrote here that it would, but would evidence from such an experiment be a helpful contribution to this ticket anyway?

Vudentz, 2024-05-17:
> Thanks for investigating! I would (naively) assume that the Central should be able to accept any legal ranges in [7.5ms 4000ms], barred only by the limitations of the radio hardware + firmware. I'm [...]

Above 1 second sounds sort of too slow imo, except if you are thinking on changing that when there is data then adjust it back when idle, but not really beats disconnecting in terms of power safe which is what I would expect the ULP IoT devices to do if they don't have any data to transmit then just disconnect.

Perhaps we could use the link supervision timeout as a guidance to maximum connection latency, that would be 420ms.

charlesnicholson, 2024-05-17:
Above 1 second is extremely useful for things like environmental sensors that run on exhaustible batteries. I shipped a mass-produced sensor that used a 1.2s connection interval to phones / dedicated hub hardware on a coin cell battery- 1.2s interval let us offer a 3+ month battery life. There's a significant number of devices like these; it's not great to support them just because they sound too slow. Remember also that staying connected is a lower-power mode than advertising, because advertising is traditionally on 3 channels, and connections are only on one channel. That's 1/3 the total number of radio transmissions, which can be significant.

For a system like this, I would carefully choose the supervision timeout to be a function of the minimum-acceptable connection interval on a device. So, for example, if my device has a 1-1.2s connection interval, I would choose maybe (1 * 4) seconds, allowing my device to miss 4 connection intervals before a disconnection.

I'm not sure what the value is in artificially limiting the devices that BlueZ supports, if for no other reason than for automated testing during development of these devices :)

Vudentz, 2024-05-17:
> Above 1 second is extremely useful for things like environmental sensors that run on exhaustible batteries. I shipped a mass-produced sensor that used a 1.2s connection interval to phones / dedicated hub hardware on a coin cell battery- 1.2s interval let us offer a 3+ month battery life. There's a significant number of devices like these; it's not great to withhold support for them just because they "sound too slow" :). Remember also that staying connected is a lower-power mode than advertising, because advertising is traditionally on 3 channels, and connections are only on one channel. That's 1/3 the total number of radio transmissions, which can be significant.

While I agree that it can save battery if you increate the interval, you probably can do it even better by only waking up every few minutes, connect, exchange the latest reads and disconnect...

> For a system like this, I would carefully choose the supervision timeout to be a function of the minimum-acceptable connection interval on a device. So, for example, if my device has a 1-1.2s [...]

Yeah, I don't know what is the purpose of that test though, so it looks like we needto specify something as maximum latency for whatever reason, but we landed that patch there was no limits really so the device could set up to 4 seconds, so patches are welcomed if you want to change the defaults by allowing the pass different values via main.conf.

Vudentz, 2024-05-17:
In fact it is already supported to overwrite the maximum latency:

https://github.com/bluez/bluez/blob/master/src/main.conf#L221

charlesnicholson, 2024-05-17:
You are describing one of many possible implementations, but for things like passive medical sensors that want to send data at regular 1s intervals, what you're proposing wouldn't work. I say this just as an example of the IoT BLE world being very large, with a bunch of needs that are different from "traditional" BLE things (kb / mouse / headphones).

My recommendation is to allow BlueZ to accept a parameter update to any legal set of values that the underlying radio + firmware supports. If the system ends up being strange, that's the developer's fault, right? BlueZ shouldn't stop me from doing such things, because BlueZ can't know my intent :). If I want a 1s connection interval with a 30s supervisory timeout, why should BlueZ actively prohibit me from doing that?

Related: If you haven't seen them, here are some interesting guidelines from Apple about intervals + timeouts:
https://www.bluetooth.com/wp-content/uploads/attachments/BluetoothDesignGuidelines.pdf

(Note the recommendation that the supervision timeout be > 3x the maximum connection interval.)

[image]

I might attempt a patch, though I'm not sure what the `MinConnectionInterval` and `MaxConnectionInterval` in `main.conf` are meant to be used for- is the intent behind them that they should specify the min + max connection intervals that a peer could attempt to update to after connection? I found that changing `MaxConnectionInterval` had no effect (I mentioned this in my initial posting). Perhaps I'm using it incorrectly?

bgix, 2024-05-17:
This sounds like a job for a
Bluetooth Mesh LPN (Low Power Node)

Vudentz, 2024-05-17:
> I might attempt a patch, though I'm not sure what the `MinConnectionInterval` and `MaxConnectionInterval` in `main.conf` are meant to be used for- is the intent behind them that they should specify [...]

Perhaps there is bug somewhere, did you reload the service after changing it? Also the range is between 0x0006-0x0c80, so it is using the units of the spec not msec.

charlesnicholson, 2024-05-17:
I changed them to this:
```
MinConnectionInterval=6
MaxConnectionInterval=3200
```

I believe that those correlate to 7.5ms and 4000ms, since the time unit is in 1.25ms, right?

Is it ok to use decimal values, or do they have to be hex? I did reload the service, but I'm happy to re-run all of my experiments- I'd love to be wrong here :)

Vudentz, 2024-05-17:
> I changed them to this:
>
> ```
> MinConnectionInterval=6
> MaxConnectionInterval=3200
> ```
>
> I believe that those correlate to 7.5ms and 4000ms, since the time unit is in 1.25ms, right?
>
> Is it ok to use [...]

3200 * 1.25 = 4000 so that should be allowed, and yeah you need to reload the bluetooth.service if you want them to take effect, note though that it will effect every connection you made, regardless of the device type, thus why I said that perhaps we need a different field but that will require changes to the kernel.

charlesnicholson, 2024-05-17:
Oh, interesting, perhaps I didn't restart the service correctly last time? Sadly, though, I'm seeing a new failure upon connection establishment: my peripheral is rejecting the Create Connection command. 

Presumably this is happening because the initial connection parameters are 7.5ms-4000ms, instead of the spec-required 30-50ms:

```
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 42                                                             #49 [hci0] 68.774458
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Random (0x01)
        Peer address: E2:96:74:76:59:4B (Static)
        Initiating PHYs: 0x03
        Entry 0: LE 1M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 7.50 msec (0x0006)
          Max connection interval: 4000.00 msec (0x0c80)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
        Entry 1: LE 2M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 7.50 msec (0x0006)
          Max connection interval: 4000.00 msec (0x0c80)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                                                      #50 [hci0] 68.776347
      LE Extended Create Connection (0x08|0x0043) ncmd 1
        Status: Invalid HCI Command Parameters (0x12)
@ MGMT Event: Connect Failed (0x000d) plen 8                                                                              {0x0001} [hci0] 68.776399
        LE Address: E2:96:74:76:59:4B (Static)
        Status: Invalid Parameters (0x0d)
```

So maybe BlueZ should hard-code the initial 30-50ms range during a connection, but then use the `main.conf` values to decide whether to accept or reject subsequent interval update requests?

charlesnicholson, 2024-05-17:
Also the min and max connection length values look incorrect at 0ms :(

Vudentz, 2024-05-17:
> Also the min and max connection length values look incorrect at 0ms :(

Don't think we ever set those, btw I think you need to adjust the supervision timeout otherwise the max interval will be bigger than it which I believed is not allowed.

charlesnicholson, 2024-05-17:
Sorry about that; I realized that they were also 0.000 in successful connections and I didn't want to add noise to the ticket. I guess you caught it before I deleted it though :)

Vudentz, 2024-05-17:
Weird that Min_CE_Length and Max_CE_Length are not explained at all, if we set to 0x0000 I would assume that the host don't a care how long it takes to connect, though why it would be given as a range really beats me, that is just a timeout if I understand it correctly and the controller is not even required to use the values, that spells like undefined behavior if you ask me.

charlesnicholson, 2024-05-17:
Oh, I think those refer to the length of connection events inside of an interval (i.e. how many MTUs / PDUs you have time to exchange once a connection interval elapses and you're transmitting) and not related to establishing an actual connection.

charlesnicholson, 2024-05-17:
Ah, that was it, thank you! I hadn't changed `ConnectionSupervisionTimeout` from its default. When I change the values to

```
MinConnectionInterval=6
MaxConnectionInterval=3200
ConnectionSupervisionTimeout=1000
```

then the update succeeds:

```
> ACL Data RX: Handle 3585 flags 0x02 dlen 16                                  #56 [hci0] 62.440154
      LE L2CAP: Connection Parameter Update Request (0x12) ident 3 len 8
        Min interval: 80
        Max interval: 160
        Peripheral latency: 0
        Timeout multiplier: 400
@ MGMT Event: New Connection Parameter (0x001c) plen 16                   {0x0001} [hci0] 62.440202
        Store hint: Reserved (0x4b)
        LE Address: 02:E2:96:74:76:59 (OUI 02-E2-96)
        Min connection interval: 80
        Max connection interval: 160
        Connection latency: 0 (0x0000)
        Supervision timeout: 400
< ACL Data TX: Handle 3585 flags 0x00 dlen 10                                  #57 [hci0] 62.440206
      LE L2CAP: Connection Parameter Update Response (0x13) ident 3 len 2
        Result: Connection Parameters accepted (0x0000)
< HCI Command: LE Connection Update (0x08|0x0013) plen 14                      #58 [hci0] 62.440232
        Handle: 3585 Address: E2:96:74:76:59:4B (Static)
        Min connection interval: 100.00 msec (0x0050)
        Max connection interval: 200.00 msec (0x00a0)
        Connection latency: 0 (0x0000)
        Supervision timeout: 4000 msec (0x0190)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                      #59 [hci0] 62.441923
      LE Connection Update (0x08|0x0013) ncmd 1
        Status: Success (0x00)
```

I agree with everything you say about these _extreme_ defaults perhaps not making sense for most devices, but since in my case the system is only using BLE for automated testing against my IoT device, hitting these values globally with a hammer works fine for me.

Thanks so much for taking a significant amount of time to help me work through this, @Vudentz, I sincerely appreciate it.

charlesnicholson, 2024-05-18:
One final thought on this ticket- I would have saved the better part of a day if BlueZ had logged _why_ it was rejecting the parameter update requests. It's good that there are constraints that prevent nonsensical values (e.g. min > max, etc), but when the only clue in the logs is `rejected`, it's very hard to figure out what to do next.

If the logs had said something like `rejected (requested [100ms-200ms], local range is [30-50ms])` or `rejected (local: supervision timeout 200ms <= interval_max 200ms)` etc, I could have very easily self-served this one and saved myself and @Vudentz a fair amount of effort.

Thanks again!

Vudentz, 2024-05-20:
@charlesnicholson Im actually taking that check out in the above change, PTS testing tool has IXIT fields that one can program with invalid range in order to trigger the reject, otherwise we shall support the full range (6-3200/7.5-4000ms).

charlesnicholson, 2024-05-20:
That's great news (at least, from my perspective about how host-side BLE stacks should behave)!

Though, if there are any supervisory timeout settings that can cause a local rejection, in their relation to the connection interval min/max and latency, it would be nice to see a reason around that too in the monitor logs...

Vudentz, 2024-05-21:
> That's great news (at least, from my perspective about how host-side BLE stacks should behave)!
>
> Though, if there are any supervisory timeout settings that can cause a local rejection, in their [...]

Adressed this on v3 above.

Vudentz, 2024-05-22:
Merged.