bluez/bluez #848: Phantom gamepad devices in /dev after disconnecting

alice-mkh, 2024-05-19:
At some point gamepad devices stopped being properly cleaned up after being disconnected, meaning that if you connect a (bt) gamepad and then disconnect it, everything will still think it's connected until a reboot.

I bisected it down to this commit:

```
ee880bee85864b0e691370197c9de8d2f7ba0af2 is the first bad commit
commit ee880bee85864b0e691370197c9de8d2f7ba0af2
Author: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
Date:   Fri Feb 9 13:50:00 2024 -0500

    input/device: Don't destroy UHID device on disconnect
    
    This makes classic HID behave like HoG which keeps the UHID device
    around while disconnected so it doesn't have to be recreated on every
    reconnection.

 profiles/input/device.c | 4 ++++
 1 file changed, 4 insertions(+)
```

As far as I can tell there's no way to tell if a device is actually connected or disconnected with this. In my case the app I'm working on automatically assigns gamepads and keyboard to different players, and this completely breaks the logic, as gamepads that the system thinks are connected but actually aren't will be prioritized over keyboard.

That change looks intentional, so I'm not necessarily saying it should be reverted, but - any ideas for how to deal with that? Preferably without involving bluez dbus api, as that needs an additional sandbox hole.

alice-mkh, 2024-05-20:
CC @jonas2515 from the matrix discussion

Vudentz, 2024-05-20:
@alice-mkh that is intentional, that said we may want to limit it to just keyboards otherwise it sometimes misses keypresses when reconnecting.

alice-mkh, 2024-05-20:
Limiting it to keyboards would really help, yes - the issue doesn't really apply there since keyboards are combined into a single virtual device anyway, while for gamepads you really want to be able to communicate with individual ones.

Vudentz, 2024-05-20:
> Limiting it to keyboards would really help, yes - the issue doesn't really apply there since keyboards are combined into a single virtual device anyway, while for gamepads you really want to be able to communicate with individual ones.

The issue is then how to detect the device type via uHID, how you guys detect the device type?

alice-mkh, 2024-05-20:
Mm, does the same logic as here not work? https://github.com/bluez/bluez/blob/62d38d6c9ef4017ff6a84af0a9cd9242a7f6a211/src/dbus-common.c#L88-L961

Otherwise udev has detection code here: https://github.com/eudev-project/eudev/blob/master/src/udev/udev-builtin-input_id.c#L313 (and in libmanette we have a simplified copy of that for detecting specifically gamepads).

Vudentz, 2024-05-20:
> Mm, does the same logic as here not work? [`62d38d6`/src/dbus-common.c#L88-L961](https://github.com/bluez/bluez/blob/62d38d6c9ef4017ff6a84af0a9cd9242a7f6a211/src/dbus-common.c#L88-L961)

Oh, good call, I sort of forget we do have some code for providing proper icon support, will try to come up with something that takes that into account.

Vudentz, 2024-05-20:
@alice-mkh could you please try with the above changes?

alice-mkh, 2024-05-20:
Just tried - yep, works fine, thanks!

closed 2024-05-20 by BluezTestBot