bluez/bluez #849: Bluetooth not working

natesworks, 2024-05-19:
src/plugin.c:init_plugin() System does not support ccp plugin

no adapters in bluetoothctl

natesworks, 2024-05-19:
also not blocked by rfkill

StuffOfSonny, 2024-05-20:
I am also having this issue.

natesworks, 2024-05-20:
I use arch btw

abhigyanmadhukalya, 2024-05-20:
Having the same problem on Aurora/Bluefin (Ublue OS).

Vudentz, 2024-05-20:
Version? Controller, HCI traces, etc, might be required as well to investigate what is going on.

StuffOfSonny, 2024-05-20:
Arch Linux 6.9.1-zen1-1-zen, I only have a brief moment of availability so that's all I could grab at the moment.

natesworks, 2024-05-21:
Im using the normal kernel trying to enable hci0 shows file not found for some reason

Harroy11, 2024-05-21:
Arch, Linux 6.9.1 normal, bluez5.76, MediaTek PCIe adapter.
Bluetoothctl shows nothing.

natesworks, 2024-05-22:
I also have mediatek maybe related(?)

abhigyanmadhukalya, 2024-05-22:
I also have a MediaTek adapter for bluetooth and bluez 5.75, `bluetoothctl` shows nothing and `journalctl` shows no error.

Vudentz, 2024-05-22:
Anything on the traces? Perhaps the firmware is not being loaded correctly or something?

tetoNidan, 2024-05-22:
`sudo dmesg | grep -i bluetooth`

    Bluetooth: Core ver 2.22
    NET: Registered PF_BLUETOOTH protocol family
    Bluetooth: HCI device and connection manager initialized
    Bluetooth: HCI socket layer initialized
    Bluetooth: L2CAP socket layer initialized
    Bluetooth: SCO socket layer initialized
    bluetooth hci0: Direct firmware load for mediatek/BT_RAM_CODE_MT7961_1a_2_hdr.bin failed with error -2
    Bluetooth: hci0: Failed to load firmware file (-2)
    Bluetooth: hci0: Failed to set up firmware (-2)
    Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
    Bluetooth: BNEP (Ethernet Emulation) ver 1.3
    Bluetooth: BNEP filters: protocol multicast
    Bluetooth: BNEP socket layer initialized

Harroy11, 2024-05-23:
I got the exact same output

mllull, 2024-05-23:
The bug is posted (and solved) here: https://bbs.archlinux.org/viewtopic.php?id=295942
(more info no this bugzilla post: https://bugzilla.kernel.org/show_bug.cgi?id=218757).

Seems to be a bug in Kernel firmwares related to `MediaTek` (I also have it in my computer).

You can solve it by create a *symlink* as specified in first link or apply a Kernel patch.

This bug seems to be unrelated to *bluez*, so maybe this issue can be closed and keep this post as per documentation about this kernel bug.

natesworks, 2024-05-23:
why is no maintainer replying

natesworks, 2024-05-25:
Solution: sudo ln -s /lib/firmware/mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin.zst /lib/firmware/mediatek/BT_RAM_CODE_MT7961_1a_2_hdr.bin.zst

If this didn't work feel free to reopen the issue. Thanks @Vudentz for helping!

closed 2024-05-25 by natesworks

natesworks, 2024-05-26:
This will be fixed in kernel 6.9.2

StuffOfSonny, 2024-05-30:
Was not fixed in kernel 6.9.2
[image]

natesworks, 2024-05-31:
> Was not fixed in kernel 6.9.2
> [image]
> 

Well just use the comment I marked as the solution. Also your using the zen kernel.

StuffOfSonny, 2024-05-31:
> Well just use the comment I marked as the solution. Also your using the zen kernel.

Yes, as mentioned above

> Arch Linux 6.9.1-zen1-1-zen, I only have a brief moment of availability so that's all I could grab at the moment.

Anywho, I did try the symlink solution and while it did make the adapter show up in system settings, it would fail to scan for any devices.