bluez/bluez #850: Bluetooth Keyboard connectivity issue

narasimha00, 2024-05-23:
My bluetooth keyboard doesn't connect to my linux machine, it shows out this error in bluetoothctl .

`Failed to connect: org.bluez.Error.Failed br-connection-page-timeout`

It was working previously , but after installing recent updates, this has begun to happen ..

Vudentz, 2024-05-23:
> My bluetooth keyboard doesn't connect to my linux machine, it shows out this error in bluetoothctl .
> 
> `Failed to connect: org.bluez.Error.Failed br-connection-page-timeout`
> 
> It was working previously , but after installing recent updates, this has begun to happen ..

Well page timeout is usually a link-layer/controller thing, are you sure this is a regression? Or perhaps the controller firmware has been updated in the meantime?

narasimha00, 2024-10-13:
> Well page timeout is usually a link-layer/controller thing, are you sure this is a regression? Or perhaps the controller firmware has been updated in the meantime?

But, it works if I downgrade my bluez, libbluetooth3 and bluetooth packages to v 5.71.1 !

closed 2025-07-25 by github-actions[bot]