bluez/bluez #851: Bluez peripheral using DBus API always ask pairing request on Android

guirespi, 2024-05-23:
As the title says, when I setup a Gatt server using bluez, when connecting to the peripheral using an Android device, the pairing request is always asked. There won't be any problems but when I accept the pairing request, the device disconnects. This doesn't happen using an IPhone. This are the information I got using 'btmon'

``
> HCI Event: LE Meta Event (0x3e) plen 31                   #1 [hci0] 23.879919
      LE Enhanced Connection Complete (0x0a)
        Status: Success (0x00)
        Handle: 3585
        Role: Peripheral (0x01)
        Peer address type: Random (0x01)
        Peer address: 56:19:27:B2:F7:33 (Resolvable)
        Local resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Peer resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Connection interval: 30.00 msec (0x0018)
        Connection latency: 0 (0x0000)
        Supervision timeout: 5000 msec (0x01f4)
        Central clock accuracy: 0x01
@ MGMT Event: Device Connected (0x000b) plen 13       {0x0001} [hci0] 23.880050
        LE Address: BC:32:B2:50:CB:C8 (OUI BC-32-B2)
        Flags: 0x00000000
        Data length: 0
< HCI Command: LE Read Remote Used... (0x08|0x0016) plen 2  #2 [hci0] 23.880164
        Handle: 3585
> HCI Event: LE Meta Event (0x3e) plen 4                    #3 [hci0] 23.880811
      LE Channel Selection Algorithm (0x14)
        Handle: 3585
        Algorithm: #2 (0x01)
> HCI Event: LE Meta Event (0x3e) plen 6                    #4 [hci0] 23.881774
      LE Advertising Set Terminated (0x12)
        Status: Success (0x00)
        Handle: 1
        Connection handle: 3585
        Number of completed extended advertising events: 0
@ MGMT Command: Remove Advertising (0x003f) plen 1    {0x0001} [hci0] 23.882187
        Instance: 1
> HCI Event: Command Status (0x0f) plen 4                   #5 [hci0] 23.882784
      LE Read Remote Used Features (0x08|0x0016) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Remove Advertisin.. (0x08|0x003c) plen 1  #6 [hci0] 23.882822
        Handle: 1
> HCI Event: LE Meta Event (0x3e) plen 12                   #7 [hci0] 23.883780
      LE Read Remote Used Features (0x04)
        Status: Success (0x00)
        Handle: 3585
        Features: 0xff 0xf9 0x01 0xf7 0x8f 0x00 0x00 0x00
          LE Encryption
          Connection Parameter Request Procedure
          Extended Reject Indication
          Peripheral-initiated Features Exchange
          LE Ping
          LE Data Packet Length Extension
          LL Privacy
          Extended Scanner Filter Policies
          LE 2M PHY
          LE Coded PHY
          LE Extended Advertising
          LE Periodic Advertising
          Channel Selection Algorithm #2
          LE Power Class 1
          Minimum Number of Used Channels Procedure
          Periodic Advertising Sync Transfer - Sender
          Periodic Advertising Sync Transfer - Recipient
          Sleep Clock Accuracy Updates
          Connected Isochronous Stream - Central
          Connected Isochronous Stream - Peripheral
          Isochronous Broadcaster
          Synchronized Receiver
          Isochronous Channels (Host Support)
          Unknown features (0x0000008e00000000)
< ACL Data TX: Handle 3585 flags 0x00 dlen 6                #8 [hci0] 23.884021
      SMP: Security Request (0x0b) len 1
[... 213 lines cut ...]
< HCI Command: LE Set Extended Sc.. (0x08|0x0038) plen 10  #41 [hci0] 50.618596
        Handle: 0x01
        Operation: Complete scan response data (0x03)
        Fragment preference: Minimize fragmentation (0x01)
        Data length: 0x06
        Name (complete): BINC
> HCI Event: Command Complete (0x0e) plen 4                #42 [hci0] 50.620481
      LE Set Extended Scan Response Data (0x08|0x0038) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Adv.. (0x08|0x0039) plen 6  #43 [hci0] 50.620548
        Extended advertising: Enabled (0x01)
        Number of sets: 1 (0x01)
        Entry 0
          Handle: 0x01
          Duration: 0 ms (0x00)
          Max ext adv events: 0
> HCI Event: Command Complete (0x0e) plen 4                #44 [hci0] 50.622442
      LE Set Extended Advertising Enable (0x08|0x0039) ncmd 2
        Status: Success (0x00)
@ MGMT Event: Command Complete (0x0001) plen 4        {0x0001} [hci0] 50.622450
      Add Extended Advertising Data (0x0055) plen 1
        Status: Success (0x00)
        Instance: 1

I'm new using bluez for BLE applications so I don't have a clue of why this pairing request is always asked. I hope someone can help me with this problem.

guirespi, 2024-05-23:
Hi, thanks for the response. I'll try it. But, there is a way in which the Android 13 device doesn't act as master and doesn't initiate the pairing request?

Vudentz, 2024-05-23:
> Hi, thanks for the response. I'll try it. But, there is a way in which the Android 13 device doesn't act as master and doesn't initiate the pairing request?

Well I guess we need to ask Android folks what is going on there, since it seems they may have broken peripherals implemented using BlueZ.

vkurkchi, 2025-01-06:
@Bi11 you saved my day, Thank YOU!

closed 2025-07-25 by github-actions[bot]