bluez/bluez #855: initializer element is not constant

navinagaraj, 2024-05-29:
```
Architecture         -> aarch64         
Kernel Architecture  -> arm64           
Kernel Version       -> 5.10    
gcc                  -> 6.5.0
glibc                -> 2.33
```

**Build Error:**  [Bluez-5.70-build-error.txt](https://github.com/bluez/bluez/files/15482685/Bluez-5.70-build-error.txt)

bgix, 2024-05-30:
I will look at these today.

bgix, 2024-05-31:
Is there any way to use a version of GCC that is > v7.4? GCC is currently on v13 and versions lower than v7.4 (including v6.5.0) have an issue with this particular warning. Upgrading to a newer GCC version should fix your problem.

In any case, this is a unit test module, which is not used in deployed solutions.

navinagaraj, 2024-06-03:
which module need to disable to build the bluez package on gcc-6.5.0 ?

bgix, 2024-06-03:
I would probably just:

$ make -k

Which lets make keep going even after encountering any kind of build error.
It builds everything it can that doesn’t have dependency build errors.
Since the file you are complaining about is part of the test engine,
probably all the important things will still get built.

closed 2025-07-25 by github-actions[bot]