bluez/bluez #856: Host initiating remove device command resulting device removed from Pair list and LE link also got disconnected.

maheshtalewad, 2024-05-31:
Hi,

Issue:
Once pairing got successful during LE outgoing connection from DUT, Host initiating remove device command resulting device removed from Pair list and LE link also got disconnected.

This issue is Linux Kernel regression issue.

**Linux BSP used: 6.6.23-lts-next-06207-g4aef2658ac28
Regression from BSP: 6.6.3**

[Logs.zip](https://github.com/user-attachments/files/15513648/Logs.zip)

Frequency:5/5.
Steps to reproduce:

1.Flash mentioned BSP on the DUT.
2. Turn On bluetoothctl and then turn on BT/BLE scan using command "scan on".
3. Set Advertise On in nRF App in the Ref device.
4. Pair Ref device in bluetoothctl from DUT.
5. LE connection can be observed in bluetoothctl using command "devices Connected".
6. In Ref device in nRF app it can be observed.
7.After some time device is removed from pair list and LE Link got disconnected.

Actual Observation: After connection established, Ref  paired device removed from the pair list and LE connection got disconnected.
Expected Observation: DUT should successfully sustain LE outgoing connection and ref device should not be 
removed from pair list.

**Below CL is causing this issue. This CL is in kernel space**
 
 757 commit 865f1f43262cff2cabb07ac7c3b5ed89e283a0eb
 758 Author: Xiao Yao <xiaoyao@rock-chips.com>
 759 Date:  Tue Dec 12 00:27:18 2023 +0800
 760
 761    Bluetooth: MGMT/SMP: Fix address type when using SMP over BREDR/LE
 762
 763    commit 59b047bc98084f8af2c41483e4d68a5adf2fa7f7 upstream.
 764
 765    If two Bluetooth devices both support BR/EDR and BLE, and also
 766    support Secure Connections, then they only need to pair once.
 767    The LTK generated during the LE pairing process may be converted
 768    into a BR/EDR link key for BR/EDR transport, and conversely, a
 769    link key generated during the BR/EDR SSP pairing process can be
 770    converted into an LTK for LE transport. Hence, the link type of
 771    the link key and LTK is not fixed, they can be either an LE LINK
 772    or an ACL LINK.
 773
 774    Currently, in the mgmt_new_irk/ltk/crsk/link_key functions, the
 775    link type is fixed, which could lead to incorrect address types
 776    being reported to the application layer. Therefore, it is necessary
 777    to add link_type/addr_type to the smp_irk/ltk/crsk and link_key,
 778    to ensure the generation of the correct address type.
 779
 780    SMP over BREDR:
 781    Before Fix:
 782    > ACL Data RX: Handle 11 flags 0x02 dlen 12
 783             BR/EDR SMP: Identity Address Information (0x09) len 7
 784             Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
 785    @ MGMT Event: New Identity Resolving Key (0x0018) plen 30
 786             Random address: 00:00:00:00:00:00 (Non-Resolvable)
 787             LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
 788    @ MGMT Event: New Long Term Key (0x000a) plen 37
 789             LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
 790             Key type: Authenticated key from P-256 (0x03)
 791
 792    After Fix:
 793    > ACL Data RX: Handle 11 flags 0x02 dlen 12
 794          BR/EDR SMP: Identity Address Information (0x09) len 7
 795             Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
 796    @ MGMT Event: New Identity Resolving Key (0x0018) plen 30
 797             Random address: 00:00:00:00:00:00 (Non-Resolvable)
 798             BR/EDR Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
 799    @ MGMT Event: New Long Term Key (0x000a) plen 37
 800             BR/EDR Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
 801             Key type: Authenticated key from P-256 (0x03)
 802
 803    SMP over LE:
 804    Before Fix:
 805    @ MGMT Event: New Identity Resolving Key (0x0018) plen 30
 806             Random address: 5F:5C:07:37:47:D5 (Resolvable)
 807             LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
 808    @ MGMT Event: New Long Term Key (0x000a) plen 37
 809             LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
 810             Key type: Authenticated key from P-256 (0x03)
 811    @ MGMT Event: New Link Key (0x0009) plen 26
 812             BR/EDR Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
 813             Key type: Authenticated Combination key from P-256 (0x08)
 814
 815    After Fix:
 816    @ MGMT Event: New Identity Resolving Key (0x0018) plen 30
 817             Random address: 5E:03:1C:00:38:21 (Resolvable)
 818             LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
 819    @ MGMT Event: New Long Term Key (0x000a) plen 37
 820             LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
 821             Key type: Authenticated key from P-256 (0x03)
 822    @ MGMT Event: New Link Key (0x0009) plen 26
 823             Store hint: Yes (0x01)
 824             LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
 825             Key type: Authenticated Combination key from P-256 (0x08)
 826
 827    Cc: stable@vger.kernel.org
 828    Signed-off-by: Xiao Yao <xiaoyao@rock-chips.com>
 829    Signed-off-by: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>
 830    Signed-off-by: Greg Kroah-Hartman <gregkh@linuxfoundation.org>

Thanks and regards,
Mahesh Vithal Talewad

Vudentz, 2024-06-13:
This seems to be with BlueZ 5.72, any chance to update to something more recent like 5.76 to check if that helps?

Vudentz, 2024-06-13:
@maheshtalewad try with the above change.

maheshtalewad, 2024-06-14:
Hi Luiz,
I tried with above changes, issue did not reproduce.
Thank you.

Thanks and regards,
Mahesh Vithal Talewad

closed 2024-06-14 by BluezTestBot