bluez/bluez #858: After bluez 5.66 -> 5.72 upgrade, multipoint prioritization is working incorrectly

runderwo, 2024-06-01:
I have a Mac and a HP Ubuntu laptop that share a Bluetooth headset.  The headset has volume and playback controls on it. 

experimental = True is set in main.conf only to enable battery life reporting.  All the other configuration is default.

With bluez 5.66, the headset playback controls controlled the last device that played audio to it unless it no longer had an active media player application.  This is the expected behavior.  There were no other anomalies.

After upgrading the HP Ubuntu laptop to bluez 5.72 (with an Ubuntu upgrade), the headset playback controls only control the Mac if it's currently or has very recently been playing audio (i.e. to pause the stream).  If more than a few minutes passes, attempting to resume the Mac stream with the headset control actually causes the HP laptop to play whatever media player context is focused.  It's as if the HP laptop is continuously grabbing priority.  Even if there is no media player context on the HP laptop, pressing resume on the headset causes the screen to unblank and a "cancel" icon to be briefly displayed either on the lock screen or over the desktop.

Another anomaly with the new stack that may be related - not present in 5.66: even when playback is resumed on the Mac successfully (because the time since pausing was short enough), the HP laptop unblanks the screen as if it's received input.

Vudentz, 2024-06-08:
> I have a Mac and a HP Ubuntu laptop that share a Bluetooth headset. The headset has volume and playback controls on it.
>
> experimental = True is set in main.conf only to enable battery life [...]

Are you sure this is due to upgrading BlueZ alone, BlueZ is not the entity that actually stream the audio, it is either pipewire or pulseaudio and I noticed the pipewire has a different behavior since it appears it doesn't sit idle like pulseaudio, so it doesn't suspend the stream.

Anyway the multi-point was never really covered that well with existing AVRCP implementation, so it is not surprise it doesn't always work if it depends on behavior of each central connected to the headset.

runderwo, 2024-06-09:
No, I'm not sure it's due to upgrading bluez alone.  I'm happy to chase it in another direction, but rather than an audio streaming issue this seemed to be more on the side of a "control plane" issue and seemed reasonable to start here - in case major changes were made to multipoint between the (previous) working and (current) incorrectly-working versions..

pv, 2024-06-09:
If you're running pipewire, you can use `pw-top` command to see if some application is keeping the device busy, which could prevent multipoint from switching.

runderwo, 2024-06-15:
Thanks, that helped me identify that the problem was occurring when pavucontrol is open, since it's listed as a client of pipewire in `R` (RUNNING) state.

```
S   ID  QUANT   RATE    WAIT    BUSY   W/Q   B/Q  ERR FORMAT           NAME                                                                                                                                                                                           
S   29      0      0    ---     ---   ---   ---     0                  Dummy-Driver
S   30      0      0    ---     ---   ---   ---     0                  Freewheel-Driver
S   42      0      0    ---     ---   ---   ---     0                  Midi-Bridge
S   50      0      0    ---     ---   ---   ---     0                  v4l2_input.pci-0000_00_14.0-usb-0_4_1.0
R   52   1024  48000 105.2us  32.8us  0.00  0.00   35    S32LE 2 48000 alsa_output.pci-0000_00_1f.3.analog-stereo
R   86      1     25  47.6us  16.5us  0.00  0.00    0       F32LE 1 25  + PulseAudio Volume Control
R   53   1024  48000 107.2us   0.8us  0.01  0.00    0    S32LE 2 48000 alsa_input.pci-0000_00_1f.3.analog-stereo
R   83      1     25  51.8us  16.3us  0.00  0.00    0       F32LE 1 25  + PulseAudio Volume Control
```

After closing pavucontrol, it disappears from the list and the headset controls are no longer interpreted as an attempt to control the Ubuntu laptop.  This hasn't changed on my end since previously - I always have pavucontrol running as a way to quickly route app outputs and manage relative volumes of app sounds.

I guess pavucontrol is registering as a sink in order to display the volume meters - I found this old issue where pavucontrol was crashing an older version of pipewire: https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/3320

A further guess is that in the case of pavucontrol, the current version of pipewire doesn't observe the difference between a sink which is passively snooping on volumes and a source which should be controlled with headset controls.

It sounds like I need to follow up on the pipewire side unless anyone else has guidance for me here.

pv, 2024-06-15:
That volume monitor streams activate sinks may be a behavior difference between PW and PA. Open an issue in Pipewire gitlab.

closed 2024-06-22 by runderwo