bluez/bluez #867: missing ~/phonebook/telecom/ causes connection drop

kolAflash, 2024-06-08:
When I connect my notebook to my car as described [here](https://www.reddit.com/r/pipewire/comments/1d9391u/play_music_on_my_cars_speakers_via_bluetooth_from/), the connection drops after about 1 minute and 10 seconds.  
Notebook: `HP EliteBook 845 G8`, `Ryzen-5650U`, `Debian-12`, `bluez-obexd-5.66 by Debian`  
Car: [Volkswagen ID.3](https://en.wikipedia.org/wiki/Volkswagen_ID.3), `firmware 3.2`

`/var/log/syslog` reports:  
`obexd[*]: stat(/home/myuser/phonebook/telecom/XXX): No such file or directory (2)`  
`obexd[*]: open(/telecom/XXX.vcf): No such file or directory (2)`  
Where `XXX` is one of these: `cch ich mch och pb`  

Creating just `~/phonebook/telecom/` fixed the problem. _(subdirectories and vcf files not needed)_  
Now my car stays connected.  

I'm not sure what happens inside of obexd. But I propose that obexd should continue working without that directory in the same way as with that directory. Or alternatively it should create that directory if it's missing.

<details>
<summary>full log</summary>

```
2024-06-08T23:14:46.952669+02:00 myhost wireplumber[1539]: RFCOMM receive command but modem not available: AT+CREG?#015
2024-06-08T23:14:47.024897+02:00 myhost wireplumber[1539]: RFCOMM receive command but modem not available: AT+NREC=0#015
2024-06-08T23:14:47.420654+02:00 myhost kernel: [  957.105284][  T888] input: STRANDWAGEN (AVRCP) as /devices/virtual/input/input34
2024-06-08T23:14:47.549097+02:00 myhost wireplumber[1539]: RFCOMM receive command but modem not available: AT+CCWA=1#015
2024-06-08T23:14:47.613916+02:00 myhost obexd[1930]: CONNECT(0x0), <unknown>(0xff)
2024-06-08T23:14:47.622143+02:00 myhost obexd[1930]: CONNECT(0x0), <unknown>(0x0)
2024-06-08T23:14:47.665801+02:00 myhost obexd[1930]: CONNECT(0x0), <unknown>(0xff)
2024-06-08T23:14:47.666101+02:00 myhost obexd[1930]: CONNECT(0x0), <unknown>(0x0)
2024-06-08T23:14:47.677426+02:00 myhost obexd[1930]: SETPATH(0x5), <unknown>(0xff)
2024-06-08T23:14:47.677759+02:00 myhost obexd[1930]: stat(/home/moritz/phonebook/telecom): No such file or directory (2)
2024-06-08T23:14:47.677982+02:00 myhost obexd[1930]: SETPATH(0x5), Not Found(0x44)
2024-06-08T23:14:47.685024+02:00 myhost obexd[1930]: SETPATH(0x5), <unknown>(0xff)
2024-06-08T23:14:47.685212+02:00 myhost obexd[1930]: SETPATH(0x5), Not Found(0x44)
2024-06-08T23:14:47.740845+02:00 myhost obexd[1930]: GET(0x3), <unknown>(0xff)
2024-06-08T23:14:47.741051+02:00 myhost obexd[1930]: open((null)): Function not implemented (38)
2024-06-08T23:14:47.741122+02:00 myhost obexd[1930]: GET(0x3), Not Implemented(0x51)
2024-06-08T23:14:47.817024+02:00 myhost wireplumber[1539]: RFCOMM receive command but modem not available: AT+CSCS=?#015
2024-06-08T23:14:47.868171+02:00 myhost bluetoothd[888]: /org/bluez/hci0/dev_00_00_00_00_00_00/sep2/fd9: fd(42) ready
2024-06-08T23:15:47.621949+02:00 myhost obexd[1930]: DISCONNECT(0x1), <unknown>(0xff)
2024-06-08T23:15:47.623115+02:00 myhost obexd[1930]: DISCONNECT(0x1), Success(0x20)
2024-06-08T23:15:47.677268+02:00 myhost obexd[1930]: disconnected: Transport got disconnected
2024-06-08T23:15:47.678164+02:00 myhost bluetoothd[888]: src/profile.c:ext_io_disconnected() Unable to get io data for Message Access: getpeername: Transport endpoint is not connected (107)
2024-06-08T23:16:01.239487+02:00 myhost obexd[1930]: disconnected: Transport got disconnected
2024-06-08T23:16:01.301441+02:00 myhost bluetoothd[888]: src/profile.c:ext_io_disconnected() Unable to get io data for Phone Book Access: getpeername: Transport endpoint is not connected (107)
2024-06-08T23:16:01.302198+02:00 myhost bluetoothd[888]: src/profile.c:ext_io_disconnected() Unable to get io data for Hands-Free Voice gateway: getpeername: Transport endpoint is not connected (107)
```
</details>

kolAflash, 2024-06-09:
P.S.  
**Copy of my Reddit post linked above.**  
_(Reddit somehow removed it 🤪)_

&nbsp;

==========

&nbsp;

I've got a notebook with Debian-12 & Pipewire-0.3.65, an Android-12 phone and a [Volkswagen ID.3](https://en.wikipedia.org/wiki/Volkswagen_ID.3) car running the firmware version 3.2.

I can easily connect my phone with the car via Bluetooth and play music from any app. And I can also connect my notebook with the car via Bluetooth. But it won't play audio.

&nbsp;

When the car is connected to the notebook, it shows up in Pipewire. If I open [Pavucontrol](https://packages.debian.org/bookworm/pavucontrol) I can see the car under *"Configuration"* offering *"High Fidelity Playback (*[A2DP](https://en.wikipedia.org/wiki/List_of_Bluetooth_profiles#Advanced_Audio_Distribution_Profile_(A2DP))*, Sink)" (SBC & SBC-XQ)* and *"Headset Head Unit (HSP/HFP)" (mSBC & CVSD)*. In *"Output Devices"* the car has a *"Auto (plugged in)"* port.

I tried playing music with different programs like VLC or Firefox. But there simply no sound coming out of my cars speakers. Also the Bluetooth connection drops after about one minute.

&nbsp;

There's on difference visible in the car when connecting the phone or the notebook. As soon as the phone's connected, the car's display shows an entry *"Bluetooth Player"* and when I click that it shows *"VLC"* under which all my audio files on the phone are listed and I can start the playback of a file from the car's display without touching my phone. *(I got* [VLC for Android](https://f-droid.org/packages/org.videolan.vlc/) *installed on my phone)*

Starting the playback from the phone is also possible. In that case any app I've tested works. *(tested VLC, Vanilla Player, Spotify and others)*

But when connecting my notebook, the the *"Bluetooth Player"* entry is missing on the car's display. My guess is, that the car is expecting some additional protocol beside A2DP via the Bluetooth connection. Something like a [DLNA Media Server (DMS)](https://en.wikipedia.org/wiki/DLNA) but for Bluetooth.

&nbsp;

Any ideas?  
Thanks! :-)

rukmal, 2024-09-24:
+1 on this, I'm running into this issue on NixOS too.

kolAflash, 2025-05-12:
I'd suggest to keep this open. But I can't remove the `stale` label.

I would think, that the fix is probably be pretty easy and would save users a lot of worrying.