bluez/bluez #868: mpris-proxy.service

kolAflash, 2024-06-08:
Some Bluetooth audio output devices ("speakers") don't seem to work without `AVRCP`, so they need `mpris-proxy` running. See [here](https://www.reddit.com/r/pipewire/comments/1d9391u/comment/l7qh62x/) for example.

I'm running a Debian-12 system and I needed to start `mpris-proxy` as [described in the Arch Linux wiki](https://wiki.archlinux.org/index.php?title=MPRIS&oldid=805916#Bluetooth) to make my Bluetooth "speaker" work.

**So I'm proposing to add: `/usr/lib/systemd/user/mpris-proxy.service`**

Vudentz, 2024-06-12:
I guess we could have a service like that, Ive been planning to integrate support for mpris players to bluetoothctl, that said it doesn't change much in terms of creating a service that proxy MPRIS players to be accessible via AVRCP/MCP profiles.

joakim-tjernlund, 2024-06-20:
Maybe somthing like this:
```
~/.config/systemd/user/mpris-proxy.service

[Unit]
Description=Forward bluetooth media controls to MPRIS

[Service]
Type=simple
ExecStart=/usr/bin/mpris-proxy

[Install]
WantedBy=default.target
```
Found at https://wiki.archlinux.org/title/MPRIS

kolAflash, 2024-06-20:
> Maybe somthing like this:
> [...]
> Found at https://wiki.archlinux.org/title/MPRIS

@joakim-tjernlund That's exactly what I'm proposing :-)

joakim-tjernlund, 2024-06-20:
> > Maybe somthing like this:
> > [...]
> > Found at https://wiki.archlinux.org/title/MPRIS
> 
> @joakim-tjernlund That's exactly what I'm proposing :-)

ahh, right you are. Didn't check your URL. Sorry about the noise
Make a patch and send it to list?

paulmenzel, 2025-04-26:
Fixed by commit daa86e06c376d (tools/mpris-proxy: Add systemd user unit)?

closed 2025-09-10 by github-actions[bot]