bluez/bluez #869: bluepy3 under BlueZ

Mausy5043, 2024-06-09:
Hello `bluez` devs,

### Introduction
I've been working on [bluepy3](https://github.com/Mausy5043/bluepy3) since february 2023.
`bluepy3` is a Python3 interface to Bluetooth LE on Linux. I based this on previous work by [Ian Harvey](https://github.com/IanHarvey/bluepy) whose repository is based on Python2 and hadn't seen activity for several years. The package is working fine on a Raspberry Pi w/ Debian Linux and within the limited use-cases that I can test. The [Raspberry Pi](https://www.raspberrypi.com/) platform, to which `bluepy3` is currently optimized, is rapidly growing as a popular IoT and desktop platform where bluetooth is, obviously, a major connectivity tool.

### Current State
`bluepy3` is currently being worked on only by me. I have limited experience with BT and BLE and limited resources (time & hardware) to test the stack in different scenarios.

### Future State
I would like to see `bluepy3` being adopted and maintained by the BlueZ team. This would help boost the package's visibility to the community and future-proof & improve the support of BT and BLE to Python3 and IoT platforms. 

### Request
Are you open to discuss the idea of transferring development and support of [bluepy3](https://github.com/Mausy5043/bluepy3) under the [bluez organisation](https://github.com/bluez) ?

Vudentz, 2024-06-12:
We already have a sort of official binding: https://github.com/bluez/pybluez

That said, that hasn't seem much activities in past 2 years, so Im open to switch to bluepy if that has more traction with python devs.

Mausy5043, 2024-06-17:
Cool. I'd love to discuss the path forward. What state would you need bluepy3 to be in to fork it over?

bgix, 2024-06-17:
Will the bluepy3 bindings support Bluetooth Mesh as well?

Mausy5043, 2024-06-20:
`bluepy3` currently does not support BT Mesh. I see no reason why this couldn't be added in the future though. 

But as I mentioned I have limited knowledge of BT and also limited resources and the package has come to a point where those limitations start hurting further development.