bluez/bluez #870: Failing to connect any bluetooth devices with latest version

svartkanin, 2024-06-09:
I've encountered a error with the latest version 5.76, I'm unable to connect to any bluetooth device at this point. 
System: Arch Linux
Kernel: 6.6.32-1-lts
bluetoothctl: 5.76

I tried to reload `btusb` but no success.

```
$ rmmod btusb
$ modprobe btusb
$ systemctl start bluetooth.service
$ systemctl enable bluetooth.service
$ systemctl status bluetooth.service
$ systemctl status bluetooth.service 
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: disabled)
     Active: active (running) since Sun 2024-06-09 22:08:45 AEST; 1min 22s ago
       Docs: man:bluetoothd(8)
   Main PID: 10855 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 37653)
     Memory: 904.0K (peak: 1.3M)
        CPU: 22ms
     CGroup: /system.slice/bluetooth.service
             └─10855 /usr/lib/bluetooth/bluetoothd

Jun 09 22:08:45 nazgul bluetoothd[10855]: Endpoint registered: sender=:1.162 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_1
Jun 09 22:08:45 nazgul bluetoothd[10855]: Endpoint registered: sender=:1.162 path=/MediaEndpoint/A2DPSource/aptx_ll_duplex_0
Jun 09 22:08:45 nazgul bluetoothd[10855]: Endpoint registered: sender=:1.162 path=/MediaEndpoint/A2DPSource/faststream
Jun 09 22:08:45 nazgul bluetoothd[10855]: Endpoint registered: sender=:1.162 path=/MediaEndpoint/A2DPSource/faststream_duplex
Jun 09 22:08:45 nazgul bluetoothd[10855]: Endpoint registered: sender=:1.162 path=/MediaEndpoint/A2DPSink/opus_05
Jun 09 22:08:45 nazgul bluetoothd[10855]: Endpoint registered: sender=:1.162 path=/MediaEndpoint/A2DPSource/opus_05
Jun 09 22:08:45 nazgul bluetoothd[10855]: Endpoint registered: sender=:1.162 path=/MediaEndpoint/A2DPSink/opus_05_duplex
Jun 09 22:08:45 nazgul bluetoothd[10855]: Endpoint registered: sender=:1.162 path=/MediaEndpoint/A2DPSource/opus_05_duplex
Jun 09 22:08:45 nazgul bluetoothd[10855]: Failed to load LTKs for hci0: Invalid Parameters (0x0d)
Jun 09 22:08:45 nazgul bluetoothd[10855]: Failed to load IRKs for hci0: Invalid Parameters (0x0d)
```

These are the logs whenever a device tries to connect 
``` 
$ journalctl -u bluetooth
Jun 09 22:03:14 nazgul bluetoothd[9446]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
Jun 09 22:03:14 nazgul bluetoothd[9446]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Jun 09 22:03:14 nazgul bluetoothd[9446]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Jun 09 22:03:14 nazgul bluetoothd[9446]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
```

svartkanin, 2024-06-11:
I managed to fix it by deleting existing devices and setting them up again

closed 2024-06-11 by svartkanin

blockbabak, 2024-10-12:
That is not a solution. 
There must be a reason behind the error.
A solution is one which explains why the error has come about and the solution to it that resolves the cause.
No one learns anything from this issue. It should have been removed rather than marked closed.

I have the same error but deleting and pairing the devices doesn't change anything.

svartkanin, 2024-10-12:
It was finally fixed in an update last month, I don't know what the package was that actually fixed it though, bluedevil, kde or something else