bluez/bluez #871: Logitech MX Anywhere 3S won't reconnect or detected

Flashwalker, 2024-06-10:
I had a problem with **Logitech Anywhere 3S** mouse not being detected on the laptop's **built-in Bluetooth adapter** (it worked fine for the long time on the first mouse slot `D3:DF:1C:10:72:74`) after rebooting in Windows and connecting this mouse on its second slot and on the third slot for my Android phone.

After that in my Linux system the mouse did not connect to the previously paired slot 1 `D3:DF:1C:10:72:74` neither automatically nor manually, so I deleted this device in **Blueman manager gui** and tried to scan and re-bind it to the first slot as it was before (long press on the mouse connect button until the light blinks rapidly (mouse available for new pairing via new mac address for this slot) and scanning for devices in **Blueman manager gui**).

The device remained undetectable for no reason no matter what I did. I tried to scan via command in terminal as well, reboot, but no luck. The mouse wasn't found.

But if I plugged another **external Bluetooth usb dongle** and scanagain, the mouse was detected and via this dongle and pairing works fine (on the first slot with a new mac `D3:DF:1C:10:72:78`) without any problem.

After one day of use on the **external Bluetooth usb dongle**, I rebooted (or maybe not, I don't remember), and tried scanning again on the laptop's **built-in Bluetooth adapter** - and miraculously this time the mouse was detected and pairingworked fine to the first slot with a new mac `D3:DF:1C:10:72:79`.

---

But...
If I reboot again in Windows and connect the mouse to paired slot 2 (via **built-in Bluetooth adapter**), and then reboot back in Linux - the mouse will not connect to paired slot 1 again (of course when I do this, I switch the mouse to the paired slot 1).
I get error:

```
$ bluetoothctl connect D3:DF:1C:10:72:79
Attempting to connect to D3:DF:1C:10:72:79
Failed to connect: org.bluez.Error.Failed le-connection-abort-by-local
```

Then if i'll delete this device (mouse) and will try to scan again (slot 1) (with long press of the mouse connection button the mouse just will gain new mac address e.g.: `D3:DF:1C:10:72:7A` for that slot and became available for the new pairing) - the mouse won't be detected at all again. And rebooting doesn't help.

(but with **external Bluetooth usb dongle** connection works fine every time)

Looks like a bug.

Here is my **built-in Bluetooth adapter**:

```
 $ hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: 14:AC:60:49:23:AC  ACL MTU: 1021:6  SCO MTU: 240:8
	UP RUNNING PSCAN 
	RX bytes:1050 acl:0 sco:0 events:97 errors:0
	TX bytes:5261 acl:0 sco:0 commands:97 errors:0
	Features: 0xbf 0x3e 0x8d 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'laptop'
	Class: 0x7c010c
	Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
	Device Class: Computer, Laptop
	HCI Version: 5.2 (0xb)  Revision: 0x612
	LMP Version: 5.2 (0xb)  Subversion: 0x2301
	Manufacturer: MediaTek, Inc. (70)

 $ rfkill
ID TYPE      DEVICE                 SOFT      HARD
 0 wlan      ideapad_wlan      unblocked unblocked
 1 bluetooth ideapad_bluetooth unblocked unblocked
 2 bluetooth hci0              unblocked unblocked
 3 wlan      phy0              unblocked unblocked
 
 $ lsusb
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 004: ID 048d:c103 Integrated Technology Express, Inc. ITE Device(8910)
Bus 005 Device 003: ID 0489:e0d8 Foxconn / Hon Hai Wireless_Device
Bus 005 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 174f:246a Syntek Integrated Camera
Bus 003 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 152d:0583 JMicron Technology Corp. / JMicron USA Technology Corp. JMS583Gen 2 to PCIe Gen3x2 Bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

---

bluez 5.64

Vudentz, 2024-06-13:
@Flashwalker looks like the btmon was cut before:

```
@ MGMT Event: Connect Failed (0x000d) plen 8                                                                                                                                      {0x0001} [hci0] 40.142133
        LE Address: D3:DF:1C:10:72:79 (Static)
        Status: Disconnected (0x0e)
```

We probably want to check if it was due to link-layer that it is disconnecting.

Flashwalker, 2024-06-27:
@Vudentz 
I just tested and verified that suspending the laptop helps.
As I wrote before, if I suspend and resume the laptop, my Anywhere mouse successfully connects via the built-in adapter. It's strange that suspending does this magic.

Flashwalker, 2024-07-27:
I've found that just plugging in an external USB bluetooth dongle helps sometimes too.
Somehow it triggers the internal bluetooth adapter to work and the mouse automatically connects to the internal adapter.

applemodus, 2024-08-16:
Hi!

Also does not work on fedora 40, x64

I have made a successful connection to my Lenovo thinkbook and after a few hours it stops connecting and wont reconnect. Even when I reset my mouse book doesn't "see" it.

Flashwalker, 2024-08-17:
In my case, the problem disappeared when I switched to a different laptop but still using the same OS.
It seems that it was a hardware related problem.

applemodus, 2024-08-18:
> Hi!
> 
> Also does not work on fedora 40, x64
> 
> I have made a successful connection to my Lenovo thinkbook and after a few hours it stops connecting and wont reconnect. Even when I reset my mouse book doesn't "see" it.

After a few days it works fine. But when I made a reboot it stopped connecting once again

stimut, 2025-05-11:
This is still an issue for me

dexif, 2025-06-25:
I have the same problem

DevenBL, 2025-06-27:
> This issue is stale because it has been open 30 days with no activity. Remove stale label or comment or this will be closed in 5 days.

What even is the point of these things. It's not like the issue goes away if you close it.

closed 2025-09-21 by github-actions[bot]