bluez/bluez #875: Logitech MX Master 3 doesn't connect in LE after a reboot

toxn, 2024-06-12:
Each movement of the mouse triggers a connection that ends after about 1s.

bluetooth.service gives an error : 

```shell-session
> systemctl status bluetooth
● bluetooth.service - Bluetooth service
     Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: enabled)
    Drop-In: /usr/lib/systemd/system/service.d
             └─10-timeout-abort.conf
     Active: active (running) since Wed 2024-06-12 10:17:23 CEST; 23s ago
       Docs: man:bluetoothd(8)
   Main PID: 8787 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 18841)
     Memory: 812.0K (peak: 5.0M)
        CPU: 54ms
     CGroup: /system.slice/bluetooth.service
             └─8787 /usr/libexec/bluetooth/bluetoothd

juin 12 10:17:27 dla.msol bluetoothd[8787]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
juin 12 10:17:27 dla.msol bluetoothd[8787]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
juin 12 10:17:27 dla.msol bluetoothd[8787]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
juin 12 10:17:27 dla.msol bluetoothd[8787]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
juin 12 10:17:27 dla.msol bluetoothd[8787]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
juin 12 10:17:27 dla.msol bluetoothd[8787]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
juin 12 10:17:27 dla.msol bluetoothd[8787]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
juin 12 10:17:27 dla.msol bluetoothd[8787]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
juin 12 10:17:27 dla.msol bluetoothd[8787]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
juin 12 10:17:27 dla.msol bluetoothd[8787]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
```

I disabled bluetooth.service and launched the daemon with `--debug=*` to get a more precise log in [bluetooth_log_2024-06-12.txt](https://gist.github.com/toxn/9af3ff40bd3cbd33e04025f90d57cdbd)

The only working solution for me is to remove the pairing and fully recreate it at each startup.

toxn, 2024-06-12:
OS / Installation informations:

```shell-session
> cat /etc/os-release
NAME="Fedora Linux"
VERSION="40 (KDE Plasma)"
ID=fedora
VERSION_ID=40
VERSION_CODENAME=""
PLATFORM_ID="platform:f40"
PRETTY_NAME="Fedora Linux 40 (KDE Plasma)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:40"
DEFAULT_HOSTNAME="fedora"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f40/system-administrators-guide/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=40
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=40
SUPPORT_END=2025-05-13
VARIANT="KDE Plasma"
VARIANT_ID=kde
```

```shell-session
> dnf list --installed \*bluez\*    
Paquets installés
bluez.x86_64                                                                                 5.76-1.fc40                                                                             @updates 
bluez-cups.x86_64                                                                            5.76-1.fc40                                                                             @updates 
bluez-libs.i686                                                                              5.76-1.fc40                                                                             @updates 
bluez-libs.x86_64                                                                            5.76-1.fc40                                                                             @updates 
bluez-obexd.x86_64                                                                           5.76-1.fc40                                                                             @updates 
kf5-bluez-qt.x86_64                                                                          5.115.0-1.fc40                                                                          @anaconda
kf6-bluez-qt.x86_64                                                                          6.3.0-1.fc40                                                                            @updates
```

toxn, 2024-06-12:
Hardware / driver infos:

```shell-session
$ lsmod | grep -E "(blue|bt)"
dell_rbtn              20480  0
btusb                  86016  0
btrtl                  36864  1 btusb
btintel                61440  1 btusb
btbcm                  24576  1 btusb
btmtk                  12288  1 btusb
bluetooth            1064960  34 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm
rfkill                 40960  11 iwlmvm,bluetooth,dell_laptop,dell_rbtn,cfg80211
```

Vudentz, 2024-06-12:
@toxn please collect the HCI traces with btmon.

Vudentz, 2024-06-12:
I starting to suspect this is due to PPCP parameters not being supported or something:

```
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 42                                                #2727 [hci0] 13:38:12.699649
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Random (0x01)
        Peer address: FE:D5:D9:EC:AB:99 (Static)
        Initiating PHYs: 0x03
        Entry 0: LE 1M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 7.50 msec (0x0006)
          Max connection interval: 4000.00 msec (0x0c80)
          Connection latency: 0 (0x0000)
          Supervision timeout: 5000 msec (0x01f4)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
        Entry 1: LE 2M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 7.50 msec (0x0006)
          Max connection interval: 4000.00 msec (0x0c80)
          Connection latency: 0 (0x0000)
          Supervision timeout: 5000 msec (0x01f4)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                                         #2728 [hci0] 13:38:12.700540
      LE Extended Create Connection (0x08|0x0043) ncmd 1
        Status: Invalid HCI Command Parameters (0x12)
```

But it works when pairing because the parameters used are different:

```
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 42                                                #5065 [hci0] 13:56:09.116312
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Random (0x01)
        Peer address: FE:D5:D9:EC:AB:99 (Static)
        Initiating PHYs: 0x03
        Entry 0: LE 1M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 30.00 msec (0x0018)
          Max connection interval: 50.00 msec (0x0028)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
        Entry 1: LE 2M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 30.00 msec (0x0018)
          Max connection interval: 50.00 msec (0x0028)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                                         #5066 [hci0] 13:56:09.117304
      LE Extended Create Connection (0x08|0x0043) ncmd 2
        Status: Success (0x00)
```

toxn, 2024-06-14:
I'm sorry, the problem has disappeared on itself, albeit I didn't update anything buez-related. (Although I did reinstall bluez, but the same version)
I'm unable to reproduce the bug.

I don' see any relevant update in my history apart maybe `selinux-policy`.

christophschw, 2024-06-21:
I do face this bug with a Logitech M720 under Ubuntu 24.04. So, this is relevant to me at least.

ifsand, 2024-06-23:
i had the same issue. It starts when I pair my phone (Redmi note 11 pro 5g android 13) . After some tests i found the problem disappears when i remove my phone from the paired devices.

grealish, 2024-07-22:
Still an issue, I can quickly reproduce this across many Logitech mice, MX 3, MX anywhere etc.. it's not specific to Logi, but something really broken.  I've testing both ` 5.77-1.fc40` and downgraded to ` 5.73-3.fc40` but still an issue 

```
Installed Packages
bluez.x86_64                                                                                                        5.73-3.fc40                                                                                                        @fedora
Available Packages
bluez.x86_64                                                                                                        5.73-3.fc40                                                                                                        fedora 
bluez.x86_64                                                                                                        5.77-1.fc40                                                                                                        updates
```

Hope the logs can help a little here, 

```
l 22 18:31:45 bluetoothd[6668]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Jul 22 18:31:45  bluetoothd[6668]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Jul 22 18:31:45 bluetoothd[6668]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Jul 22 18:31:45 bluetoothd[6668]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Jul 22 18:31:45 bluetoothd[6668]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
Jul 22 18:31:45 bluetoothd[6668]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Jul 22 18:31:45 bluetoothd[6668]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
```

only work around is to remove the pair device and re-pair, then it works until the laptop sleeps or reboots, so number of times a day you need to re-pair

Vudentz, 2024-07-22:
@grealish sounds liek a pairing issue then, perhaps attach the HCI trace (using btmon -w).

grealish, 2024-07-23:
Hi @Vudentz  - added now the output, it write a binary file, do you need that too?

<details> <summary>sudo btmon -w</summary>
```

HCI Event: LE Meta Event (0x3e) plen 26                                                                                             #6887 [hci0] 45.945885
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0015
            Props: 0x0015
              Connectable
              Directed
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: ADV_DIRECT_IND (0x0015)
          Address type: Random (0x01)
          Address: C4:EA:1C:CD:5B:C3 (Static)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -53 dBm (0xcb)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 74:13:EA:B6:52:F5 (Intel Corporate)
          Data length: 0x00
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                       #6888 [hci0] 45.967717
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                                           #6889 [hci0] 45.969820
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 13                                                                  #6890 [hci0] 45.970024
        Own address type: Public (0x00)
        Filter policy: Ignore not in accept list (0x01)
        PHYs: 0x05
        Entry 0: LE 1M
          Type: Passive (0x00)
          Interval: 60.000 msec (0x0060)
          Window: 30.000 msec (0x0030)
        Entry 1: LE Coded
          Type: Passive (0x00)
          Interval: 180.000 msec (0x0120)
          Window: 90.000 msec (0x0090)
> HCI Event: Command Complete (0x0e) plen 4                                                                                           #6891 [hci0] 45.970789
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                       #6892 [hci0] 45.970890
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                                           #6893 [hci0] 45.971775
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                       #6894 [hci0] 45.971879
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                                           #6895 [hci0] 45.974730
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 26                                                                    #6896 [hci0] 45.974776
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Random (0x01)
[... 139 lines cut ...]
        Own address type: Public (0x00)
        Filter policy: Ignore not in accept list (0x01)
        PHYs: 0x05
        Entry 0: LE 1M
          Type: Passive (0x00)
          Interval: 60.000 msec (0x0060)
          Window: 30.000 msec (0x0030)
        Entry 1: LE Coded
          Type: Passive (0x00)
          Interval: 180.000 msec (0x0120)
          Window: 90.000 msec (0x0090)
> HCI Event: Command Complete (0x0e) plen 4                                                                                           #6916 [hci0] 48.216804
      LE Set Extended Scan Parameters (0x08|0x0041) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                       #6917 [hci0] 48.216927
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                                           #6918 [hci0] 48.217795
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)

```
</details>

grealish, 2024-07-24:
Other related issues:
https://github.com/bluez/bluez/issues/778 
https://github.com/bluez/bluez/issues/899 
https://github.com/bluez/bluez/issues/752

Vudentz, 2024-07-24:
@grealish User Confirmation Negative Reply

Vudentz, 2024-07-24:
I think the culprit is actually this:

Failed to load LTKs for hci0: Invalid Parameters (0x0d)

btmon should be able to decode what is the contents of the command that makes it fail, perhaps something is corrupting the LTK on reboot or something like that.

karcsesz, 2024-08-24:
I have also started to experience this same issue with the exact same log messages about a month ago.

BlueZ `5.77-1` on EndeavourOS w/ Plasma 6 and kernel `6.10.6-arch1-1`. Hardware is an AMD Framework 13 which I believe uses its Mediatek MT7922 for Bluetooth as well.

Mouse works fine until reboot, after which I have to remove it from the paired device list and re-add it, which once again fixes the issue until reboot. I use the builtin Bluetooth management of KDE Plasma to do so.

r2rien, 2024-08-25:
Hi, same issue here with same intel Wireless-AC 9260 and a MX anywhere 3S;
(each mouse request gets a reconnect but then immediately disconnect)
..with same exact logs, in detail:
first connect:
```
Aug 24 22:42:14 bluetoothd[1418]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
Aug 24 22:42:14 bluetoothd[1418]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Aug 24 22:42:14 bluetoothd[1418]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Aug 24 22:42:14 bluetoothd[1418]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Aug 24 22:42:14 bluetoothd[1418]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Aug 24 22:42:14 bluetoothd[1418]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Aug 24 22:42:14 bluetoothd[1418]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Aug 24 22:42:14 bluetoothd[1418]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
Aug 24 22:42:14 bluetoothd[1418]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
```
then each further one, no more `report_read_cb()`:
```
Aug 24 22:42:17 bluetoothd[1418]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
Aug 24 22:42:17 bluetoothd[1418]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Aug 24 22:42:17 bluetoothd[1418]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Aug 24 22:42:17 bluetoothd[1418]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Aug 24 22:42:17 bluetoothd[1418]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error
Aug 24 22:42:17 bluetoothd[1418]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
```

lshw when functional (extract):
```
description: Laptop
product: XPS 13 9343 (0665)
vendor: Dell Inc.
  *-core
     *-pci
          description: Host bridge
          product: Broadwell-U Host Bridge -OPI
          vendor: Intel Corporation
          physical id: 100
          bus info: pci@0000:00:00.0
          version: 09
          width: 32 bits
          clock: 33MHz
          configuration: driver=bdw_uncore
          resources: irq:0
        *-usb:0
             description: USB controller
             product: Wildcat Point-LP USB xHCI Controller
             vendor: Intel Corporation
             physical id: 14
             bus info: pci@0000:00:14.0
             version: 03
             width: 64 bits
             clock: 33MHz
             capabilities: pm msi xhci bus_master cap_list
             configuration: driver=xhci_hcd latency=0
             resources: irq:47 memory:f7200000-f720ffff
           *-usbhost:0
                product: xHCI Host Controller
                vendor: Linux 6.10.6-amd64 xhci-hcd
                physical id: 0
                bus info: usb@1
                logical name: usb1
                version: 6.10
                capabilities: usb-2.00
                configuration: driver=hub slots=11 speed=480Mbit/s
              *-usb:0
                   description: Bluetooth wireless interface
                   product: Wireless-AC 9260 Bluetooth Adapter
                   vendor: Intel Corp.
                   physical id: 3
                   bus info: usb@1:3
                   version: 0.02
                   capabilities: bluetooth usb-2.00
                   configuration: driver=btusb maxpower=100mA speed=12Mbit/s
  *-input:8
       product: MX Anywhere 3S Mouse
       physical id: a
       logical name: input25
       logical name: /dev/input/event20
       logical name: /dev/input/mouse4
       capabilities: bluetooth
```
..ready for any supplementary testing/debug/bisecting/whatever needed, cheers

r2rien, 2024-08-25:
btmon log when success 
[btmon_MX3S_connect.log](https://github.com/user-attachments/files/16738467/btmon_MX3S_connect.log)

r2rien, 2024-08-25:
btmon reconnect failure:
```
Bluetooth monitor ver 5.77
btmon[18819]: = Note: Linux version 6.10.6-amd64 (x86_64)                                                                                                           0.301084
btmon[18819]: = Note: Bluetooth subsystem version 2.22                                                                                                              0.301087
= New Index: 64:6E:E0:E7:81:4F (Primary,USB,hci0)                                                                                                            [hci0] 0.301088
= Open Index: 64:6E:E0:E7:81:4F                                                                                                                              [hci0] 0.301089
= Index Info: 64:6E:E0:E7:81:4F (Intel Corp.)                                                                                                                [hci0] 0.301090
bluetoothd[1095]: @ MGMT Open: bluetoothd (privileged) version 1.22                                                                                        {0x0001} 0.301091
> HCI Event: LE Meta Event (0x3e) plen 26                                                                                                                 #1 [hci0] 5.005681
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0015
            Props: 0x0015
              Connectable
              Directed
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: ADV_DIRECT_IND (0x0015)
          Address type: Random (0x01)
          Address: D7:2A:DE:1E:73:9F (Static)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -38 dBm (0xda)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 64:6E:E0:E7:81:4F (Intel Corporate)
          Data length: 0x00
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                                           #2 [hci0] 5.005788
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                               #3 [hci0] 5.007633
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 26                                                                                        #4 [hci0] 5.007707
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Random (0x01)
        Peer address: D7:2A:DE:1E:73:9F (Static)
        Initiating PHYs: 0x01
        Entry 0: LE 1M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 7.50 msec (0x0006)
          Max connection interval: 11.25 msec (0x0009)
          Connection latency: 44 (0x002c)
          Supervision timeout: 2160 msec (0x00d8)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                 #5 [hci0] 5.009596
      LE Extended Create Connection (0x08|0x0043) ncmd 2
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 31                                                                                                                 #6 [hci0] 5.012610
      LE Enhanced Connection Complete (0x0a)
        Status: Success (0x00)
        Handle: 3585
        Role: Central (0x00)
        Peer address type: Random (0x01)
        Peer address: D7:2A:DE:1E:73:9F (Static)
        Local resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Peer resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Connection interval: 7.50 msec (0x0006)
        Connection latency: 44 (0x002c)
        Supervision timeout: 2160 msec (0x00d8)
        Central clock accuracy: 0x00
[... 113 lines cut ...]
bluetoothd[1095]: = profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error     5.314912
bluetoothd[1095]: = profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error     5.314985
bluetoothd[1095]: = profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error                5.315050
< HCI Command: LE Set Extended Scan Parameters (0x08|0x0041) plen 8                                                                                      #26 [hci0] 5.368532
        Own address type: Public (0x00)
        Filter policy: Ignore not in accept list (0x01)
        PHYs: 0x01
        Entry 0: LE 1M
          Type: Passive (0x00)
          Interval: 60.000 msec (0x0060)
          Window: 30.000 msec (0x0030)
[... cut ...]
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                                          #28 [hci0] 5.370035
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                              #29 [hci0] 5.370724
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
```

cheers

Vudentz, 2024-08-26:
> btmon reconnect failure:
>
> ```
> Bluetooth monitor ver 5.77
> btmon[18819]: = Note: Linux version 6.10.6-amd64 [...]

Weird, looks like the keys are not being reprogrammed or something, does this happens when you restart bluetooth service or only when you reboot? An what is your kernel version?

r2rien, 2024-08-26:
Hi,
kernel 6.10.6-amd64 from debian
each reboot/wake from suspend/hibernate

..let me try now after restarting bluetooth service..
..yes reconnection after a systemctl restart bluetooth.service successfull
(tell me if you need its btmon logs)
so only when reboot/resume

at your disposal for any other info/testing - thanks
cheers

r2rien, 2024-08-26:
Ok so you need btmon logs, I presume - preparing them..
thanks

Vudentz, 2024-08-26:
> Ok so you need btmon logs, I presume - preparing them.. thanks

The tricky part is how to get the logs from the boot, maybe you will need a custom service that start btmon at boot just before bluetoothd.

r2rien, 2024-08-26:
here are reconnect logs after systemctl restart:
[btmon_MX3S_restart-bluetooth.log](https://github.com/user-attachments/files/16753543/btmon_MX3S_restart-bluetooth.log)
(I just clicked on left button mouse to initiate reconnection to not clutter logs)

r2rien, 2024-08-26:
..ah ok you need reboot logs,
will try to handle that tweaking bluetooth.target..

r2rien, 2024-08-26:
meanwhile, Cf 
> is just the LTK since the device can still reconnect
 can you inform me where is it saved ?

Vudentz, 2024-08-26:
> ```
> bluetoothd[46328]: @ MGMT Command: Load.. (0x0013) plen 74  {0x0001} [hci0] 1.343037
>         Keys: 2
>         BR/EDR Address: C0:DC:DA:A5:E5:47 (Samsung Electronics Co.,Ltd)
>         Key type: Authenticated key from P-256 (0x03)
>         Central: 0x00
>         Encryption size: 16
>         Diversifier[2]: 0000
>         Randomizer[8]: 0000000000000000
>         Key[16]: 6ed96089bd9765be2f2c971b0b95f624
>         LE Address: D7:2A:DE:1E:73:A2 (Static)
>         Key type: Unauthenticated key from P-256 (0x02)
>         Central: 0x00
>         Encryption size: 16
>         Diversifier[2]: 0000
>         Randomizer[8]: 0000000000000000
>         Key[16]: 87dd2546ededda380ffcdc0a8faa4597
> @ MGMT Event: Command Status (0x0002) plen 3                {0x0001} [hci0] 1.343039
>       Load Long Term Keys (0x0013)
>         Status: Invalid Parameters (0x0d)
> ```
> 
> So, it is failing to load the keys, which explain why it doesn't work on reboot since if you just restart perhaps the kernel already has the old keys, still it shall not fail like this.

BR/EDR Address: C0:DC:DA:A5:E5:47 (Samsung Electronics Co.,Ltd)

```
static bool ltk_is_valid(struct mgmt_ltk_info *key)
{
	if (key->initiator != 0x00 && key->initiator != 0x01)
		return false;

	switch (key->addr.type) {
	case BDADDR_LE_PUBLIC:
		return true;

	case BDADDR_LE_RANDOM:
		/* Two most significant bits shall be set */
		if ((key->addr.bdaddr.b[5] & 0xc0) != 0xc0)
			return false;
		return true;
	}

	return false;
}
```

It seems LTKs cannot be of type BR/EDR according to the comment above but later on we have:

```
		/* When using SMP over BR/EDR, the addr type should be set to BREDR */
		if (key->addr.type == BDADDR_BREDR)
			addr_type = BDADDR_BREDR;
```

r2rien, 2024-08-26:
why `BR/EDR Address: C0:DC:DA:A5:E5:47 (Samsung Electronics Co.,Ltd)` ?
```
↳ bluetoothctl devices
Device D7:2A:DE:1E:73:A2 MX Anywhere 3S
```
..let me try the reboot, to check agains its id - brb

r2rien, 2024-08-26:
```
Bluetooth monitor ver 5.77
btmon[2489]: = Note: Linux version 6.10.6-amd64 (x86_64)                                                                                                            0.426588
btmon[2489]: = Note: Bluetooth subsystem version 2.22                                                                                                               0.426591
= New Index: 64:6E:E0:E7:81:4F (Primary,USB,hci0)                                                                                                            [hci0] 0.426592
= Open Index: 64:6E:E0:E7:81:4F                                                                                                                              [hci0] 0.426592
= Index Info: 64:6E:E0:E7:81:4F (Intel Corp.)                                                                                                                [hci0] 0.426593
bluetoothd[1105]: @ MGMT Open: bluetoothd (privileged) version 1.22                                                                                        {0x0001} 0.426594
> HCI Event: LE Meta Event (0x3e) plen 26                                                                                                                 #1 [hci0] 2.712437
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0015
            Props: 0x0015
              Connectable
              Directed
              Use legacy advertising PDUs
            Data status: Complete
          Legacy PDU Type: ADV_DIRECT_IND (0x0015)
          Address type: Random (0x01)
          Address: D7:2A:DE:1E:73:A2 (Static)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -29 dBm (0xe3)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 64:6E:E0:E7:81:4F (Intel Corporate)
          Data length: 0x00
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                                           #2 [hci0] 2.712590
        Extended scan: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                               #3 [hci0] 2.713288
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 26                                                                                        #4 [hci0] 2.713345
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Random (0x01)
        Peer address: D7:2A:DE:1E:73:A2 (Static)
        Initiating PHYs: 0x01
        Entry 0: LE 1M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 7.50 msec (0x0006)
          Max connection interval: 11.25 msec (0x0009)
          Connection latency: 44 (0x002c)
          Supervision timeout: 2160 msec (0x00d8)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                 #5 [hci0] 2.715316
      LE Extended Create Connection (0x08|0x0043) ncmd 2
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 31                                                                                                                 #6 [hci0] 2.718432
      LE Enhanced Connection Complete (0x0a)
        Status: Success (0x00)
        Handle: 3585
        Role: Central (0x00)
        Peer address type: Random (0x01)
        Peer address: D7:2A:DE:1E:73:A2 (Static)
        Local resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Peer resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Connection interval: 7.50 msec (0x0006)
        Connection latency: 44 (0x002c)
        Supervision timeout: 2160 msec (0x00d8)
        Central clock accuracy: 0x00
@ MGMT Event: Device Connected (0x000b) plen 13                                                                                                     {0x0001} [hci0] 2.718499
[... 110 lines cut ...]
bluetoothd[1105]: = profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error     3.021072
bluetoothd[1105]: = profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error                       3.021132
bluetoothd[1105]: = profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error     3.021193
bluetoothd[1105]: = profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error     3.021252
bluetoothd[1105]: = profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error                3.021312
[... cut ...]
< HCI Command: LE Set Extended Scan Enable (0x08|0x0042) plen 6                                                                                          #28 [hci0] 3.060499
        Extended scan: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
        Duration: 0 msec (0x0000)
        Period: 0.00 sec (0x0000)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                              #29 [hci0] 3.062268
      LE Set Extended Scan Enable (0x08|0x0042) ncmd 2
        Status: Success (0x00)
```

Vudentz, 2024-08-26:
@r2rien it is failing because the samsumg device you have is store as LTK with address type as BR/EDR, maybe something has changed and we used to store it like that but this doesn't work when reloading.

Vudentz, 2024-08-26:
```
commit 59b047bc98084f8af2c41483e4d68a5adf2fa7f7
Author: Xiao Yao <xiaoyao@rock-chips.com>
Date:   Tue Dec 12 00:27:18 2023 +0800

    Bluetooth: MGMT/SMP: Fix address type when using SMP over BREDR/LE
    
    If two Bluetooth devices both support BR/EDR and BLE, and also
    support Secure Connections, then they only need to pair once.
    The LTK generated during the LE pairing process may be converted
    into a BR/EDR link key for BR/EDR transport, and conversely, a
    link key generated during the BR/EDR SSP pairing process can be
    converted into an LTK for LE transport. Hence, the link type of
    the link key and LTK is not fixed, they can be either an LE LINK
    or an ACL LINK.
    
    Currently, in the mgmt_new_irk/ltk/crsk/link_key functions, the
    link type is fixed, which could lead to incorrect address types
    being reported to the application layer. Therefore, it is necessary
    to add link_type/addr_type to the smp_irk/ltk/crsk and link_key,
    to ensure the generation of the correct address type.
    
    SMP over BREDR:
    Before Fix:
    > ACL Data RX: Handle 11 flags 0x02 dlen 12
            BR/EDR SMP: Identity Address Information (0x09) len 7
            Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
    @ MGMT Event: New Identity Resolving Key (0x0018) plen 30
            Random address: 00:00:00:00:00:00 (Non-Resolvable)
            LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
    @ MGMT Event: New Long Term Key (0x000a) plen 37
            LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
            Key type: Authenticated key from P-256 (0x03)
    
    After Fix:
    > ACL Data RX: Handle 11 flags 0x02 dlen 12
          BR/EDR SMP: Identity Address Information (0x09) len 7
            Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
    @ MGMT Event: New Identity Resolving Key (0x0018) plen 30
            Random address: 00:00:00:00:00:00 (Non-Resolvable)
            BR/EDR Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
    @ MGMT Event: New Long Term Key (0x000a) plen 37
            BR/EDR Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
            Key type: Authenticated key from P-256 (0x03)
    
    SMP over LE:
    Before Fix:
    @ MGMT Event: New Identity Resolving Key (0x0018) plen 30
            Random address: 5F:5C:07:37:47:D5 (Resolvable)
            LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
    @ MGMT Event: New Long Term Key (0x000a) plen 37
            LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
            Key type: Authenticated key from P-256 (0x03)
    @ MGMT Event: New Link Key (0x0009) plen 26
            BR/EDR Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
            Key type: Authenticated Combination key from P-256 (0x08)
    
    After Fix:
    @ MGMT Event: New Identity Resolving Key (0x0018) plen 30
            Random address: 5E:03:1C:00:38:21 (Resolvable)
            LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
    @ MGMT Event: New Long Term Key (0x000a) plen 37
            LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
            Key type: Authenticated key from P-256 (0x03)
    @ MGMT Event: New Link Key (0x0009) plen 26
            Store hint: Yes (0x01)
            LE Address: F8:7D:76:F2:12:F3 (OUI F8-7D-76)
            Key type: Authenticated Combination key from P-256 (0x08)
    
    Cc: stable@vger.kernel.org
    Signed-off-by: Xiao Yao <xiaoyao@rock-chips.com>
    Signed-off-by: Luiz Augusto von Dentz <luiz.von.dentz@intel.com>

```

Looks like the above introduced the capability of generating LTK with address type of BR/EDR, but I don't think the author tested reloading because ltk_is_valid clearly doesn't support it.

r2rien, 2024-08-26:
Thanks, 
Do you still need btmon logs before starting bluetooth ?
Where could I purge anything stored, deeper than gui 'remove device' ?

r2rien, 2024-08-26:
Thanks indeed, deleting this old samsung device (which I don't use anymore) solves the issue.

karcsesz, 2024-08-26:
> ```
> @ MGMT Command: Load.. (0x0013) plen 74  {0x0001} [hci0] 1.343037
>         Keys: 2
>         BR/EDR Address: C0:DC:DA:A5:E5:47 (Samsung Electronics Co.,Ltd)
>         Key type: Authenticated key from [...]
       
Can confirm, running `systemctl restart bluetooth.service` while having btmon attached listed something similar for me too, with the BR/EDR address being attached to my Android phone. Removing that from the paired devices also seems to have fixed the issue.

Thank you!

Vudentz, 2024-08-27:
@r2rien @karcsesz can you try with the changes above? It needs to be applied to the kernel though, so you maybe required to reboot after rebuilding and installing the kernel image.

karcsesz, 2024-08-27:
On it, will check.

Should re-adding my phone to the device list break it again without the patches?

karcsesz, 2024-08-27:
Re-added phone, mouse still works without an issue even on stock kernel. How would you recommend I verify the fix?

Vudentz, 2024-08-28:
@karcsesz lets try with the userspace fix up changes first, Ive scraped the idea of doing the fix up in the kernel since it might take a lot more time to propagate to the distros.

closed 2024-08-28 by BluezTestBot

maks, 2024-09-22:
I seem to be running into similar issue of having moved to Ubuntu 24.04 (fresh install) which has a kernel 6.8.0-45 and now the  Logitech MX master 3S mouse will no reconnect after a sleep or reboot. 

**BUT** I'm not sure if this is the same issue as here because: a. I don't have any other Bluetooth device paired and b. because I can see the BT MAC address of the mouse increments by one each time that I need to do the forget-reconnect dance. @Vudentz would you have any thoughts as to why/how? the mouses address can be incrementing each time? 
I found [another person reporting](https://unix.stackexchange.com/questions/690808/mouse-logitech-mx-anywhere-3-keeps-changing-its-bluetooth-mac-address-on-every) the same curious address incrementing happening and like one of the commentors on that SO post I have a intel AX200. 

Happy to provide more information if needed, not this mouse was working perfectly on this very same laptop when I previously had Ubuntu 23.10 installed (sorry don't remember which kernel version it was).

samodadela, 2024-11-12:
Don't know if it's the same issue - the description matches, though. 
Steps to reproduce:
- pair the MX Master 3 (via BT), mouse works
- reboot the computer, the mouse keeps connecting/disconnecting each 5 seconds or so (cursor does not move)
- to fix: remove the pairing (using bluetooth manager), re-pair the mouse -> works until next reboot

I also tried to reinstall bluez ([like](https://github.com/bluez/bluez/issues/875#issuecomment-2167336474))... This of course removed all my devices and indeed it seemed to have worked (also across reboots)... until I added another device (a phone). Then after reboot the same problem again (mouse doesn't work, connect, disconnect...). Removing the phone pairing makes the mouse work again.

Using bluez `5.72-0ubuntu5`, kernel:
```
Linux a 6.8.0-48-generic #48-Ubuntu SMP PREEMPT_DYNAMIC Fri Sep 27 14:04:52 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

maks, 2024-11-13:
The workaround that I've been using that *mostly* works for me is editing the info file: `/var/lib/bluetooth/<MAC-ADAPTER>/<MAC-MOUSE>/info` (note need to have su perms to access that path on Ubuntu)

and removing the 2 lines
```
[IdentityResolvingKey]
Key=CDF...
```
and then restarting bluetoothd: `sudo systemctl restart bluetooth` 

This mostly but not always survives both suspect and laptop restarts.

I think I found the above via a thread on [Arch forum](https://bbs.archlinux.org/viewtopic.php?id=259954&p=2) and I've also filed a bug about this in the [Ubuntu bugtracker](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/2081663).