bluez/bluez #876: Bluetooth keyboard issues when switching from 5.75-1 to 5.76-1 (Arch Linux)

jagaimoworks, 2024-06-12:
When I update to  version 5.76-1 everything works fine until I reboot the system. The keyboard still connects without issues but the input is very laggy as if the connection was not stable or polling has broken somehow. There is very noticeable delay and if the input is too fast it will partially drop key events.

When I try to re-pair the device it gets even worse. as the keyboard will loop through connected an then instantly disconnected state, even resetting the keyboard itself. 

But it gets even weirder. If I then reboot again, I can re-pair the keyboard and it will work fine without issues until it starts lagging again on a subsequent reboot.

The lagging will persist even after downgrading back to 5.75-1, until I re-pair it with that version.

Please let me know how I can help you help me.

Vudentz, 2024-06-12:
@jagaimoworks we probably need to start with HCI traces/btmon to check what is going on.

Vudentz, 2024-06-12:
@jagaimoworks btw, check if is not something like the following going on:

https://github.com/bluez/bluez/issues/875#issuecomment-2163613344

jagaimoworks, 2024-06-12:
This is what you meant. right? The dump contains the intial connection until `#129 [hci0] 14.471786`. After that it is just a few keypresses and thats it. This is when the keyboard is laggy like so: 

https://github.com/bluez/bluez/assets/54439198/510dd568-8791-4f33-bdf9-d10f1f9a8205

```
sudo btmon -p7 -w hcidump.log
```
```
Bluetooth monitor ver 5.76
= Note: Linux version 6.9.3-zen1-1-zen (x86_64)                        0.342735
= Note: Bluetooth subsystem version 2.22                               0.342736
= New Index: 00:1A:7D:DA:71:15 (Primary,USB,hci0)               [hci0] 0.342737
= Open Index: 00:1A:7D:DA:71:15                                 [hci0] 0.342737
= Index Info: 00:1A:7D:DA:71:15 (Cambridge Silicon Radio)       [hci0] 0.342737
@ MGMT Open: bluetoothd (privileged) version 1.22             {0x0001} 0.342738
> HCI Event: LE Meta Event (0x3e) plen 31                    #1 [hci0] 2.849607
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Random (0x01)
        Address: C9:80:3C:7F:7E:48 (Static)
        Data length: 19
        Appearance: Keyboard (0x03c1)
        Flags: 0x04
          BR/EDR Not Supported
        16-bit Service UUIDs (complete): 1 entry
          Human Interface Device (0x1812)
        Name (complete): AM AFA
        RSSI: -44 dBm (0xd4)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2       #2 [hci0] 2.849638
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                  #3 [hci0] 2.851446
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Create Connection (0x08|0x000d) plen 25    #4 [hci0] 2.851466
        Scan interval: 60.000 msec (0x0060)
        Scan window: 60.000 msec (0x0060)
        Filter policy: Accept list is not used (0x00)
        Peer address type: Random (0x01)
        Peer address: C9:80:3C:7F:7E:48 (Static)
        Own address type: Public (0x00)
        Min connection interval: 7.50 msec (0x0006)
        Max connection interval: 200.00 msec (0x00a0)
        Connection latency: 6 (0x0006)
        Supervision timeout: 3000 msec (0x012c)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                    #5 [hci0] 2.857447
      LE Create Connection (0x08|0x000d) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 19                    #6 [hci0] 2.922613
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 75
        Role: Central (0x00)
        Peer address type: Random (0x01)
        Peer address: C9:80:3C:7F:7E:48 (Static)
        Connection interval: 200.00 msec (0x00a0)
        Connection latency: 6 (0x0006)
        Supervision timeout: 3000 msec (0x012c)
        Central clock accuracy: 0x00
@ MGMT Event: Device Connected (0x000b) plen 32        {0x0001} [hci0] 2.922627
        LE Address: C9:80:3C:7F:7E:48 (Static)
        Flags: 0x00000008
          Connection Locally Initiated
        Data length: 19
        Appearance: Keyboard (0x03c1)
        Flags: 0x04
          BR/EDR Not Supported
[... 1087 lines cut ...]
> ACL Data RX: Handle 72 flags 0x02 dlen 23               #192 [hci0] 37.071786
      ATT: Handle Value Notification (0x1b) len 18
        Handle: 0x0018
          Data[16]: 00000014020000000000000000000000
> ACL Data RX: Handle 72 flags 0x02 dlen 23               #193 [hci0] 37.071786
      ATT: Handle Value Notification (0x1b) len 18
        Handle: 0x0018
          Data[16]: 00000004020000000000000000000000
> ACL Data RX: Handle 72 flags 0x02 dlen 23               #194 [hci0] 37.071786
      ATT: Handle Value Notification (0x1b) len 18
        Handle: 0x0018
          Data[16]: 00000004000000000000000000000000
> ACL Data RX: Handle 72 flags 0x02 dlen 23               #195 [hci0] 37.073783
      ATT: Handle Value Notification (0x1b) len 18
        Handle: 0x0018
          Data[16]: 00000000000000000000000000000000
> ACL Data RX: Handle 72 flags 0x02 dlen 23               #196 [hci0] 38.271786
      ATT: Handle Value Notification (0x1b) len 18
        Handle: 0x0018
          Data[16]: 00000000000000020000000000000000
> ACL Data RX: Handle 72 flags 0x02 dlen 23               #197 [hci0] 38.271787
      ATT: Handle Value Notification (0x1b) len 18
        Handle: 0x0018
          Data[16]: 00000000000000000000000000000000
```

Vudentz, 2024-06-12:
```
< HCI Command: LE Create Connection (0x08|0x000d) plen 25   #93 [hci0] 9.299572
        Scan interval: 60.000 msec (0x0060)
        Scan window: 60.000 msec (0x0060)
        Filter policy: Accept list is not used (0x00)
        Peer address type: Random (0x01)
        Peer address: C9:80:3C:7F:7E:48 (Static)
        Own address type: Public (0x00)
        Min connection interval: 7.50 msec (0x0006)
        Max connection interval: 200.00 msec (0x00a0)
        Connection latency: 6 (0x0006)
        Supervision timeout: 3000 msec (0x012c)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                   #94 [hci0] 9.304382
      LE Create Connection (0x08|0x000d) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 19                   #95 [hci0] 9.315552
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 72
        Role: Central (0x00)
        Peer address type: Random (0x01)
        Peer address: C9:80:3C:7F:7E:48 (Static)
        Connection interval: 200.00 msec (0x00a0)
        Connection latency: 6 (0x0006)
        Supervision timeout: 3000 msec (0x012c)
        Central clock accuracy: 0x00
```

Yeah, the controller is selecting 200ms it seems, from a range of 7.5-200ms which is probably what we discover via PPCP, perhaps it tries to prioritize power over latency, so without us knowing exactly what policy applies based on the connection min-max this is a hit and miss logic.

Btw, try checking if the following helps:

https://patchwork.kernel.org/project/bluetooth/list/?series=861326

jagaimoworks, 2024-06-12:
> Yeah, the controller is selecting 200ms it seems, from a range of 7.5-200ms which is probably what we discover via PPCP, perhaps it tries to prioritize power over latency, so without us knowing exactly what policy applies based on the connection min-max this is a hit and miss logic.

Interestingly, when looking at the HCI dump after re-pairing the keyboard the min and max connection intervals are different from before:
```
< HCI Command: LE Create Connection (0x08|0x000d) plen 25    #4 [hci0] 3.307167
        Scan interval: 60.000 msec (0x0060)
        Scan window: 60.000 msec (0x0060)
        Filter policy: Accept list is not used (0x00)
        Peer address type: Random (0x01)
        Peer address: C9:80:3C:7F:7E:49 (Static)
        Own address type: Public (0x00)
        Min connection interval: 30.00 msec (0x0018)
        Max connection interval: 50.00 msec (0x0028)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                    #5 [hci0] 3.311971
      LE Create Connection (0x08|0x000d) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 19                    #6 [hci0] 3.370138
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 73
        Role: Central (0x00)
        Peer address type: Random (0x01)
        Peer address: C9:80:3C:7F:7E:49 (Static)
        Connection interval: 50.00 msec (0x0028)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Central clock accuracy: 0x00
```
At least in this case the controller seems to just go with the max interval which 50ms which feels fine to me. As to why it is 200ms after a reboot I do not quite understand.

> Btw, try checking if the following helps:
> 
> https://patchwork.kernel.org/project/bluetooth/list/?series=861326

This looks very promising. Especially the "Fix default PPCP connection intervals" patch seems to address exactly this. I'll try it later tomorrow and report back.

jagaimoworks, 2024-06-13:
Well damn, it was the first patch (https://patchwork.kernel.org/project/bluetooth/patch/20240612191145.1379415-1-luiz.dentz@gmail.com/) that fixed the problem instead of the one I thought to be promising.

So I guess the issue will be solved in the next release?

Vudentz, 2024-06-13:
> Well damn, it was the first patch ([patchwork.kernel.org/project/bluetooth/patch/20240612191145.1379415-1-luiz.dentz@gmail.com](https://patchwork.kernel.org/project/bluetooth/patch/20240612191145.1379415-1-luiz.dentz@gmail.com/)) that fixed the problem instead of the one I thought to be promising.
> 
> So I guess the issue will be solved in the next release?

Most likely yes, that said with newer kernel we would still use the PPCP values so I wonder what we shall do then, Im thinking in perhaps introduce a policy that adjust the connection interval based on traffic, but that most likely need some kernel changes as well.

closed 2024-06-13 by BluezTestBot