bluez/bluez #877: Bluetooth: hci0: Malformed MSFT vendor event: 0x02

metemaddar, 2024-06-14:
**Summary**: Malformed MSFT vendor event in Bluetooth during startup

**Description**:
The following Bluetooth error message appears during startup:

19:48:53 kernel: Bluetooth: hci0: Malformed MSFT vendor event: 0x02

**System Information**:
- Distribution: Ubuntu 22.04
- BlueZ Version: 5.64-0ubuntu1.3
- Bluetooth Device: Bluetooth headset [Samsung Level U2](https://www.samsung.com/levant/mobile-accessories/samsung-level-u2-eo-b3300blegww/)

**Additional Information**:
This message does not affect Bluetooth functionality but should be reviewed.

joakim-tjernlund, 2024-06-14:
Intel BT AX... controllers has printed this error for a very long time.  Either fix the underlying problem or
stop printing this msg for Intel AX... controllers if cannot be fixed.

Tractor777, 2024-06-19:
not just the Intel AX... series the 8265 card has also been givin this error message for as long as I can remember. Bluetooth does work. Ubuntu 24.04 Bluez version 5.72-0Ubuntu5

pomergranate, 2024-08-02:
I have met this problem too, when boot ubuntu, which makes me unable to start gnome. 
Blow is the detailed problem. Hope someone can solve this problem. 
[image]

chaffra, 2024-08-16:
Me too. Same issue after enabling Bluetooth on my ubuntu

Emmanuel-Coutlee, 2024-08-28:
Got the same issue. Also blocking gnome from starting.

BlazarKnight, 2024-11-01:
I can get to gnome, but i can't use Bluetooth.

joakim-tjernlund, 2024-11-03:
The ```hci0: Malformed MSFT vendor event: 0x02``` error has been fixed upstream:

https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/commit/?id=4d67eb2b5444e1c872db4f04e7921ae8a95251fe

And I think it just an harmless printout

closed 2025-07-23 by github-actions[bot]