bluez/bluez #880: Bluez 5.75+ breaks steam

youpie, 2024-06-19:
After upgrading to blues 5.75 and connecting a steam controller via bluetooth. Steam does not freeze however it does not respond to the controller input and pressing buttons in the Steam UI with the mouse does nothing until you disconnect the controller again. Downgrading to 5.73 fixes this issue and makes steam work normally again.

I use fedora 40

Vudentz, 2024-06-20:
@youpie there was some hiccups with 5.75 wrt to HID devices, so I do recommend testing with 5.76 instead.

benaryorg, 2024-06-21:
I can confirm that Steam and bluez 5.75 are having issues.
It seems a bit random which case I hit, sometimes Steam just freezes, other times bluetoothd just straight out segfaults, however it does not seem to fault consistently (or on the other hand the stacktrace captured by systemd-coredump might be off).

```text
Module libcap.so.2 without build-id.
Module libsystemd.so.0 without build-id.
Module libpcre2-8.so.0 without build-id.
Module libudev.so.1 without build-id.
Module libasound.so.2 without build-id.
Module bluetoothd without build-id.
Stack trace of thread 1777:
#0  0x00000000004bd863 queue_remove_all (bluetoothd + 0xbd863)
#1  0x00000000004bd8a5 queue_destroy (bluetoothd + 0xbd8a5)
#2  0x00000000004c1763 uhid_free (bluetoothd + 0xc1763)
#3  0x000000000043c122 hog_free (bluetoothd + 0x3c122)
#4  0x000000000043a3ea hog_remove (bluetoothd + 0x3a3ea)
#5  0x000000000048e179 service_remove (bluetoothd + 0x8e179)
#6  0x000000000049dc2a device_remove (bluetoothd + 0x9dc2a)
#7  0x0000000000480a0b adapter_remove (bluetoothd + 0x80a0b)
#8  0x000000000048525b adapter_cleanup (bluetoothd + 0x8525b)
#9  0x000000000040a6e2 main (bluetoothd + 0xa6e2)
#10 0x00007fcb2dbae10e __libc_start_call_main (libc.so.6 + 0x2a10e)
#11 0x00007fcb2dbae1c9 __libc_start_main@@GLIBC_2.34 (libc.so.6 + 0x2a1c9)
#12 0x000000000040aea5 _start (bluetoothd + 0xaea5)
ELF object binary architecture: AMD x86-64
```

```text
Module libcap.so.2 without build-id.
Module libsystemd.so.0 without build-id.
Module libpcre2-8.so.0 without build-id.
Module libudev.so.1 without build-id.
Module libasound.so.2 without build-id.
Module bluetoothd without build-id.
Stack trace of thread 19934:
#0  0x00000000004bd52e queue_pop_head (bluetoothd + 0xbd52e)
#1  0x00000000004c219b bt_uhid_record (bluetoothd + 0xc219b)
#2  0x00000000004c22d1 bt_uhid_set_report_reply (bluetoothd + 0xc22d1)
#3  0x000000000043ac48 set_report_cb (bluetoothd + 0x3ac48)
#4  0x000000000045ab40 attrib_callback_result (bluetoothd + 0x5ab40)
#5  0x00000000004c64a8 can_read_data (bluetoothd + 0xc64a8)
#6  0x00000000004e95d5 watch_callback (bluetoothd + 0xe95d5)
#7  0x00007f062705ae39 g_main_dispatch (libglib-2.0.so.0 + 0x5be39)
#8  0x00007f062705dfd7 g_main_context_iterate_unlocked.isra.0 (libglib-2.0.so.0 + 0x5efd7)
#9  0x00007f062705e88f g_main_loop_run (libglib-2.0.so.0 + 0x5f88f)
#10 0x00000000004e9c65 mainloop_run (bluetoothd + 0xe9c65)
#11 0x00000000004ea0bc mainloop_run_with_signal (bluetoothd + 0xea0bc)
#12 0x000000000040a6bd main (bluetoothd + 0xa6bd)
#13 0x00007f0626c8310e __libc_start_call_main (libc.so.6 + 0x2a10e)
#14 0x00007f0626c831c9 __libc_start_main@@GLIBC_2.34 (libc.so.6 + 0x2a1c9)
#15 0x000000000040aea5 _start (bluetoothd + 0xaea5)
ELF object binary architecture: AMD x86-64
```

Updating to 5.76 *did not help*.
I'm not sure which version the stacktraces above are from, sorry.
If there is need for further testing of specific versions or debug infos, I can try.

habeebweeb, 2024-06-24:
People over in ValveSoftware/steam-for-linux#10203 are reporting having the same issue as well.

I've bisected the issue back to commit 256d0b594d044222975f55e3b2d02990e0f88f87 with the migration over to the new uhid opcode functions. This commit is a handful of commits after the `5.73` release so that's the last version that works with the steam controller.

I tried my hand at fixing the issue myself and the only thing I noticed was that in `profiles/input/hog-lib.c` in the `report_reply` function when calling `bt_uhid_get_report_reply`, the `status` and `id` arguments seem to be passed in the "wrong" order where `status` is the argument for the `number` parameter and `id` is the argument for the `status` parameter.

Swapping the two arguments did seem to work like it did in `5.73` but I think I had further issues with the controller not responding after a while so I'm not sure if what I found was a red herring.

Vudentz, 2024-06-24:
> People over in [ValveSoftware/steam-for-linux#10203](https://github.com/ValveSoftware/steam-for-linux/issues/10203) are reporting having the same issue as well.
> 
> I've bisected the issue back to commit [256d0b5](https://github.com/bluez/bluez/commit/256d0b594d044222975f55e3b2d02990e0f88f87) with the migration over to the new uhid opcode functions. This commit is a handful of commits after the `5.73` release so that's the last version that works with the steam controller.
> 
> I tried my hand at fixing the issue myself and the only thing I noticed was that in `profiles/input/hog-lib.c` in the `report_reply` function when calling `bt_uhid_get_report_reply`, the `status` and `id` arguments seem to be passed in the "wrong" order where `status` is the argument for the `number` parameter and `id` is the argument for the `status` parameter.
> 
> Swapping the two arguments did seem to work like it did in `5.73` but I think I had further issues with the controller not responding after a while so I'm not sure if what I found was a red herring.

Do you have a diff or a patch? Btw, it would probably be a good idea to check if that wasn't resolved in 5.76 already.

habeebweeb, 2024-06-24:
I've tried both 5.76 and main and the issue still persists.

Here's a patch of the fix I attempted on the main branch. I `git blame`d the line and found that there haven't been any changes since the commit I mentioned earlier.
```patch
diff --git a/profiles/input/hog-lib.c b/profiles/input/hog-lib.c
index 67a0e832d..1b4aca074 100644
--- a/profiles/input/hog-lib.c
+++ b/profiles/input/hog-lib.c
@@ -904,7 +904,7 @@ static void report_reply(struct bt_hog *hog, uint8_t status, uint8_t id,
 
 	hog->getrep_att = 0;
 
-	err = bt_uhid_get_report_reply(hog->uhid, hog->getrep_id, status, id,
+	err = bt_uhid_get_report_reply(hog->uhid, hog->getrep_id, id, status,
 					data, len);
 	if (err < 0)
 		error("bt_uhid_get_report_reply: %s", strerror(-err));

```

Vudentz, 2024-06-24:
@habeebweeb check with the changes above.

habeebweeb, 2024-06-24:
> @habeebweeb check with the changes above.

I built bluez from the BluezTestBot/bluez#2295 and ran `bluetoothd -n -d`. Steam does not freeze when opening while the steam controller is connected and opening the settings window in steam now works.

closed 2024-06-25 by BluezTestBot