bluez/bluez #882: LE Audio output+input with Samsung Galaxy Buds2 Pro broken in new firmware

pv, 2024-06-22:
Since recent firmware upgrade to Samsung Galaxy Buds2 Pro (to version R510XXU0AXC2 Jun 2024), simultaneous input + output doesn't work, as the device replies with Unspecified Error to ASE Enable command.

```
> ACL Data RX: Handle 2049 flags 0x02 dlen 15                                                                   #235 [hci1] 12.949319
      ATT: Handle Value Notification (0x1b) len 10
        Handle: 0x0043 Type: ASE Control Point (0x2bc6)
          Data[8]: 0302030000010e0b
            Opcode: Enable (0x03)
            Number of ASE(s): 2
            ASE: #0
            ASE ID: 0x03
            ASE Response Code: Success (0x00)
            ASE Response Reason: None (0x00)
            ASE: #1
            ASE ID: 0x01
            ASE Response Code: Unspecified Error (0x0e)
            ASE Response Reason: Reserved (0x0b)
```

The same Codec+QoS configuration used to work on earlier firmware version of this device, so not clear what it is unhappy about. (Quite likely this is device side issue, but attaching the logs below just in case something can be seen.)

The device audio output works if we don't configure the mic input stream. Unlike Sony LinkBuds S, reducing output stream sampling rate from 48 kHz to something lower doesn't allow enabling the mic input simultaneously, and so far I didn't find a working input+output configuration (tried different config / QoS).

btmon: [gbuds2-text.log.txt](https://github.com/user-attachments/files/15936077/gbuds2-text.log.txt) [gbuds2.log.txt](https://github.com/user-attachments/files/15936078/gbuds2.log.txt)

Jah-On, 2024-07-03:
Was this using latest source or latest release? I know that you mentioned this was also an issue for the Zen Hybrids Pros in another issue.

ValeZAA, 2025-06-11:
I was able to make it work on a commit right before https://github.com/bluez/bluez/commit/4a7ee30978da919cc820116be6ced323b35e4ebd

But only on one earbud for some reason and with terrible artefacts. Also, I have buds3 pro, not Buds2

ValeZAA, 2025-06-12:
>UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)

Where is UUID: Coordinated Set Identif.. (00001846-0000-1000-8000-00805f9b34fb)? Also info command should show Sets Key: /org/bluez/hciX/set_xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

Jah-On, 2025-06-12:
> > UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)
> 
> Where is UUID: Coordinated Set Identif.. (00001846-0000-1000-8000-00805f9b34fb)? Also info command should show Sets Key: /org/bluez/hciX/set_xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

These are headphones so the CSIS/P is not needed.

closed 2025-06-14 by pv

ValeZAA, 2025-06-26:
> I was able to make it work on a commit right before [4a7ee30](https://github.com/bluez/bluez/commit/4a7ee30978da919cc820116be6ced323b35e4ebd)
> 
> But only on one earbud for some reason and with terrible artefacts. Also, I have buds3 pro, not Buds2

https://github.com/bluez/bluez/commit/ae1b7f6ba805f82742bbc32ff275e268248ef9f8 this is the issue with only one earbud connecting in BAP mode i had. Will test whether it is fixed and update the comment.