bluez/bluez #884: What are the Trusted devices?

ratijas, 2024-06-24:
Looking at [this page](https://github.com/bluez/bluez/blob/52bda9d45572ab4629bf2f686616c0398f489dad/doc/org.bluez.Device.rst#L221), the only meaningful explanation for the Trusted property is in the semi-unrelated `void Disconnect()` method:

> For non-trusted devices connected over LE bearer calling this method will disable incoming connections until Connect method is called again.

In KDE Bluetooth KCM (system settings page) we have a checkbox for the Trusted property.

[image]

However, we find it hard to [present an explanation](https://invent.kde.org/plasma/bluedevil/-/merge_requests/173) hint for it in layman's terms.  "LE bearer" is too technical of a jargon.

What do developers of BlueZ thing about it?

weliem, 2024-06-25:
I have the same question. Can't figure out from the documentation what it means. I experimented a bit but it doesn't seem to do anything....at least, not that I could notice.

ratijas, 2024-06-28:
Apparently we check if the device is Trusted to branch in at least one place in KDE:

```
~/kde/src/bluedevil/src/kded/receivefilejob.cpp:
  102      switch (FileReceiverSettings::self()->autoAccept()) {
  103      case 0: // Never auto-accept transfers
  104          showNotification();
  105          break;
  106  
  107:     case 1: // Auto-accept only from trusted devices
  108:         if (device->isTrusted()) {
  109:             qCDebug(BLUEDEVIL_KDED_LOG) << "Auto-accepting transfer for trusted device";
  110              slotAccept();
  111          } else {
  112              showNotification();
  113          }
  114          break;
```

pv, 2024-06-29:
Trusted devices are allowed to do things like connect and use bluetooth services exposed by the local host, without requiring explicit confirmation from user (technically the confirmations go through org.bluez.Agent1 dbus interface, maybe KDE etc support it).

For example, if you untrust your earbuds, they can't connect A2DP/HFP automatically any more when powered on. Conversely, if you set your earbuds as trusted, they automatically connect and appear as audio device when taken out from the case. (Most earbuds automatically connect to the device they were last used with.)

weliem, 2024-06-29:
So are you saying that trusting allows for autoconnecting of devices? I only do LE stuff but I can guarantee you that no autoconnection will take place if I trust a device....

pv, 2024-06-29:
No, I'm saying trusting generally allows the device to use local services.

weliem, 2024-06-29:
But if the device is paired already, why wouldn't a connection be allowed? Is pairing only uni-directional?

pv, 2024-06-29:
BlueZ asks for authorization when connecting services, trusted means it will be granted directly.

For the question of LE connections, the source code says this:
```
static DBusMessage *dev_disconnect(DBusConnection *conn, DBusMessage *msg,
                                                        void *user_data)
{
        struct btd_device *device = user_data;

        /*
         * If device is not trusted disable connections through passive
         * scanning until Device1.Connect is called
         */
        if (device->auto_connect && !device->trusted) {
                device->disable_auto_connect = TRUE;
                device_set_auto_connect(device, FALSE);
        }
```
I think here the setting controls whether host will try to autoconnect. I guess you can just read the sources to find out exactly what it is does.

closed 2025-07-23 by github-actions[bot]