bluez/bluez #890: Cannot connect Smart Wireless Keyboard (VG-KBD1000) on Linux (Fedora & Arch)

Ramen-LadyHKG, 2024-07-03:
## Steps to reproduce
1. Pair and connect the keyboard in (Gnome/Plasma)'s Bluetooth Setting
2. Pair Successfully but didn't recognized as a `keyboard` but a `peripheral`
3. When Connect to the keyboard and start tpying,  it disconnect and reconnect on itself
4. None of key triggered when pressing.

## Hardware info (Keyboard):
> bluetoothctl
>> info E4:E0:C5:9C:99:28
> > > 	Device E4:E0:C5:9C:99:28 (public)
> > > 	Name: Samsung Wireless Keyboard
> > > 	Alias: Samsung Wireless Keyboard
> > > 	Class: 0x000005cc (1484)
> > > 	Paired: yes
> > > 	Bonded: yes
> > > 	Trusted: yes
> > >	Blocked: no
> > >	Connected: no
> > >	WakeAllowed: yes
> > >	LegacyPairing: no
> > >	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
> > >	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
> > >	Modalias: usb:v04E8p2080d0100

## System info (Fedora 40):
> **`Linux Kernel` version:** `6.9.7-1.surface.fc40.x86_64`
> **`bluez` version:** `bluez-5.76-1.fc40.src.rpm`

## System info (Arch Linux):
> **`Linux Kernel` version:** `6.9.7-arch1-1-surface`
> **`bluez` version:** `bluez-5.76-1`

## Error shown in system log:
```
Jul 03 22:34:27 sp5-20230727-f38 bluetoothd[1937]: profiles/input/device.c:hidp_report_req_timeout() Device E4:E0:C5:9C:99:28 HIDP GET_REPORT request timed out
Jul 03 22:34:30 sp5-20230727-f38 bluetoothd[1937]: profiles/input/device.c:hidp_report_req_timeout() Device E4:E0:C5:9C:99:28 HIDP GET_REPORT request timed out
Jul 03 22:34:33 sp5-20230727-f38 bluetoothd[1937]: profiles/input/device.c:hidp_report_req_timeout() Device E4:E0:C5:9C:99:28 HIDP GET_REPORT request timed out
Jul 03 22:34:33 sp5-20230727-f38 bluetoothd[1937]: profiles/input/device.c:hidp_send_message() BT socket not connected
Jul 03 22:34:58 sp5-20230727-f38 bluetoothd[1937]: profiles/input/device.c:hidp_report_req_timeout() Device E4:E0:C5:9C:99:28 HIDP GET_REPORT request timed out
Jul 03 22:35:01 sp5-20230727-f38 bluetoothd[1937]: profiles/input/device.c:hidp_report_req_timeout() Device E4:E0:C5:9C:99:28 HIDP GET_REPORT request timed out
Jul 03 22:35:04 sp5-20230727-f38 bluetoothd[1937]: profiles/input/device.c:hidp_report_req_timeout() Device E4:E0:C5:9C:99:28 HIDP GET_REPORT request timed out
```

Vudentz, 2024-07-03:
> ## Error shown in system log:
>
> ```
> Jul 03 22:34:27 sp5-20230727-f38 bluetoothd[1937]: profiles/input/device.c:hidp_report_req_timeout() Device E4:E0:C5:9C:99:28 HIDP GET_REPORT request timed out
> Jul [...]

You may need a HID driver if it doesn't work with hidraw which I assume is what it ends up selecting.

Ramen-LadyHKG, 2024-07-04:
> You may need a HID driver if it doesn't work with hidraw which I assume is what it ends up selecting.

umm how to achieve that?
I though HID driver is embedded in linux kernel.

Ramen-LadyHKG, 2024-09-08:
I just found out, it does connect and able to type

but has to wait for about a minute

markmb88, 2024-11-23:
I have the same issue with my Keychron K1 SE V5. It will connect but keyboard is unresponsive. after a few seconds keyboard will disconnect and reconnect and then it works fine.

```
Nov 23 12:09:14 archlinux kernel: apple 0005:05AC:024F.000C: unknown main item tag 0x0
Nov 23 12:09:14 archlinux kernel: apple 0005:05AC:024F.000C: Non-apple keyboard detected; function keys will default to fnmode=2 behavior
Nov 23 12:09:14 archlinux kernel: input: Keychron K1 SE as /devices/virtual/misc/uhid/0005:05AC:024F.000C/input/input36
Nov 23 12:09:14 archlinux kernel: apple 0005:05AC:024F.000C: input,hidraw0: BLUETOOTH HID v1.03 Keyboard [Keychron K1 SE] on 20:23:51:09:c5:4f
Nov 23 12:09:14 archlinux systemd-logind[609]: Watching system buttons on /dev/input/event2 (Keychron K1 SE)
Nov 23 12:09:35 archlinux bluetoothd[607]: profiles/input/device.c:hidp_report_req_timeout() Device DC:2C:26:2B:7C:7C HIDP GET_REPORT request timed out
Nov 23 12:09:45 archlinux kernel: apple 0005:05AC:024F.000D: unknown main item tag 0x0
Nov 23 12:09:45 archlinux kernel: apple 0005:05AC:024F.000D: Non-apple keyboard detected; function keys will default to fnmode=2 behavior
Nov 23 12:09:45 archlinux kernel: input: Keychron K1 SE as /devices/virtual/misc/uhid/0005:05AC:024F.000D/input/input37
Nov 23 12:09:45 archlinux kernel: apple 0005:05AC:024F.000D: input,hidraw0: BLUETOOTH HID v1.03 Keyboard [Keychron K1 SE] on 20:23:51:09:c5:4f
```

Ramen-LadyHKG, 2025-05-11:
> I have the same issue with my Keychron K1 SE V5. Once paired and after a reboot, It will connect but keyboard is unresponsive. after a few seconds keyboard will disconnect and reconnect and then it works fine.
> 
> ```
> Nov 23 12:09:14 archlinux kernel: apple 0005:05AC:024F.000C: unknown main item tag 0x0
> Nov 23 12:09:14 archlinux kernel: apple 0005:05AC:024F.000C: Non-apple keyboard detected; function keys will default to fnmode=2 behavior
> Nov 23 12:09:14 archlinux kernel: input: Keychron K1 SE as /devices/virtual/misc/uhid/0005:05AC:024F.000C/input/input36
> Nov 23 12:09:14 archlinux kernel: apple 0005:05AC:024F.000C: input,hidraw0: BLUETOOTH HID v1.03 Keyboard [Keychron K1 SE] on 20:23:51:09:c5:4f
> Nov 23 12:09:14 archlinux systemd-logind[609]: Watching system buttons on /dev/input/event2 (Keychron K1 SE)
> Nov 23 12:09:35 archlinux bluetoothd[607]: profiles/input/device.c:hidp_report_req_timeout() Device DC:2C:26:2B:7C:7C HIDP GET_REPORT request timed out
> Nov 23 12:09:45 archlinux kernel: apple 0005:05AC:024F.000D: unknown main item tag 0x0
> Nov 23 12:09:45 archlinux kernel: apple 0005:05AC:024F.000D: Non-apple keyboard detected; function keys will default to fnmode=2 behavior
> Nov 23 12:09:45 archlinux kernel: input: Keychron K1 SE as /devices/virtual/misc/uhid/0005:05AC:024F.000D/input/input37
> Nov 23 12:09:45 archlinux kernel: apple 0005:05AC:024F.000D: input,hidraw0: BLUETOOTH HID v1.03 Keyboard [Keychron K1 SE] on 20:23:51:09:c5:4f
> ```

mine takes about 20 seconds.

Your keyboard is pretty new, it has `bluetooth 5.1`.

 I thought it was because my bluetooth keyboard was too old, only has `bluetooth2.0`

closed 2025-09-21 by github-actions[bot]