bluez/bluez #892: DualSense Audio Support

citrizon, 2024-07-05:
Hello! I am an Arch Linux user and i'm powering my bluetooth stack using Bluez. I configured my system and everything works fine.
I can connect my DS4 controller to my PC, however when i try to connect a headphone, there is no sound output. I am checking the error through `dmesg`. the result is here:
```r
[ 3120.827352] input: Wireless Controller as /devices/virtual/misc/uhid/XXXX:XXXX:XXXX:XXXX/input/input34
[ 3120.828103] input: Wireless Controller Motion Sensors as /devices/virtual/misc/uhid/XXXX:XXXX:XXXX:XXXX/input/input35
[ 3120.828493] input: Wireless Controller Touchpad as /devices/virtual/misc/uhid/XXXX:XXXX:XXXX:XXXX/input/input36
[ 3120.829440] playstation XXXX:XXXX:XXXX:XXXX: Registered DualShock4 controller hw_version=0x00006404 fw_version=0x00007008
[ 3174.454748] playstation XXXX:XXXX:XXXX:XXXX: Unhandled reportID=24
[ 3174.465644] playstation XXXX:XXXX:XXXX:XXXX: Unhandled reportID=19
[ 3174.482671] playstation XXXX:XXXX:XXXX:XXXX: Unhandled reportID=19
[ 3174.501570] playstation XXXX:XXXX:XXXX:XXXX: Unhandled reportID=19
[ 3174.512648] playstation XXXX:XXXX:XXXX:XXXX: Unhandled reportID=19
[ 3174.536696] playstation XXXX:XXXX:XXXX:XXXX: Unhandled reportID=19
[ 3174.549577] playstation XXXX:XXXX:XXXX:XXXX: Unhandled reportID=19
[ 3174.561715] playstation XXXX:XXXX:XXXX:XXXX: Unhandled reportID=19
[ 3174.579633] playstation XXXX:XXXX:XXXX:XXXX: Unhandled reportID=19
[ 3174.596593] playstation XXXX:XXXX:XXXX:XXXX: Unhandled reportID=19
```
(Potentially sensitive information is redacted.)

I would like to learn about this problem and i would like to contribute to fix this issue (hopefully... maybe)

Vudentz, 2024-07-08:
> Hello! I am an Arch Linux user and i'm powering my bluetooth stack using Bluez. I configured my system and everything works fine. I can connect my DS4 controller to my PC, however when i try to [...]

Doesn't seem like it is using A2DP/HFP for audio, so I guess it is transmitted over HID so it needs a kernel driver that can handle these reports.

mirh, 2025-02-01:
You are correct. Previous episodes:
https://github.com/chrippa/ds4drv/pull/80
https://github.com/sensepost/dual-pod-shock/issues/1

egormanga, 2025-07-23:
No, it's only quad-channel when wired. Over BT it's two separate stereo streams, and you're right to guess that the non-haptic ones are Opus.