bluez/bluez #893: btmgmt: --index option does not work in non-interactive mode [patch]

adopmeer, 2024-07-06:
### btmgmt in interactive mode:

```
$ btmgmt 
[mgmt]# info
[mgmt]# Index list with 1 item         <==== notice generic listing of all adapters
[mgmt]# hci0:	Primary controller
```

```
$ btmgmt -i 0
[hci0]# info
[hci0]# hci0:	Primary controller      <==== info for supplied index
```

```
$ btmgmt -i 1
[hci1]# info
[hci1]# Reading hci1 info failed with status 0x11 (Invalid Index)
```

### btmgmt in non-interactive mode:

```
$ btmgmt -i 0 info
Index list with 1 item                  <==== notice generic listing of all adapters
hci0:	Primary controller
```

```
$ btmgmt -i 1 info
Index list with 1 item                  <==== notice generic listing of all adapters
hci0:	Primary controller
```

The reason the `--index` option does not work in non-interactive mode is because the call to `mgmt_set_index()` is made too
late. In this mode the entire command line is run by the call to `bt_shell_attach()` however the call to `mgmt_set_index()`
comes after that. Fix this by hoisting the call to `mgmt_set_index()` above the call to `bt_shell_attach()` like so:

```diff
--- btmgmt.c.ORIG	2024-07-06 21:06:48.826453028 +0200
+++ btmgmt.c	2024-07-06 21:07:10.802429413 +0200
@@ -51,8 +51,8 @@
 		return EXIT_FAILURE;
 	}
 
-	bt_shell_attach(fileno(stdin));
 	mgmt_set_index(index_option);
+	bt_shell_attach(fileno(stdin));
 	status = bt_shell_run();
 
 	mgmt_remove_submenu();
```

Vudentz, 2024-07-08:
@adopmeer patch looks good, do you care the send it to the mailing list:

https://github.com/bluez/bluez/blob/master/HACKING#L98

adopmeer, 2024-07-09:
OK, this took some googling because the information in the HACKING document is certainly not enough to get `git send-email` working, but the patch has been sent to the mailing list. The online mailing list archive shows the email as received.

closed 2024-07-09 by BluezTestBot