bluez/bluez #894: bluez 5.75 down after errror: "kernel: Bluetooth: Unexpected start frame (len 14)"

MaximBoguk, 2024-07-08:
bluez on dell laptop on Ubuntu with 6.9.3 kernel and bluez 5.76 goes down few times per day on active usage
with error "Bluetooth: Unexpected start frame (len 14)".
Only bluetooth mouse and keyboard connected.

Can compile/debug/run gdb/etc  if need but need guidance for future actions.

igorog, 2024-08-06:
I get the same error from time to time, then it goes away for a while. I didn’t find any patterns.
Operating System: Arch Linux 
KDE Plasma Version: 6.1.3
KDE Frameworks Version: 6.4.0
Qt Version: 6.7.2
Kernel Version: 6.10.3-zen1-1-zen (64-bit)
Graphics Platform: X11
Processors: 12 × 12th Gen Intel® Core™ i7-1255U
Memory: 31.1 ГиБ of RAM
Graphics Processor: Mesa Intel® Graphics

pmhahn, 2024-09-13:
Also on Debian-12-Buster with `6.10.6+bpo-amd64`.

Bluez: 5.66-1+deb12u2

Strangely only my `BLUETOOTH HID v0.22 Mouse [Logitech MX Ergo Multi-Device Trackball ]` is affected, but my `EPOS ADAPT 360` is not affected and continues working.

Workaround: disable and re-enable Bluetooth.

closed 2025-07-22 by github-actions[bot]